grodansparadis/vscp #45: Typo in VSCPworks: "Full Uppdate"

waaslandwolf, 2015-01-12:
Ake,

There's a small typo in VSCPworks v 1.0.0.47: in the Device Conf window there is a check box "Full Uppdate" on the bottom right of the screen. Obviously "Update" is spelled with only one "p".

Obviously it is a minor issue but just reporting it so that it can be fixed it in some next release..

David.

grodansparadis, 2015-01-12:
Thanks!  Fixed.

Hopefully there will be a new release out this week that mostly is a fix 
for some issues in VSCP Works that never got attended to before. It will 
be better but not perfect. ;-)

/Ake

waaslandwolf skrev den 2015-01-12 12:36:

> Ake,
>
> There's a small typo in VSCPworks v 1.0.0.47: in the Device Conf
> window there is a check box "Full Uppdate" on the bottom right of the
> screen. Obviously "Update" is spelled with only one [...]

waaslandwolf, 2015-01-12:
Looking forward to that new release. Will the pagination issue be fixed in that one as well?

grodansparadis, 2015-01-12:
waaslandwolf skrev den 2015-01-12 12:52:
>
> Looking forward to that new release. Will the pagination issue be 
> fixed in that one as well?
>
Hopefully. The code beneath is totally rewritten. The paging should at 
least work a lot better I hope.

/Ake

-- 
Be Hungry - Stay Foolish!
Åke Hedman, Grodans Paradis AB / Paradise of the Frog
Brattbergavägen 17, 82050 LOS, SWEDEN
Phone: +46 (0)840011835 CellPhone: +46 (0)730-533146
http://www.grodansparadis.com http://www.vscp.org

---
Detta e-postmeddelande har sökts igenom efter virus med antivirusprogram från Avast.
http://www.avast.com

waaslandwolf, 2015-01-12:
Great to hear. I'll be sure to put it through its paces!

---
This email has been checked for viruses by Avast antivirus software.
http://www.avast.com

closed 2015-02-02 by grodansparadis