grodansparadis/vscp #46: Feature request: filter heartbeat messages

waaslandwolf, 2015-01-13:
Ake,

I find that when monitoring VSCP traffic with the VSCP works monitor window, after a while the receive event window gets flooded with heartbeat messages. It would be nice if there was an option to filter out these events so that they don't appear in the received event window any more. Perhaps this is something of value which is easy to implement?

David.

grodansparadis, 2015-01-13:
Hi David,

cant you just set a filter that filter them out? With a bit of thougth 
you should be able to receive all other events or at least the once you 
are interested in.

But generally I agree it would have been nice with a mechanism where you 
could tell which events you want to see and which you don't want.  Has 
been on my list of todo's for a while and some day it might be a reality 
even.

/Ake

waaslandwolf skrev den 2015-01-13 10:58:

> Ake,
>
> I find that when monitoring VSCP traffic with the VSCP works monitor
> window, after a while the receive event window gets flooded with
> heartbeat messages. It would be nice if there was an [...]

waaslandwolf, 2015-01-13:
Ake,

Yes setting a filter make the most sense, but when I tested this by 
selecting Configuration - Set Filter, I got a message "Sorry this 
functionality isn't available yet..."

David.

---
This email has been checked for viruses by Avast antivirus software.
http://www.avast.com

grodansparadis, 2015-01-13:
No what you get there is exactly what we talk about here. But you can 
set a filter on the interface also.

/Ake

waaslandwolf skrev den 2015-01-13 11:29:

> Ake,
>
> Yes setting a filter make the most sense, but when I tested this by
> selecting Configuration - Set Filter, I got a message "Sorry this
> functionality isn't available yet..."
>
> David.
>
> > Hi David,
> >
> > cant you just set a filter that filter them out? With a bit of thougth
> > you should be able to receive all other events or at least the once you
> > are interested in.
> >
> > But generally I agree it would have been nice with a mechanism where you
> > could tell which events you want to see and which you don't want. Has
> > been on my list of todo's for a while and some day it might be a reality
> > even.
> >
> > /Ake
> >
> > waaslandwolf skrev den 2015-01-13 10:58:
> > >
> > > Ake,
> > >
> > > I find that when monitoring VSCP traffic with the VSCP works monitor
> > > window, after a while the receive event window gets flooded with
> > > heartbeat messages. It would be nice if there was an option to filter
> > > out these events so that they don't appear in the received event
> > > window any more. Perhaps this is something of value which is easy to
> > > implement?
> > >
> > > David.
> > >
> > >
> >
> > Be Hungry - Stay Foolish!
> > Åke Hedman, Grodans Paradis AB / Paradise of the Frog
> > Brattbergavägen 17, 82050 LOS, SWEDEN
> > Phone: +46 (0)840011835 CellPhone: +46 (0)730-533146
> > http://www.grodansparadis.com http://www.vscp.org
> >
> >
> >
> > ---
> > Detta e-postmeddelande har sökts igenom efter virus med
> > antivirusprogram från Avast.
> > http://www.avast.com
> >
> > 
> >
>
>
>
> ---
> This email has been checked for viruses by Avast antivirus software.
> http://www.avast.com

waaslandwolf, 2015-01-13:
Ah I see... I'll give that a go.

---
This email has been checked for viruses by Avast antivirus software.
http://www.avast.com

closed 2015-02-02 by grodansparadis

waaslandwolf, 2015-02-18:
Ake,

Just wanted to let you know that I am enjoying the new filter 
functionality every day. Such a great feature to be able to save and 
load filters on the fly! I have >20 nodes on the bus now so filtering 
events sure helps troubleshooting!

David.

---
This email has been checked for viruses by Avast antivirus software.
http://www.avast.com

grodansparadis, 2015-02-18:
David,

Good to here. I actually find it useful myself to. ;-)

/Ake

waaslandwolf skrev den 2015-02-18 20:35:

> Ake,
>
> Just wanted to let you know that I am enjoying the new filter
> functionality every day. Such a great feature to be able to save and
> load filters on the fly! I have >20 nodes on the bus now so filtering
> events sure helps troubleshooting!
>
> David.
>
> >
> > Closed #46 <https://github.com/grodansparadis/vscp_software/issues/46>.
> >
> > 
> >
>
>
>
> ---
> This email has been checked for viruses by Avast antivirus software.
> http://www.avast.com