grodansparadis/vscp #47: VSCPworks 1.0.0.54 core dumps

waaslandwolf, 2015-01-23:
Ake,

I have compiled & installed yesterday's vscp_software git pull. The deamon seems ok, but VSCPworks core dumps.

In one terminal I do:
sudo ip link set can0 up type can bitrate 250000 sample-point 0.875
sudo /usr/local/bin/vscpd -s &

In the other terminal I do:
root@ubuntu-john:~/vscp/vscp_software#  /usr/local/bin/vscpworks &

- Works starts
- I open a client window, select interface, Edit, test connector or get interfaces -> core dump
18:47:27: Debug: ============================================================
18:47:27: Debug: Connect in progress with server localhost:9598
18:47:27: Debug: ============================================================
[1]+  Segmentation fault      (core dumped) /usr/local/bin/vscpworks

- I open a device configuration window, select my existing interface, click ok -> core dump
18:53:25: Debug: strInterface = 
18:53:25: Debug: localhost:9598;admin;secret;
18:53:25: Debug: strHostname = 
18:53:25: Debug: localhost:9598
18:53:25: Debug: strUsername = 
18:53:25: Debug: admin
18:53:25: Debug: strPassword = 
18:53:25: Debug: secret
18:53:25: Debug: ============================================================
18:53:25: Debug: Connect in progress with server localhost:9598
18:53:25: Debug: ============================================================
[1]+  Segmentation fault      (core dumped) /usr/local/bin/vscpworks

waaslandwolf, 2015-01-23:
Ake,

I did some further testing and found I cannot connect to the deamon on my Ubuntu machine from my Windows machine. Perhaps VSCPworks crashes because of the deamon?

Deamon starts with:
root@ubuntu-john:~# /usr/local/bin/vscpd &
../src/common/socket.cpp(295): assert "app" failed in Init(): sockets can't be initialized without wxApp
20:11:53: Debug: ControlObject: Configfile =/etc/vscp/vscpd.conf

vscpd.conf is pretty much stock. Just changed GUID and enabled level II socket can driver.

grodansparadis, 2015-01-23:
Hi,

My standard test setup here is 64-bit win 7 / 32-bit  Ubuntu - VSCP 
Works <->  64-bit Ubuntu / 64-bit Debin / 32-bit Debian - Daemon and I 
have no problems at all. I probably connect several hundreds of times a 
day.

So something must be different in our setups?

/Ake

waaslandwolf skrev den 2015-01-23 20:16:

> Ake,
>
> I did some further testing and found I cannot connect to the deamon on
> my Ubuntu machine from my Windows machine. Perhaps VSCPworks crashes
> because of the deamon?
>
> Deamon starts with: [...]

waaslandwolf, 2015-01-24:
Ake,

My test setup is:
- Ubuntu 14.04.4 LTS with VSCPworks 1.0.0.54
- Windows XP SP3 with VSCPworks 1.0.0.43

Perhaps it is indeed something on my machine. I've rolled back to an 
earlier git version and get the same core dump. I'll investigate some more.

David.

---
This email has been checked for viruses by Avast antivirus software.
http://www.avast.com

grodansparadis, 2015-01-24:
Hi

wx 2.8.12?   32-bit?

XP is not supported from this version.  It may work or not.

/Ake

waaslandwolf skrev den 2015-01-24 09:09:

> Ake,
>
> My test setup is:
> - Ubuntu 14.04.4 LTS with VSCPworks 1.0.0.54
> - Windows XP SP3 with VSCPworks 1.0.0.43
>
> Perhaps it is indeed something on my machine. I've rolled back to an
> earlier git version and get the same core dump. I'll investigate some 
> more.
>
> David.
>
> > Hi,
> >
> > My standard test setup here is 64-bit win 7 / 32-bit Ubuntu - VSCP
> > Works <-> 64-bit Ubuntu / 64-bit Debin / 32-bit Debian - Daemon and I
> > have no problems at all. I probably connect several hundreds of times a
> > day.
> >
> > So something must be different in our setups?
> >
> > /Ake
> >
> > waaslandwolf skrev den 2015-01-23 20:16:
> > >
> > > Ake,
> > >
> > > I did some further testing and found I cannot connect to the deamon on
> > > my Ubuntu machine from my Windows machine. Perhaps VSCPworks crashes
> > > because of the deamon?
> > >
> > > Deamon starts with:
> > > root@ubuntu-john:~# /usr/local/bin/vscpd &
> > > ../src/common/socket.cpp(295): assert "app" failed in Init(): sockets
> > > can't be initialized without wxApp
> > > 20:11:53: Debug: ControlObject: Configfile =/etc/vscp/vscpd.conf
> > >
> > > vscpd.conf is pretty much stock. Just changed GUID and enabled level
> > > II socket can driver.
> > >
> > >
> > 
> > >
> >
> > Be Hungry - Stay Foolish!
> > Åke Hedman, Grodans Paradis AB / Paradise of the Frog
> > Brattbergavägen 17, 82050 LOS, SWEDEN
> > Phone: +46 (0)840011835 CellPhone: +46 (0)730-533146
> > http://www.grodansparadis.com http://www.vscp.org
> >
> >
> >
> > ---
> > Detta e-postmeddelande har sökts igenom efter virus med
> > antivirusprogram från Avast.
> > http://www.avast.com
> >
> > 
> >
>
>
>
> ---
> This email has been checked for viruses by Avast antivirus software.
> http://www.avast.com

waaslandwolf, 2015-01-24:
wx3, 32 bit.

I'm not upgrading my Win XP VSCPworks 43 just yet:-)

---
This email has been checked for viruses by Avast antivirus software.
http://www.avast.com

waaslandwolf, 2015-01-24:
Ake,

I did a fresh git pull, ./configure; make clean; make; make install. Then edited vscpd.conf to enable the can driver, started the deamon and attempted to connect with VSCPworks.

The log files from this testing exercise are [here](http://www.steeman.be/vscpworks%201.0.0.55%20install%20logs.zip), perhaps you can find something unusual.

Thanks in advance,
David.

grodansparadis, 2015-01-25:
David,

thanks for all the info.

I have tried to get the same messages when I start the daemon as you do 
for most of this afternoon but can't. I am not able to repeat the crash 
either. So this is very strange to me. But the search goes on.

It looks like you get some asserts for the  the path string to the 
configuration file. Which is

What pussels me is that you get debug outputs but is running a release 
compiled daemon.

We start with

If you issue   "wx-config --list" what does that give you?

/Ake

waaslandwolf skrev den 2015-01-24 21:04:

> Ake,
>
> I did a fresh git pull, ./configure; make clean; make; make install.
> Then edited vscpd.conf to enable the can driver, started the deamon
> and attempted to connect with VSCPworks.
>
> The log [...]

waaslandwolf, 2015-01-26:
Ake,

Thanks for your effort.

 >If you issue "wx-config --list" what does that give you?

john@ubuntu-john:~$ *su -*
Password:

root@ubuntu-john:~# cd vscp

root@ubuntu-john:~/vscp# *wx-config --list*

     Default config is gtk2-unicode-3.0

   Default config will be used for output

   Alternate matches:
     base-unicode-3.0

root@ubuntu-john:~/vscp# *ip link set can0 up type can bitrate 250000 
sample-point 0.875*

root@ubuntu-john:~/vscp# *ip link*
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode 
DEFAULT group default
     link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast 
state UP mode DEFAULT group default qlen 1000
     link/ether 00:16:d3:80:10:aa brd ff:ff:ff:ff:ff:ff
3: can0: <NOARP,UP,LOWER_UP,ECHO> mtu 16 qdisc pfifo_fast state UNKNOWN 
mode DEFAULT group default qlen 10
     link/can
4: wlan0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast 
state DOWN mode DORMANT group default qlen 1000
     link/ether 00:60:b3:61:9a:b0 brd ff:ff:ff:ff:ff:ff

---
This email has been checked for viruses by Avast antivirus software.
http://www.avast.com

grodansparadis, 2015-01-26:
Hi,

looks OK.  This is really strange.   If you have possibility could you 
clean the local repository and clone it again and build from fresh and 
see if that makes a difference.  The only difference I can see between 
your setup and my is that I work on a 64-bit platform. But that should 
not matter anyway.  But we will hunt it down!

/Ake

waaslandwolf skrev den 2015-01-26 09:30:

> Ake,
>
> Thanks for your effort.
>
> >If you issue "wx-config --list" what does that give you?
>
> john@ubuntu-john:~$ *su -*
> Password:
>
> root@ubuntu-john:~# cd vscp
>
> root@ubuntu-john:~/vscp# *wx-config --list*
>
> Default config is gtk2-unicode-3.0
>
> Default config will be used for output
>
> Alternate matches:
> base-unicode-3.0
>
> root@ubuntu-john:~/vscp# *ip link set can0 up type can bitrate 250000
> sample-point 0.875*
>
> root@ubuntu-john:~/vscp# *ip link*
> 1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode
> DEFAULT group default
> link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
> 2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast
> state UP mode DEFAULT group default qlen 1000
> link/ether 00:16:d3:80:10:aa brd ff:ff:ff:ff:ff:ff
> 3: can0: <NOARP,UP,LOWER_UP,ECHO> mtu 16 qdisc pfifo_fast state UNKNOWN
> mode DEFAULT group default qlen 10
> link/can
> 4: wlan0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast
> state DOWN mode DORMANT group default qlen 1000
> link/ether 00:60:b3:61:9a:b0 brd ff:ff:ff:ff:ff:ff
>
>
>
>
> > David,
> >
> > thanks for all the info.
> >
> > I have tried to get the same messages when I start the daemon as you do
> > for most of this afternoon but can't. I am not able to repeat the crash
> > either. So this is very strange to me. But the search goes on.
> >
> > It looks like you get some asserts for the the path string to the
> > configuration file. Which is
> >
> > What pussels me is that you get debug outputs but is running a release
> > compiled daemon.
> >
> > We start with
> >
> > If you issue "wx-config --list" what does that give you?
> >
> > /Ake
> >
> >
> > waaslandwolf skrev den 2015-01-24 21:04:
> > >
> > > Ake,
> > >
> > > I did a fresh git pull, ./configure; make clean; make; make install.
> > > Then edited vscpd.conf to enable the can driver, started the deamon
> > > and attempted to connect with VSCPworks.
> > >
> > > The log files from this testing exercise are here
> > > <http://www.steeman.be/vscpworks%201.0.0.55%20install%20logs.zip>,
> > > perhaps you can find something unusual.
> > >
> > > Thanks in advance,
> > > David.
> > >
> > >
> > 
> > >
> >
> > Be Hungry - Stay Foolish!
> > Åke Hedman, Grodans Paradis AB / Paradise of the Frog
> > Brattbergavägen 17, 82050 LOS, SWEDEN
> > Phone: +46 (0)840011835 CellPhone: +46 (0)730-533146
> > http://www.grodansparadis.com http://www.vscp.org
> >
> >
> >
> > ---
> > Detta e-postmeddelande har sökts igenom efter virus med
> > antivirusprogram från Avast.
> > http://www.avast.com
> >
> > 
> >
>
>
>
> ---
> This email has been checked for viruses by Avast antivirus software.
> http://www.avast.com

grodansparadis, 2015-01-26:
David,

do you get my replies. They never appear on the list as far as I can see.

/Ake

waaslandwolf skrev den 2015-01-24 21:04:

> Ake,
>
> I did a fresh git pull, ./configure; make clean; make; make install.
> Then edited vscpd.conf to enable the can driver, started the deamon
> and attempted to connect with VSCPworks.
>
> The log [...]

waaslandwolf, 2015-01-26:
yes I do :-)

---
This email has been checked for viruses by Avast antivirus software.
http://www.avast.com

grodansparadis, 2015-01-26:
I don't know if it's just this thread bit replies does not go to the 
list.  At least I does not see my replies there.

/Ake

waaslandwolf, 2015-01-26:
Ah ok, I see what you mean now. I've checked your replies in this 
thread, they all come from github.com and the vscp google group is not 
in the to: or cc:. So indeed, it seems that these mails don't go to the 
group.

---
This email has been checked for viruses by Avast antivirus software.
http://www.avast.com

waaslandwolf, 2015-01-26:
Ake,

I removed the vscp_software folder which is synced to git and cloned again. I get the same issue. vscp_software used to work just fine on my machine, until I did a git pull of v1.0.0.55 and installed it. Nothing else changed, it is a VSCP machine only. Maybe I fumbled a bit with synergy and x11vnc, but that shouldn't impact VSCP.

Here's the logs of today's session:
[Git clone, ./configure; make; make install](http://www.steeman.be/fileshare/installog.txt)
[Start vscpd](http://www.steeman.be/fileshare/runlog-vscpd.txt)
[Start VSCPworks and test interface connection](http://www.steeman.be/fileshare/runlog-vscpworks.txt)

grodansparadis, 2015-01-26:
David,

can you replace the Makefile that is in src/vscp/vscpworks with the 
attached Makefile and then go to this folder and do "make clean;make" 
and then (if it builds OK) "./vscpworks"

I want to check that the problems is not related to the new 
communication library.

/Ake

# =========================================================================
#                      Autogenerated from Makefile.in         
# =========================================================================

prefix = /usr/local
exec_prefix = ${prefix}
INSTALL = /usr/bin/install -c
top_srcdir = ../../..
top_wxsrcdir = @top_wxsrcdir@
top_builddir = ../../..
top_wxbuilddir = @top_wxbuilddir@
CXX = g++
CXXFLAGS = -g -O2 
CPPFLAGS =  -I/usr/local/lib/wx/include/gtk2-unicode-3.0 -I/usr/local/include/wx-3.0 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -D NS_ENABLE_SSL -DwxUSE_GUI=1 -I../common -I../../common
LDFLAGS = 
EXTRALIBS =  -lmosquittopp -lssl -lpcap -lz -lrt -lm -lcrypto -lpthread -ldl -lcurses  -L/usr/local/lib -pthread   -lwx_gtk2u_xrc-3.0 -lwx_gtk2u_html-3.0 -lwx_gtk2u_qa-3.0 -lwx_gtk2u_adv-3.0 -lwx_gtk2u_core-3.0 -lwx_baseu_xml-3.0 -lwx_baseu_net-3.0 -lwx_baseu-3.0 
STRIP = strip
INSTALL_PROGRAM = ${INSTALL}
INSTALL_DATA = ${INSTALL} -m 644
INSTALL_DIR = /usr/bin/install -c -d
srcdir = .
top_srcdir = .
top_builddir =
bindir = ${exec_prefix}/bin
libdir = ${exec_prefix}/lib
datadir = ${prefix}/share
includedir = ${prefix}/include
DLLPREFIX = lib

### Variables: ###

DESTDIR = 
WX_RELEASE = 2.8
LIBDIRNAME = $(top_wxbuilddir)lib

VSCPWORKS_OBJECTS =  \
	vscpworks.o dlgvscpmsg.o frmmain.o frmvscpsession.o dlgnewvscpsession.o dlgconfiguration.o dlgvscpinterfacesettings.o dlgvscpfilter.o vscpbootloader.o dlgvscptrmit.o frmdeviceconfig.o frmscanfordevices.o dlgselectdaemoninterface.o frmmdfeditor.o dlgabout.o ../common/canalsuperwrapper.o ../../common/crc.o ../../common/crc8.o ../../common/net_skeleton.o ../common/dllwrapper.o ../common/guid.o threadvscpwork.o ../common/vscphelper.o ../common/vscpeventhelper.o ../common/vscpremotetcpif.o ../common/mdf.o ../common/register.o vscptxobj.o gridtable.o dlgeditlevel1dmrow.o dlgsetfiltermask.o dlgabstractionedit.o devicebootloaderwizard.o bootdevice.o bootdevice_pic1.o bootdevice_vscp.o merlin.o frmdaemondecisionmartrixeditor.o frmdaemonvariableeditor.o frmsimpleuserinterfacegenerator.o dlgeventfilter.o dlgclasstypeselect.o $(RESOURCEOBJECT)

### Targets: ###

all: vscpworks

install: all
	$(INSTALL_PROGRAM) vscpworks $(bindir)

uninstall: 

install-strip: install

clean: 
	rm -rf ./.deps ./.pch
	rm -f ./*.o
	rm -f ../common/*.o
	rm -f ../../common/*.o
	rm -f vscpworks

distclean: clean
	rm -f config.cache config.log config.status bk-deps bk-make-pch shared-ld-sh Makefile

vscpworks: $(VSCPWORKS_OBJECTS)
	$(CXX) -o $@ $(VSCPWORKS_OBJECTS) $(LDFLAGS) -L$(LIBDIRNAME) $(WXLIBS) $(EXTRALIBS)
	

# Include dependency info, if present:
-include .deps/*.d

.PHONY: all install uninstall clean distclean data

waaslandwolf, 2015-01-26:
Ok let me test that. In the future can you attach the file as a proper attachment instead of copying it in? If I copy and paste it the formatting gets screwed up.

grodansparadis, 2015-01-26:
Her it is

# =========================================================================
#                      Autogenerated from Makefile.in         
# =========================================================================

prefix = /usr/local
exec_prefix = ${prefix}
INSTALL = /usr/bin/install -c
top_srcdir = ../../..
top_wxsrcdir = @top_wxsrcdir@
top_builddir = ../../..
top_wxbuilddir = @top_wxbuilddir@
CXX = g++
CXXFLAGS = -g -O2 
CPPFLAGS =  -I/usr/local/lib/wx/include/gtk2-unicode-3.0 -I/usr/local/include/wx-3.0 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -D NS_ENABLE_SSL -DwxUSE_GUI=1 -I../common -I../../common
LDFLAGS = 
EXTRALIBS =  -lmosquittopp -lssl -lpcap -lz -lrt -lm -lcrypto -lpthread -ldl -lcurses  -L/usr/local/lib -pthread   -lwx_gtk2u_xrc-3.0 -lwx_gtk2u_html-3.0 -lwx_gtk2u_qa-3.0 -lwx_gtk2u_adv-3.0 -lwx_gtk2u_core-3.0 -lwx_baseu_xml-3.0 -lwx_baseu_net-3.0 -lwx_baseu-3.0 
STRIP = strip
INSTALL_PROGRAM = ${INSTALL}
INSTALL_DATA = ${INSTALL} -m 644
INSTALL_DIR = /usr/bin/install -c -d
srcdir = .
top_srcdir = .
top_builddir =
bindir = ${exec_prefix}/bin
libdir = ${exec_prefix}/lib
datadir = ${prefix}/share
includedir = ${prefix}/include
DLLPREFIX = lib

### Variables: ###

DESTDIR = 
WX_RELEASE = 2.8
LIBDIRNAME = $(top_wxbuilddir)lib

VSCPWORKS_OBJECTS =  \
	vscpworks.o dlgvscpmsg.o frmmain.o frmvscpsession.o dlgnewvscpsession.o dlgconfiguration.o dlgvscpinterfacesettings.o dlgvscpfilter.o vscpbootloader.o dlgvscptrmit.o frmdeviceconfig.o frmscanfordevices.o dlgselectdaemoninterface.o frmmdfeditor.o dlgabout.o ../common/canalsuperwrapper.o ../../common/crc.o ../../common/crc8.o ../../common/net_skeleton.o ../common/dllwrapper.o ../common/guid.o threadvscpwork.o ../common/vscphelper.o ../common/vscpeventhelper.o ../common/vscpremotetcpif.o ../common/mdf.o ../common/register.o vscptxobj.o gridtable.o dlgeditlevel1dmrow.o dlgsetfiltermask.o dlgabstractionedit.o devicebootloaderwizard.o bootdevice.o bootdevice_pic1.o bootdevice_vscp.o merlin.o frmdaemondecisionmartrixeditor.o frmdaemonvariableeditor.o frmsimpleuserinterfacegenerator.o dlgeventfilter.o dlgclasstypeselect.o $(RESOURCEOBJECT)

### Targets: ###

all: vscpworks

install: all
	$(INSTALL_PROGRAM) vscpworks $(bindir)

uninstall: 

install-strip: install

clean: 
	rm -rf ./.deps ./.pch
	rm -f ./*.o
	rm -f ../common/*.o
	rm -f ../../common/*.o
	rm -f vscpworks

distclean: clean
	rm -f config.cache config.log config.status bk-deps bk-make-pch shared-ld-sh Makefile

vscpworks: $(VSCPWORKS_OBJECTS)
	$(CXX) -o $@ $(VSCPWORKS_OBJECTS) $(LDFLAGS) -L$(LIBDIRNAME) $(WXLIBS) $(EXTRALIBS)
	

# Include dependency info, if present:
-include .deps/*.d

.PHONY: all install uninstall clean distclean data

waaslandwolf, 2015-01-26:
Ake,

Did as described but it doesn't build. [Log is here](http://www.steeman.be/fileshare/Makefile-test.txt).

grodansparadis, 2015-01-26:
What are all the strange control characters in the file?

waaslandwolf skrev den 2015-01-26 20:53:

>
> Ake,
>
> Did as described but it doesn't build. Log is here 
> <http://www.steeman.be/fileshare/Makefile-test.txt>.

waaslandwolf, 2015-01-26:
Just artifacts because I used the 'script' command. Each time I use tab completion it puts in these chars.

grodansparadis, 2015-01-26:
Looks like your wxwidgets libs are installed in another place then I 
have them.

But you can do the edit yourself after "./configure". Edit the Makefile 
in vscpworks and

1.) Take away -DUSE_FOSSA from the line that starts with cppflags
2.) replace fossa.o with net_skeleton.o in the line that starts with 
VSCP_WORKS_OBJECTS

Then make

/Ake

waaslandwolf skrev den 2015-01-26 20:53:

>
> Ake,
>
> Did as described but it doesn't build. Log is here 
> <http://www.steeman.be/fileshare/Makefile-test.txt>.

grodansparadis, 2015-01-26:
OK hard to read. I sent a new mail with some suggestions to test.

waaslandwolf skrev den 2015-01-26 21:04:

>
> Just artifacts because I used the 'script' command. Each time I use 
> tab completion it puts in these chars.

waaslandwolf, 2015-01-26:
Ok, tried that. No cigar. [Here's the log](http://www.steeman.be/fileshare/Makefile-test2.txt).

grodansparadis, 2015-01-26:
That indicates that the makefile is messed up in some way. Some strange 
character left in it or something. Probably on the VSCPWORKS_OBJECTS 
row. Very sensitive makefiles. You could try to mode the long line up a 
row and take away the \ and see if it helps.  Or send it over here so I 
can look at it.

waaslandwolf skrev den 2015-01-26 21:10:

>
> Ok, tried that. No cigar. Here's the log 
> <http://www.steeman.be/fileshare/Makefile-test2.txt>.

waaslandwolf, 2015-01-26:
Ok, I just sent the edited Makefile to your private address.

waaslandwolf, 2015-01-26:
Oh, as a side comment: when I first got the core dump after doing a new git pull, I reverted back to the previous git version (git reset --hard HEAD...) and recompiled, got the same core dump situation. Maybe that's an interesting piece of info.

grodansparadis, 2015-01-26:
But did't you remove the repo and fetched it again. If not try that.

I can't see anything strange in the Makefile. Bloody hard to debug as 
they are so sensitive.

waaslandwolf, 2015-01-26:
Yes I did. Did 
rm -rf vscp_software
and then
git clone 

Will try again. Do you want me to edit the Makefile in in vscpworks and 
1.) Take away -DUSE_FOSSA from the line that starts with cppflags 
2.) replace fossa.o with net_skeleton.o in the line that starts with VSCP_WORKS_OBJECTS
before I attempt to make it?

grodansparadis, 2015-01-26:
OK

Try the attached instead. I have used my net_skeleton modified file 
which builds and put in your directories.

# =========================================================================
#                      Autogenerated from Makefile.in         
# =========================================================================

prefix = /usr/local
exec_prefix = ${prefix}
INSTALL = /usr/bin/install -c
top_srcdir = /root/vscp/vscp_software
top_wxsrcdir = @top_wxsrcdir@
top_builddir = ../../..
top_wxbuilddir = @top_wxbuilddir@
CXX = g++
CXXFLAGS = -g -O2 
CPPFLAGS =  -I/usr/lib/i386-linux-gnu/wx/include/gtk2-unicode-3.0 -I/usr/include/wx-3.0 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -DNS_ENABLE_SSL -DwxUSE_GUI=1 -I../common -I../../common
LDFLAGS = 
EXTRALIBS =  -lmosquittopp -lssl -lpcap -lz -lrt -lm -lcrypto -lpthread -ldl -lcurses  -L/usr/lib/i386-linux-gnu -pthread   -lwx_gtk2u_xrc-3.0 -lwx_gtk2u_html-3.0 -lwx_gtk2u_qa-3.0 -lwx_gtk2u_adv-3.0 -lwx_gtk2u_core-3.0 -lwx_baseu_xml-3.0 -lwx_baseu_net-3.0 -lwx_baseu-3.0 
STRIP = strip
INSTALL_PROGRAM = ${INSTALL}
INSTALL_DATA = ${INSTALL} -m 644
INSTALL_DIR = /usr/bin/install -c -d
srcdir = .
top_srcdir = .
top_builddir =
bindir = ${exec_prefix}/bin
libdir = ${exec_prefix}/lib
datadir = ${prefix}/share
includedir = ${prefix}/include
DLLPREFIX = lib

### Variables: ###

DESTDIR = 
WX_RELEASE = 2.8
LIBDIRNAME = $(top_wxbuilddir)lib

VSCPWORKS_OBJECTS =  \
	vscpworks.o dlgvscpmsg.o frmmain.o frmvscpsession.o dlgnewvscpsession.o dlgconfiguration.o dlgvscpinterfacesettings.o dlgvscpfilter.o vscpbootloader.o dlgvscptrmit.o frmdeviceconfig.o frmscanfordevices.o dlgselectdaemoninterface.o frmmdfeditor.o dlgabout.o ../common/canalsuperwrapper.o ../../common/crc.o ../../common/crc8.o ../../common/net_skeleton.o ../common/dllwrapper.o ../common/guid.o threadvscpwork.o ../common/vscphelper.o ../common/vscpeventhelper.o ../common/vscpremotetcpif.o ../common/mdf.o ../common/register.o vscptxobj.o gridtable.o dlgeditlevel1dmrow.o dlgsetfiltermask.o dlgabstractionedit.o devicebootloaderwizard.o bootdevice.o bootdevice_pic1.o bootdevice_vscp.o merlin.o frmdaemondecisionmartrixeditor.o frmdaemonvariableeditor.o frmsimpleuserinterfacegenerator.o dlgeventfilter.o dlgclasstypeselect.o $(RESOURCEOBJECT)

### Targets: ###

all: vscpworks

install: all
	$(INSTALL_PROGRAM) vscpworks $(bindir)

uninstall: 

install-strip: install

clean: 
	rm -rf ./.deps ./.pch
	rm -f ./*.o
	rm -f ../common/*.o
	rm -f ../../common/*.o
	rm -f vscpworks

distclean: clean
	rm -f config.cache config.log config.status bk-deps bk-make-pch shared-ld-sh Makefile

vscpworks: $(VSCPWORKS_OBJECTS)
	$(CXX) -o $@ $(VSCPWORKS_OBJECTS) $(LDFLAGS) -L$(LIBDIRNAME) $(WXLIBS) $(EXTRALIBS)
	

# Include dependency info, if present:
-include .deps/*.d

.PHONY: all install uninstall clean distclean data

grodansparadis, 2015-01-26:
If the latest makefile did not work you can check out

git reset --hard 452d2b5f0b022c8b1980d9d600a508a5100e6af2

and ./configure/Make etc.

/Ake

waaslandwolf skrev den 2015-01-26 21:44:

> Yes I did. Did
> rm -rf vscp_software
> and then
> git clone
>
> Will try again. Do you want me to edit the Makefile in in vscpworks and
> 1.) Take away -DUSE_FOSSA from the line that starts with cppflags
> 2.) [...]

waaslandwolf, 2015-01-27:
Ake,

I tried [the latest Makefile](http://www.steeman.be/fileshare/Makefile-Ake2.txt), no success. Tried git reset, no success. [The logfile is here](http://www.steeman.be/fileshare/installog3.txt).

I am prepared to re-image my machine so that we have a similar configuration running. If that would make sense then please provide some pointers on what packages to install on top of the stock Ubuntu. I find that the installation instructions on your website can be sometimes unclear (for novice Linux users that is).

Thanks,
David.

grodansparadis, 2015-01-27:
Hi,

./configure creates the Makefile so if you copy it  before issuing 
"./configure" it becomes the stock one.  "make clean;./configure", copy 
in the makefile,  build vscp works.

Did your try the older version I sent mail about.

It would be very good if you could work with me to find out what is the 
problem. If you update and get things to work someone els will come 
around and have the same problem later. Understand of course if you want 
to move on.

For the installation please suggest on places that should be make 
clearer. As of now I am unable to think of where it needs to be. I'm bad 
gods.

/Ake

waaslandwolf skrev den 2015-01-27 09:08:

> Ake,
>
> I tried the latest Makefile
> <http://www.steeman.be/fileshare/Makefile-Ake2.txt>, no success. Tried
> git reset, no success. The logfile is here [...]

waaslandwolf, 2015-01-27:
>./configure creates the Makefile so if you copy it before issuing "./configure" it becomes the stock one. 

I understand now. Sorry, Linux is not my native language so you'll have to bear with me and explain things here and there.

I cloned a fresh copy from git, then copied your last Makefile into ./src/vscp/vscpworks, then did make clean; make as requested. [Here's the log](http://www.steeman.be/fileshare/installog4.txt). It cops out immediately with this message:
root@ubuntu-john:~/vscp/vscp_software/src/vscp/vscpworks# make clean; make
Makefile:45: *** missing separator.  Stop.
Makefile:45: *** missing separator.  Stop.

>"make clean;./configure", copy in the makefile, build vscp works. 

I tried again in this order, files are here:
[install log](http://www.steeman.be/fileshare/installog5.txt)
[your modified Makefile](http://www.steeman.be/fileshare/Makefile)
[original Makefile (after ./configure](http://www.steeman.be/fileshare/Makefile-orig)

>Did your try the older version I sent mail about. 

Yes, I did 
git reset --hard 452d2b5f0b022c8b1980d9d600a508a5100e6af2
then re-built 
[install log](http://www.steeman.be/fileshare/installog6.txt)

>It would be very good if you could work with me to find out what is the problem. If you update and get things to work someone els will come around and have the same problem later. 

I understand and as mentioned before, I'm committed to improve VSCP together with you and the rest of the group. So let's continue this troubleshooting. 

Sometimes it is frustrating though because I'm making good progress with my node's hardware and firmware and feel blocked by the pagination issue and now this; of course that's what happens when you help me resolve issues in my own code, then I make too much progress and other stuff becomes the bottleneck :-)

>Understand of course if you want to move on. 

I see the value in finding the root cause and fixing it once and for all, so I'm happy to give it a few days and keep trying.

>For the installation please suggest on places that should be make clearer. As of now I am unable to think of where it needs to be. I'm bad gods.

I'll send you a separate mail related to this.

I could give you remote access to my Linux machine through VNC if you like. That would allow you to troubleshoot this a bit quicker I guess. I don't always know 100% what I am doing on Linux. Would that be a good idea? As mentioned, this is a VSCP-only machine so I don't mind if it gets trashed :-)

David.

grodansparadis, 2015-01-27:
It may be better to just revert backwards then.

Can you remember about when in time things did work for you?  Did the 
1.0.0 release code work for you for example? Now I mean without a core dump.

waaslandwolf skrev den 2015-01-27 13:19:

>
>     ./configure creates the Makefile so if you copy it before issuing
>     "./configure" it becomes the stock one.
>
> I understand now. Sorry, Linux is not my native language so you'll 
> have to bear with me and explain things here and there.
>
> I cloned a fresh copy from git, then copied your last Makefile into 
> ./src/vscp/vscpworks, then did make clean; make as requested. Here's 
> the log <http://www.steeman.be/fileshare/installog4.txt>. It cops out 
> immediately with this message:
> root@ubuntu-john:~/vscp/vscp_software/src/vscp/vscpworks# make clean; make
> Makefile:45: *** missing separator. Stop.
> Makefile:45: *** missing separator. Stop.
>
>     "make clean;./configure", copy in the makefile, build vscp works.
>
> I tried again in this order, files are here:
> install log <http://www.steeman.be/fileshare/installog5.txt>
> your modified Makefile <http://www.steeman.be/fileshare/Makefile>
> original Makefile (after ./configure 
> <http://www.steeman.be/fileshare/Makefile-orig>
>
>     Did your try the older version I sent mail about.
>
> Yes, I did
> git reset --hard 452d2b5 
> then re-built
> install log <http://www.steeman.be/fileshare/installog6.txt>
>
>     It would be very good if you could work with me to find out what
>     is the problem. If you update and get things to work someone els
>     will come around and have the same problem later.
>
> I understand and as mentioned before, I'm committed to improve VSCP 
> together with you and the rest of the group. So let's continue this 
> troubleshooting.
>
> Sometimes it is frustrating though because I'm making good progress 
> with my node's hardware and firmware and feel blocked by the 
> pagination issue and now this; of course that's what happens when you 
> help me resolve issues in my own code, then I make too much progress 
> and other stuff becomes the bottleneck :-)
>
>     Understand of course if you want to move on.
>
> I see the value in finding the root cause and fixing it once and for 
> all, so I'm happy to give it a few days and keep trying.
>
>     For the installation please suggest on places that should be make
>     clearer. As of now I am unable to think of where it needs to be.
>     I'm bad gods.
>
> I'll send you a separate mail related to this.
>
> I could give you remote access to my Linux machine through VNC if you 
> like. That would allow you to troubleshoot this a bit quicker I guess. 
> I don't always know 100% what I am doing on Linux. Would that be a 
> good idea? As mentioned, this is a VSCP-only machine so I don't mind 
> if it gets trashed :-)
>
> David.

grodansparadis, 2015-01-27:
What ubuntu version did you run again?

waaslandwolf skrev den 2015-01-27 13:19:

> ./configure creates the Makefile so if you copy it before issuing
>     "./configure" it becomes the stock one.
>
> I understand now. Sorry, Linux is not my native language so you'll
> have to bear with [...]

waaslandwolf, 2015-01-27:
Ake,

I was up to date with the latest head code and it always worked, until 
the last git pull a few days ago. v1.0.0.51. I should have saved a git 
reflog list but didn't :-(

My Ubuntu is the latest long term support version, 14.04 LTS.

Dvid.

---
This email has been checked for viruses by Avast antivirus software.
http://www.avast.com

waaslandwolf, 2015-01-27:
Ake,

Well we're not giving up yet, are we? :-)

The release code 1.0.0 worked fine, I always kept up to date with head 
code on git as well, only the last time I pulled the repository it 
crapped out after building. So the version from a couple of days ago 
messed things up for me.

I'm happy to continue investigating the way we are now, or I could give 
you remote access. That would IMHO speed up troubleshooting tremendously 
since I feel like the un-educated man in the middle at the moment :-)

Let me know what you want to do, I'm happy either way :-)

David.

---
This email has been checked for viruses by Avast antivirus software.
http://www.avast.com

grodansparadis, 2015-01-27:
I will set up a machine with 14.04 LTS here today and see if I can fins 
something.

waaslandwolf skrev den 2015-01-27 14:49:

> Ake,
>
> I was up to date with the latest head code and it always worked, until
> the last git pull a few days ago. v1.0.0.51. I should have saved a git
> reflog list but didn't :-(
>
> My Ubuntu is the [...]

waaslandwolf, 2015-01-27:
ok thanks!

---
This email has been checked for viruses by Avast antivirus software.
http://www.avast.com

grodansparadis, 2015-01-27:
I have some other things to do now but I will come back later tonight. I 
have some 32-bit machines also but non of them with graphical 
interface.   If You want you can try to step backwards through the 
commits 
(https://github.com/grodansparadis/vscp_software/commits/master?page=4). 
The release is at Nov 14, 2014 so it should be somwhere in between that 
and current.

waaslandwolf skrev den 2015-01-27 14:52:

> Ake,
>
> Well we're not giving up yet, are we? :-)
>
> The release code 1.0.0 worked fine, I always kept up to date with head
> code on git as well, only the last time I pulled the repository it
> crapped out after building. So the version from a couple of days ago
> messed things up for me.
>
> I'm happy to continue investigating the way we are now, or I could give
> you remote access. That would IMHO speed up troubleshooting tremendously
> since I feel like the un-educated man in the middle at the moment :-)
>
> Let me know what you want to do, I'm happy either way :-)
>
> David.
>
> > It may be better to just revert backwards then.
> >
> > Can you remember about when in time things did work for you? Did the
> > 1.0.0 release code work for you for example? Now I mean without a core
> > dump.
> >
> > waaslandwolf skrev den 2015-01-27 13:19:
> > >
> > > ./configure creates the Makefile so if you copy it before issuing
> > > "./configure" it becomes the stock one.
> > >
> > > I understand now. Sorry, Linux is not my native language so you'll
> > > have to bear with me and explain things here and there.
> > >
> > > I cloned a fresh copy from git, then copied your last Makefile into
> > > ./src/vscp/vscpworks, then did make clean; make as requested. Here's
> > > the log <http://www.steeman.be/fileshare/installog4.txt>. It cops out
> > > immediately with this message:
> > > root@ubuntu-john:~/vscp/vscp_software/src/vscp/vscpworks# make
> > clean; make
> > > Makefile:45: *** missing separator. Stop.
> > > Makefile:45: *** missing separator. Stop.
> > >
> > > "make clean;./configure", copy in the makefile, build vscp works.
> > >
> > > I tried again in this order, files are here:
> > > install log <http://www.steeman.be/fileshare/installog5.txt>
> > > your modified Makefile <http://www.steeman.be/fileshare/Makefile>
> > > original Makefile (after ./configure
> > > <http://www.steeman.be/fileshare/Makefile-orig>
> > >
> > > Did your try the older version I sent mail about.
> > >
> > > Yes, I did
> > > git reset --hard 452d2b5
> > >
> > 
> > > then re-built
> > > install log <http://www.steeman.be/fileshare/installog6.txt>
> > >
> > > It would be very good if you could work with me to find out what
> > > is the problem. If you update and get things to work someone els
> > > will come around and have the same problem later.
> > >
> > > I understand and as mentioned before, I'm committed to improve VSCP
> > > together with you and the rest of the group. So let's continue this
> > > troubleshooting.
> > >
> > > Sometimes it is frustrating though because I'm making good progress
> > > with my node's hardware and firmware and feel blocked by the
> > > pagination issue and now this; of course that's what happens when you
> > > help me resolve issues in my own code, then I make too much progress
> > > and other stuff becomes the bottleneck :-)
> > >
> > > Understand of course if you want to move on.
> > >
> > > I see the value in finding the root cause and fixing it once and for
> > > all, so I'm happy to give it a few days and keep trying.
> > >
> > > For the installation please suggest on places that should be make
> > > clearer. As of now I am unable to think of where it needs to be.
> > > I'm bad gods.
> > >
> > > I'll send you a separate mail related to this.
> > >
> > > I could give you remote access to my Linux machine through VNC if you
> > > like. That would allow you to troubleshoot this a bit quicker I guess.
> > > I don't always know 100% what I am doing on Linux. Would that be a
> > > good idea? As mentioned, this is a VSCP-only machine so I don't mind
> > > if it gets trashed :-)
> > >
> > > David.
> > >
> > >
> > 
> > >
> >
> > Be Hungry - Stay Foolish!
> > Åke Hedman, Grodans Paradis AB / Paradise of the Frog
> > Brattbergavägen 17, 82050 LOS, SWEDEN
> > Phone: +46 (0)840011835 CellPhone: +46 (0)730-533146
> > http://www.grodansparadis.com http://www.vscp.org
> >
> >
> >
> > ---
> > Detta e-postmeddelande har sökts igenom efter virus med
> > antivirusprogram från Avast.
> > http://www.avast.com
> >
> > 
> >
>
>
>
> ---
> This email has been checked for viruses by Avast antivirus software.
> http://www.avast.com

waaslandwolf, 2015-01-27:
Ake,

I just tried with the commit from Jan 19, 2015 (53f17d93c429810a025ee251adab5a300187ca12) and I get the same issue. 

As a second test I did 
rm -rf vscp_software
git clone https://github.com/grodansparadis/vscp_software
git reset --hard 53f17d93c429810a025ee251adab5a300187ca12
mv /etc/vscp /etc/vscp-backup
rm /usr/local/bin/vscp*
cd vscp_software
./configure; make; make install

Still the same issue. 

So I guess something else is broken other than the vscp_software respository. Perhaps some config file? Perhaps the binaries in /usr/local/bin don't get updated any more? (I'm just guessing here).

grodansparadis, 2015-01-28:
Hi David,

no I haven't forgot about your problem, have been working on it since 
yesterday and have been able to reproduce it on a 14.04 LTE machine.  
Starting to debug it the problem disappeared and now everything works. 
But the question is why of course and I can't answer that yet.

I have to install the system from the ground again and test which will 
take a while. My thought is that it all depens on some package getting 
updated by the system. Could you help me to try this

sudo apt-get  update
sudo apt-get  update
sudo apt-get  install libgtk2.0-dev

and then test. This is a real long shoot but worth to test.

/Ake

waaslandwolf skrev den 2015-01-27 14:52:

> Ake,
>
> Well we're not giving up yet, are we? :-)
>
> The release code 1.0.0 worked fine, I always kept up to date with head
> code on git as well, only the last time I pulled the repository it
> crapped out after building. So the version from a couple of days ago
> messed things up for me.
>
> I'm happy to continue investigating the way we are now, or I could give
> you remote access. That would IMHO speed up troubleshooting tremendously
> since I feel like the un-educated man in the middle at the moment :-)
>
> Let me know what you want to do, I'm happy either way :-)
>
> David.
>
> > It may be better to just revert backwards then.
> >
> > Can you remember about when in time things did work for you? Did the
> > 1.0.0 release code work for you for example? Now I mean without a core
> > dump.
> >
> > waaslandwolf skrev den 2015-01-27 13:19:
> > >
> > > ./configure creates the Makefile so if you copy it before issuing
> > > "./configure" it becomes the stock one.
> > >
> > > I understand now. Sorry, Linux is not my native language so you'll
> > > have to bear with me and explain things here and there.
> > >
> > > I cloned a fresh copy from git, then copied your last Makefile into
> > > ./src/vscp/vscpworks, then did make clean; make as requested. Here's
> > > the log <http://www.steeman.be/fileshare/installog4.txt>. It cops out
> > > immediately with this message:
> > > root@ubuntu-john:~/vscp/vscp_software/src/vscp/vscpworks# make
> > clean; make
> > > Makefile:45: *** missing separator. Stop.
> > > Makefile:45: *** missing separator. Stop.
> > >
> > > "make clean;./configure", copy in the makefile, build vscp works.
> > >
> > > I tried again in this order, files are here:
> > > install log <http://www.steeman.be/fileshare/installog5.txt>
> > > your modified Makefile <http://www.steeman.be/fileshare/Makefile>
> > > original Makefile (after ./configure
> > > <http://www.steeman.be/fileshare/Makefile-orig>
> > >
> > > Did your try the older version I sent mail about.
> > >
> > > Yes, I did
> > > git reset --hard 452d2b5
> > >
> > 
> > > then re-built
> > > install log <http://www.steeman.be/fileshare/installog6.txt>
> > >
> > > It would be very good if you could work with me to find out what
> > > is the problem. If you update and get things to work someone els
> > > will come around and have the same problem later.
> > >
> > > I understand and as mentioned before, I'm committed to improve VSCP
> > > together with you and the rest of the group. So let's continue this
> > > troubleshooting.
> > >
> > > Sometimes it is frustrating though because I'm making good progress
> > > with my node's hardware and firmware and feel blocked by the
> > > pagination issue and now this; of course that's what happens when you
> > > help me resolve issues in my own code, then I make too much progress
> > > and other stuff becomes the bottleneck :-)
> > >
> > > Understand of course if you want to move on.
> > >
> > > I see the value in finding the root cause and fixing it once and for
> > > all, so I'm happy to give it a few days and keep trying.
> > >
> > > For the installation please suggest on places that should be make
> > > clearer. As of now I am unable to think of where it needs to be.
> > > I'm bad gods.
> > >
> > > I'll send you a separate mail related to this.
> > >
> > > I could give you remote access to my Linux machine through VNC if you
> > > like. That would allow you to troubleshoot this a bit quicker I guess.
> > > I don't always know 100% what I am doing on Linux. Would that be a
> > > good idea? As mentioned, this is a VSCP-only machine so I don't mind
> > > if it gets trashed :-)
> > >
> > > David.
> > >
> > >
> > 
> > >
> >
> > Be Hungry - Stay Foolish!
> > Åke Hedman, Grodans Paradis AB / Paradise of the Frog
> > Brattbergavägen 17, 82050 LOS, SWEDEN
> > Phone: +46 (0)840011835 CellPhone: +46 (0)730-533146
> > http://www.grodansparadis.com http://www.vscp.org
> >
> >
> >
> > ---
> > Detta e-postmeddelande har sökts igenom efter virus med
> > antivirusprogram från Avast.
> > http://www.avast.com
> >
> > 
> >
>
>
>
> ---
> This email has been checked for viruses by Avast antivirus software.
> http://www.avast.com

grodansparadis, 2015-01-28:
Never mind my last mail . It does not help.

BUT

A debug build works. So "./configure --enable-debug" followed by "make" 
and "make install" make things work. Something must be wring with the 
stock 32-bit wx libs. I will investigate tat a bit further. But this 
works as a workaround.

/Ake

waaslandwolf skrev den 2015-01-27 19:52:

>
> Ake,
>
> I just tried with the commit from Jan 19, 2015 (53f17d9 
> <https://github.com/grodansparadis/vscp_software/commit/53f17d93c429810a025ee251adab5a300187ca12>) 
> and I get the same issue.
>
> As a second test I did
> rm -rf vscp_software
> git clone https://github.com/grodansparadis/vscp_software
> git reset --hard 53f17d9 
> mv /etc/vscp /etc/vscp-backup
> rm /usr/local/bin/vscp*
> cd vscp_software
> ./configure; make; make install
>
> Still the same issue.
>
> So I guess something else is broken other than the vscp_software 
> respository. Perhaps some config file? Perhaps the binaries in 
> /usr/local/bin don't get updated any more? (I'm just guessing here).

waaslandwolf, 2015-01-28:
Ake,

No I wasn't worried, you told me you'd have to set up a specific test image with 14.04 LTS so I figured that wouldn't happen in 10 minutes :-) Thanks for the work you're putting into this. 

>BUT...A debug build works. So "./configure --enable-debug" followed by "make" and "make install" make things work. 

I tried this but still that same core dump happens... I pulled the latest git before testing.

I believe I have been running the debug version all the time, since the messages I am getting from vscpd are the same in what is now configured as --enable-debug explicitly.

So best of luck debugging this, I bet you've been swearing over it...

David.

waaslandwolf, 2015-01-28:
In my comment above I did 
./configure --enable-debug
make
make install
in the $VSCPHOME directory

When I tried your other comment
"./configure --enable-debug"
go to "($VSCPHOME)/src/vscp/vscpworks" and issu
"make" and "make install" there. 
I get:
root@ubuntu-john:~/vscp/vscp_software/src/vscp/vscpworks# make
make: Nothing to be done for `all'.

grodansparadis, 2015-01-28:
Hi,

I have installed the machine several time and tested this here and it 
works without problem with the debug. Strange it did not work for you.

It is probably something strange in the libs as debug output also is 
sent out when building a release version which is strange.

/Ake

waaslandwolf skrev den 2015-01-28 21:16:

> Ake,
>
> No I wasn't worried, you told me you'd have to set up a specific test
> image with 14.04 LTS so I figured that wouldn't happen in 10 minutes
> :-) Thanks for the work you're putting into [...]

grodansparadis, 2015-01-28:
"make clean" first

waaslandwolf skrev den 2015-01-28 21:20:

> In my comment above I did
> ./configure --enable-debug
> make
> make install
> in the $VSCPHOME directory
>
> When I tried your other comment
> "./configure --enable-debug"
> go to "($VSCPHOME)/src/vscp/vscpworks" [...]

waaslandwolf, 2015-01-28:
Ah, that's better :-) It makes now, and works connects to the deamon. I 
cannot see events on the bus yet, but I need to look into it further to 
see if I am not doing something stupid. I'll provide more feedback in my 
next testing session, gotta close shop for tonight.

---
This email has been checked for viruses by Avast antivirus software.
http://www.avast.com

grodansparadis, 2015-01-28:
Two :) in a night is good. For the time they last at least. There is 
always another bug.

waaslandwolf skrev den 2015-01-28 21:48:

> Ah, that's better :-) It makes now, and works connects to the deamon. I
> cannot see events on the bus yet, but I need to look into it further to
> see if I am not doing something stupid. I'll provide more feedback in my
> next testing session, gotta close shop for tonight.
>
>
> > "make clean" first
> >
> > waaslandwolf skrev den 2015-01-28 21:20:
> > >
> > > In my comment above I did
> > > ./configure --enable-debug
> > > make
> > > make install
> > > in the $VSCPHOME directory
> > >
> > > When I tried your other comment
> > > "./configure --enable-debug"
> > > go to "($VSCPHOME)/src/vscp/vscpworks" and issu
> > > "make" and "make install" there.
> > > I get:
> > > root@ubuntu-john:~/vscp/vscp_software/src/vscp/vscpworks# make
> > > make: Nothing to be done for `all'.
> > >
> > >
> > 
> > >
> >
> > Be Hungry - Stay Foolish!
> > Åke Hedman, Grodans Paradis AB / Paradise of the Frog
> > Brattbergavägen 17, 82050 LOS, SWEDEN
> > Phone: +46 (0)840011835 CellPhone: +46 (0)730-533146
> > http://www.grodansparadis.com http://www.vscp.org
> >
> >
> >
> > ---
> > Detta e-postmeddelande har sökts igenom efter virus med
> > antivirusprogram från Avast.
> > http://www.avast.com
> >
> > 
> >
>
>
>
> ---
> This email has been checked for viruses by Avast antivirus software.
> http://www.avast.com

waaslandwolf, 2015-01-29:
Ake,

Bad news :-( Works now can connect to the deamon, but doesn't display 
any events on the bus in the monitor window. Reading registers also 
fails. If I hook up my USB2CAN interface to my windows machine I see 
events on the bus.

David.

---
This email has been checked for viruses by Avast antivirus software.
http://www.avast.com

waaslandwolf, 2015-01-29:
By the way, I can connect to the deamon running on Linux with VSCPworks 
on my Windows machine, but don't see events there either.

---
This email has been checked for viruses by Avast antivirus software.
http://www.avast.com

grodansparadis, 2015-01-29:
can you se any event in the TCP/IP IF?

Skickat från min iPhone

waaslandwolf, 2015-01-29:
Ake,

Assuming I would see those in my terminal window when running a debug 
version of vscpd; no.

What I do:
sudo ip link set can0 up type can bitrate 250000 sample-point 0.875
sudo /usr/local/bin/vscpd &
/usr/local/bin/vscpworks & (other terminal window)

In VSCPworks:
VSCP menu - client window option
TCP/IP: Localhost, Edit... option
Server URL: localhost:9598
Get Interfaces
- Internal Daemon VSCP worker. Started
- VSCP_Level_II_SocketCAN_Driver_1. started
- VSCP_Level_II_Logger_Driver_1. Started.
- Internal Daemon DM client. Started.
- Remote TCP/IP Client. Started.

I select VSCP_Level_II_SocketCAN_Driver_1
Test connection -> Successful connect to server. I don't see any debug 
messages in the daemon's terminal window.

I open the client window. Status shows connected. TCP/IP TX and RX: 
connection open.

Generating events on the bus doesn't show any activity in the client 
window nor in the daemon's terminal window.

David.

---
This email has been checked for viruses by Avast antivirus software.
http://www.avast.com

grodansparadis, 2015-01-29:
First make sure that your daemon actually is functioning on the 
interface(s) you need.

telnet to host

user: admin
pass: secret
(or other credentials if you have changed)
issue "rcvloop"

and then you should see events sent by your driver.    If you don't see 
events check that the correct driver is installed. Check settings for 
it. Restart.

/Ake

waaslandwolf skrev den 2015-01-29 18:57:

> Ake,
>
> Assuming I would see those in my terminal window when running a debug
> version of vscpd; no.
>
> What I do:
> sudo ip link set can0 up type can bitrate 250000 sample-point 0.875
> sudo /usr/local/bin/vscpd &
> /usr/local/bin/vscpworks & (other terminal window)
>
> In VSCPworks:
> VSCP menu - client window option
> TCP/IP: Localhost, Edit... option
> Server URL: localhost:9598
> Get Interfaces
> - Internal Daemon VSCP worker. Started
> - VSCP_Level_II_SocketCAN_Driver_1. started
> - VSCP_Level_II_Logger_Driver_1. Started.
> - Internal Daemon DM client. Started.
> - Remote TCP/IP Client. Started.
>
> I select VSCP_Level_II_SocketCAN_Driver_1
> Test connection -> Successful connect to server. I don't see any debug
> messages in the daemon's terminal window.
>
> I open the client window. Status shows connected. TCP/IP TX and RX:
> connection open.
>
> Generating events on the bus doesn't show any activity in the client
> window nor in the daemon's terminal window.
>
> David.
>
> > can you se any event in the TCP/IP IF?
> >
> > Skickat från min iPhone
> >
> > >
> > > By the way, I can connect to the deamon running on Linux with 
> VSCPworks
> > > on my Windows machine, but don't see events there either.
> > >
> > > > Ake,
> > > >
> > > > Bad news :-( Works now can connect to the deamon, but doesn't 
> display
> > > > any events on the bus in the monitor window. Reading registers also
> > > > fails. If I hook up my USB2CAN interface to my windows machine I see
> > > > events on the bus.
> > > >
> > > > David.
> > > >
> > > >> Two :) in a night is good. For the time they last at least. 
> There is
> > > >> always another bug.
> > > >>
> > > >> waaslandwolf skrev den 2015-01-28 21:48:
> > > >> > Ah, that's better :-) It makes now, and works connects to the
> > deamon. I
> > > >> > cannot see events on the bus yet, but I need to look into it
> > further to
> > > >> > see if I am not doing something stupid. I'll provide more 
> feedback
> > > >> in my
> > > >> > next testing session, gotta close shop for tonight.
> > > >> >
> > > >> >
> > > >> > > "make clean" first
> > > >> > >
> > > >> > > waaslandwolf skrev den 2015-01-28 21:20:
> > > >> > > >
> > > >> > > > In my comment above I did
> > > >> > > > ./configure --enable-debug
> > > >> > > > make
> > > >> > > > make install
> > > >> > > > in the $VSCPHOME directory
> > > >> > > >
> > > >> > > > When I tried your other comment
> > > >> > > > "./configure --enable-debug"
> > > >> > > > go to "($VSCPHOME)/src/vscp/vscpworks" and issu
> > > >> > > > "make" and "make install" there.
> > > >> > > > I get:
> > > >> > > > root@ubuntu-john:~/vscp/vscp_software/src/vscp/vscpworks# 
> make
> > > >> > > > make: Nothing to be done for `all'.
> > > >> > > >
> > > >> > > >
> > > >> > >
> > > >> >
> > > >>
> > 
> > > >> > > >
> > > >> > >
> > > >> > > Be Hungry - Stay Foolish!
> > > >> > > Åke Hedman, Grodans Paradis AB / Paradise of the Frog
> > > >> > > Brattbergavägen 17, 82050 LOS, SWEDEN
> > > >> > > Phone: +46 (0)840011835 CellPhone: +46 (0)730-533146
> > > >> > > http://www.grodansparadis.com http://www.vscp.org
> > > >> > >
> > > >> > >
> > > >> > >
> > > >> > > ---
> > > >> > > Detta e-postmeddelande har sökts igenom efter virus med
> > > >> > > antivirusprogram från Avast.
> > > >> > > http://www.avast.com
> > > >> > >
> > > >> > >
> > > >> >
> > > >>
> > 
> > > >> > >
> > > >> >
> > > >> >
> > > >> >
> > > >> > ---
> > > >> > This email has been checked for viruses by Avast antivirus
> > software.
> > > >> > http://www.avast.com
> > > >> >
> > > >> >
> > > >>
> > 
> > > >> >
> > > >>
> > > >> Be Hungry - Stay Foolish!
> > > >> Åke Hedman, Grodans Paradis AB / Paradise of the Frog
> > > >> Brattbergavägen 17, 82050 LOS, SWEDEN
> > > >> Phone: +46 (0)840011835 CellPhone: +46 (0)730-533146
> > > >> http://www.grodansparadis.com http://www.vscp.org
> > > >>
> > > >>
> > > >>
> > > >> ---
> > > >> Detta e-postmeddelande har sökts igenom efter virus med
> > > >> antivirusprogram från Avast.
> > > >> http://www.avast.com
> > > >>
> > > >>
> > 
> > > >>
> > > >
> > >
> > >
> > >
> > > ---
> > > This email has been checked for viruses by Avast antivirus software.
> > > http://www.avast.com
> > >
> >
> > 
> >
>
>
>
> ---
> This email has been checked for viruses by Avast antivirus software.
> http://www.avast.com

waaslandwolf, 2015-01-29:
Ake,

rcvloop outputs "+OK" every second or so. I guess I should be seeing 
polling type of messages to and from the driver instead?

My vscpd.conf reads:
         
             <driver enable="true" >
             <name>VSCP Level II SocketCAN Driver 1</name>
<path>/usr/local/lib/vscp2drv_socketcan.so</path>
             <config>can0</config>
<guid>37:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00</guid>
         </driver>

The vscp2drv_socketcan.so is in /usr/local/lib/ as specified, and I 
haven't touched it in a while and before it was working. So I would 
doubt this is the issue. Where did this driver come from? Was it 
included in my Ubuntu distro or did the vscp_software repository install it?

Any clues on further troubleshooting?

Thanks,
David.

---
This email has been checked for viruses by Avast antivirus software.
http://www.avast.com

grodansparadis, 2015-01-29:
Noop that should work.

waaslandwolf skrev den 2015-01-29 20:32:

> Ake,
>
> rcvloop outputs "+OK" every second or so. I guess I should be seeing
> polling type of messages to and from the driver instead?
>
> My vscpd.conf reads:
> 
> <driver enable="true" >
> <name>VSCP Level II SocketCAN Driver 1</name>
> <path>/usr/local/lib/vscp2drv_socketcan.so</path>
> <config>can0</config>
> <guid>37:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00</guid>
> </driver>
>
> The vscp2drv_socketcan.so is in /usr/local/lib/ as specified, and I
> haven't touched it in a while and before it was working. So I would
> doubt this is the issue. Where did this driver come from? Was it
> included in my Ubuntu distro or did the vscp_software repository 
> install it?
>
> Any clues on further troubleshooting?
>
> Thanks,
> David.
>
>
>
> > First make sure that your daemon actually is functioning on the
> > interface(s) you need.
> >
> > telnet to host
> >
> > user: admin
> > pass: secret
> > (or other credentials if you have changed)
> > issue "rcvloop"
> >
> > and then you should see events sent by your driver. If you don't see
> > events check that the correct driver is installed. Check settings for
> > it. Restart.
> >
> > /Ake
> >
> > waaslandwolf skrev den 2015-01-29 18:57:
> > > Ake,
> > >
> > > Assuming I would see those in my terminal window when running a debug
> > > version of vscpd; no.
> > >
> > > What I do:
> > > sudo ip link set can0 up type can bitrate 250000 sample-point 0.875
> > > sudo /usr/local/bin/vscpd &
> > > /usr/local/bin/vscpworks & (other terminal window)
> > >
> > > In VSCPworks:
> > > VSCP menu - client window option
> > > TCP/IP: Localhost, Edit... option
> > > Server URL: localhost:9598
> > > Get Interfaces
> > > - Internal Daemon VSCP worker. Started
> > > - VSCP_Level_II_SocketCAN_Driver_1. started
> > > - VSCP_Level_II_Logger_Driver_1. Started.
> > > - Internal Daemon DM client. Started.
> > > - Remote TCP/IP Client. Started.
> > >
> > > I select VSCP_Level_II_SocketCAN_Driver_1
[... 150 lines cut ...]
> > >
> > 
> > >
> >
> > Be Hungry - Stay Foolish!
> > Åke Hedman, Grodans Paradis AB / Paradise of the Frog
> > Brattbergavägen 17, 82050 LOS, SWEDEN
> > Phone: +46 (0)840011835 CellPhone: +46 (0)730-533146
> > http://www.grodansparadis.com http://www.vscp.org
> >
> >
> >
> > ---
> > Detta e-postmeddelande har sökts igenom efter virus med
> > antivirusprogram från Avast.
> > http://www.avast.com
> >
> > 
> >
>
>
>
> ---
> This email has been checked for viruses by Avast antivirus software.
> http://www.avast.com

waaslandwolf, 2015-01-29:
? Not sure what you mean?

---
This email has been checked for viruses by Avast antivirus software.
http://www.avast.com

grodansparadis, 2015-01-29:
I mean everything  looks OK from the VSCP side.

waaslandwolf skrev den 2015-01-29 20:35:

>
> ? Not sure what you mean?
>
> > Noop that should work.
> >
> > waaslandwolf skrev den 2015-01-29 20:32:
> > > Ake,
> > >
> > > rcvloop outputs "+OK" every second or so. I guess I should be seeing
> > > polling type of messages to and from the driver instead?
> > >
> > > My vscpd.conf reads:
> > > 
> > > <driver enable="true" >
> > > <name>VSCP Level II SocketCAN Driver 1</name>
> > > <path>/usr/local/lib/vscp2drv_socketcan.so</path>
> > > <config>can0</config>
> > > <guid>37:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00</guid>
> > > </driver>
> > >
> > > The vscp2drv_socketcan.so is in /usr/local/lib/ as specified, and I
> > > haven't touched it in a while and before it was working. So I would
> > > doubt this is the issue. Where did this driver come from? Was it
> > > included in my Ubuntu distro or did the vscp_software repository
> > > install it?
> > >
> > > Any clues on further troubleshooting?
> > >
> > > Thanks,
> > > David.
> > >
> > >
> > >
> > > > First make sure that your daemon actually is functioning on the
> > > > interface(s) you need.
> > > >
> > > > telnet to host
> > > >
> > > > user: admin
> > > > pass: secret
> > > > (or other credentials if you have changed)
> > > > issue "rcvloop"
> > > >
> > > > and then you should see events sent by your driver. If you don't see
> > > > events check that the correct driver is installed. Check 
> settings for
> > > > it. Restart.
> > > >
> > > > /Ake
> > > >
> > > > waaslandwolf skrev den 2015-01-29 18:57:
> > > > > Ake,
> > > > >
> > > > > Assuming I would see those in my terminal window when running a
> > debug
> > > > > version of vscpd; no.
> > > > >
> > > > > What I do:
> > > > > sudo ip link set can0 up type can bitrate 250000 sample-point 
> 0.875
> > > > > sudo /usr/local/bin/vscpd &
> > > > > /usr/local/bin/vscpworks & (other terminal window)
> > > > >
> > > > > In VSCPworks:
> > > > > VSCP menu - client window option
> > > > > TCP/IP: Localhost, Edit... option
[... 210 lines cut ...]
> > >
> > 
> > >
> >
> > Be Hungry - Stay Foolish!
> > Åke Hedman, Grodans Paradis AB / Paradise of the Frog
> > Brattbergavägen 17, 82050 LOS, SWEDEN
> > Phone: +46 (0)840011835 CellPhone: +46 (0)730-533146
> > http://www.grodansparadis.com http://www.vscp.org
> >
> >
> >
> > ---
> > Detta e-postmeddelande har sökts igenom efter virus med
> > antivirusprogram från Avast.
> > http://www.avast.com
> >
> > 
> >
>
>
>
> ---
> This email has been checked for viruses by Avast antivirus software.
> http://www.avast.com

waaslandwolf, 2015-01-31:
Is there anything else I can do to (help) troubleshoot this?

---
This email has been checked for viruses by Avast antivirus software.
http://www.avast.com

grodansparadis, 2015-01-31:
Does candump work?
Flags=1 in driver settings?

Skickat från min iPhone

[15 more replies]

closed 2015-02-04 by grodansparadis