grodansparadis/vscp #68: VSCPworks crashes on certain chars in MDF

waaslandwolf, 2015-02-03:
Ake,

I noticed that VSCPworks crashes when attempting to read registers associated with an MDF that contains certain characters such as %, ° etc. Perhaps something Unicode-related. Perhaps you can add a test when parsing MDF files which ignores these chars? Or some other way of handling them and preventing VSCPworks from crashing?

Mind: I tested this using an old VSCPworks v 1.0.0.43 on Windows, perhaps it is already fixed in a newer version. But I can't test that at the moment so I'm logging it here as a reminder.

David.

grodansparadis, 2015-02-03:
This is expat (the xml engine) so it's out of my control.

/Ake

waaslandwolf skrev den 2015-02-03 13:54:

> Ake,
>
> I noticed that VSCPworks crashes when attempting to read registers
> associated with an MDF that contains certain characters such as %, °
> etc. Perhaps something Unicode-related. Perhaps you [...]

grodansparadis, 2015-02-03:
Can you send me a file that is a sample of this. Can't get it to crash here.

waaslandwolf skrev den 2015-02-03 13:54:

> Ake,
>
> I noticed that VSCPworks crashes when attempting to read registers
> associated with an MDF that contains certain characters such as %, °
> etc. Perhaps something Unicode-related. Perhaps you [...]

waaslandwolf, 2015-02-03:
Here's an example:

         <reg page="0" offset="122">
             <name lang="en">PWM test register (TEST_PWM_OUT1)</name>
             <description lang="en">
                 This register holds a PWM value in tens of percents (0 
- 10 for 0 to 100%).
             </description>
             <access>rw</access>
         </reg>

This crashes VSCPworks on the % sign.

Do mind that I am testing using an old Windows version of VSCPworks so 
this issue might already have been fixed. I'm just  reporting any issue 
I find, trusting you'd be the judge of what is important to fix :-)

David.

---
This email has been checked for viruses by Avast antivirus software.
http://www.avast.com

grodansparadis, 2015-02-03:
Confirm this to be a bug!

grodansparadis, 2015-02-03:
Fixed

closed 2015-02-03 by grodansparadis

grodansparadis, 2015-02-03:
David,

can you test one thing for me.

On the system you built where vscpworks crashes

go to  /src/vscp/vscpworks
do "make clean"
do "make"
When built "./vscpworks"
Connect to server. Does it still crash?

/Ake

waaslandwolf skrev den 2015-02-03 16:36:

> Here's an example:
>
> 
> <reg page="0" offset="122">
> <name lang="en">PWM test register (TEST_PWM_OUT1)</name>
> <description lang="en">
> This register holds a PWM value in tens of percents (0
> - 10 for 0 to 100%).
> </description>
> <access>rw</access>
> </reg>
>
>
> This crashes VSCPworks on the % sign.
>
> Do mind that I am testing using an old Windows version of VSCPworks so
> this issue might already have been fixed. I'm just reporting any issue
> I find, trusting you'd be the judge of what is important to fix :-)
>
> David.
>
> > Can you send me a file that is a sample of this. Can't get it to crash
> > here.
> >
> > waaslandwolf skrev den 2015-02-03 13:54:
> > >
> > > Ake,
> > >
> > > I noticed that VSCPworks crashes when attempting to read registers
> > > associated with an MDF that contains certain characters such as %, °
> > > etc. Perhaps something Unicode-related. Perhaps you can add a test
> > > when parsing MDF files which ignores these chars? Or some other way of
> > > handling them and preventing VSCPworks from crashing?
> > >
> > > Mind: I tested this using an old VSCPworks v 1.0.0.43 on Windows,
> > > perhaps it is already fixed in a newer version. But I can't test that
> > > at the moment so I'm logging it here as a reminder.
> > >
> > > David.
> > >
> > >
> >
> > Be Hungry - Stay Foolish!
> > Åke Hedman, Grodans Paradis AB / Paradise of the Frog
> > Brattbergavägen 17, 82050 LOS, SWEDEN
> > Phone: +46 (0)840011835 CellPhone: +46 (0)730-533146
> > http://www.grodansparadis.com http://www.vscp.org
> >
> >
> >
> > ---
> > Detta e-postmeddelande har sökts igenom efter virus med
> > antivirusprogram från Avast.
> > http://www.avast.com
> >
> > 
> >
>
>
>
> ---
> This email has been checked for viruses by Avast antivirus software.
> http://www.avast.com

waaslandwolf, 2015-02-04:
Ake,

Just tried that, log is attached. This works, I can do "get interfaces", 
"test connection" and can see events in the monitor window. Doesn't core 
dump any more.

David.

---
This email has been checked for viruses by Avast antivirus software.
http://www.avast.com

Script started on Mit 04 Feb 2015 06:53:42 CET
]0;root@ubuntu: ~root@ubuntu:~# cd /src/vscp/vscpworks./src/vscp/vscpworks[C/src/vscp/vscpworksv/src/vscp/vscpworkss/src/vscp/vscpworksc/src/vscp/vscpworksp/src/vscp/vscpworks_/src/vscp/vscpworkss/src/vscp/vscpworkso/src/vscp/vscpworksf/src/vscp/vscpworkst/src/vscp/vscpworksw/src/vscp/vscpworksa/src/vscp/vscpworksr/src/vscp/vscpworkse/src/vscp/vscpworks
]0;root@ubuntu: ~/vscp_software/src/vscp/vscpworksroot@ubuntu:~/vscp_software/src/vscp/vscpworks# make clean
rm -rf ./.deps ./.pch
rm -f ./*.o
rm -f ../common/*.o
rm -f ../../common/*.o
rm -f vscpworks
]0;root@ubuntu: ~/vscp_software/src/vscp/vscpworksroot@ubuntu:~/vscp_software/src/vscp/vscpworks# 
]0;root@ubuntu: ~/vscp_software/src/vscp/vscpworksroot@ubuntu:~/vscp_software/src/vscp/vscpworks# 
]0;root@ubuntu: ~/vscp_software/src/vscp/vscpworksroot@ubuntu:~/vscp_software/src/vscp/vscpworks# 
]0;root@ubuntu: ~/vscp_software/src/vscp/vscpworksroot@ubuntu:~/vscp_software/src/vscp/vscpworks# maker[K
g++ -g -O2  -I/usr/lib/i386-linux-gnu/wx/include/gtk2-unicode-3.0 -I/usr/include/wx-3.0 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -DwxUSE_GUI=1 -DUSE_FOSSA -I../common -I../../common  -c -o vscpworks.o vscpworks.cpp
g++ -g -O2  -I/usr/lib/i386-linux-gnu/wx/include/gtk2-unicode-3.0 -I/usr/include/wx-3.0 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -DwxUSE_GUI=1 -DUSE_FOSSA -I../common -I../../common  -c -o dlgvscpmsg.o dlgvscpmsg.cpp
g++ -g -O2  -I/usr/lib/i386-linux-gnu/wx/include/gtk2-unicode-3.0 -I/usr/include/wx-3.0 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -DwxUSE_GUI=1 -DUSE_FOSSA -I../common -I../../common  -c -o frmmain.o frmmain.cpp
g++ -g -O2  -I/usr/lib/i386-linux-gnu/wx/include/gtk2-unicode-3.0 -I/usr/include/wx-3.0 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -DwxUSE_GUI=1 -DUSE_FOSSA -I../common -I../../common  -c -o frmvscpsession.o frmvscpsession.cpp
g++ -g -O2  -I/usr/lib/i386-linux-gnu/wx/include/gtk2-unicode-3.0 -I/usr/include/wx-3.0 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -DwxUSE_GUI=1 -DUSE_FOSSA -I../common -I../../common  -c -o dlgnewvscpsession.o dlgnewvscpsession.cpp
g++ -g -O2  -I/usr/lib/i386-linux-gnu/wx/include/gtk2-unicode-3.0 -I/usr/include/wx-3.0 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -DwxUSE_GUI=1 -DUSE_FOSSA -I../common -I../../common  -c -o dlgconfiguration.o dlgconfiguration.cpp
g++ -g -O2  -I/usr/lib/i386-linux-gnu/wx/include/gtk2-unicode-3.0 -I/usr/include/wx-3.0 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -DwxUSE_GUI=1 -DUSE_FOSSA -I../common -I../../common  -c -o dlgvscpinterfacesettings.o dlgvscpinterfacesettings.cpp
g++ -g -O2  -I/usr/lib/i386-linux-gnu/wx/include/gtk2-unicode-3.0 -I/usr/include/wx-3.0 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -DwxUSE_GUI=1 -DUSE_FOSSA -I../common -I../../common  -c -o dlgvscpfilter.o dlgvscpfilter.cpp
g++ -g -O2  -I/usr/lib/i386-linux-gnu/wx/include/gtk2-unicode-3.0 -I/usr/include/wx-3.0 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -DwxUSE_GUI=1 -DUSE_FOSSA -I../common -I../../common  -c -o vscpbootloader.o vscpbootloader.cpp
g++ -g -O2  -I/usr/lib/i386-linux-gnu/wx/include/gtk2-unicode-3.0 -I/usr/include/wx-3.0 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -DwxUSE_GUI=1 -DUSE_FOSSA -I../common -I../../common  -c -o dlgvscptrmit.o dlgvscptrmit.cpp
g++ -g -O2  -I/usr/lib/i386-linux-gnu/wx/include/gtk2-unicode-3.0 -I/usr/include/wx-3.0 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -DwxUSE_GUI=1 -DUSE_FOSSA -I../common -I../../common  -c -o frmdeviceconfig.o frmdeviceconfig.cpp
frmdeviceconfig.cpp: In member function ‘void frmDeviceConfig::OnLeftDClick(wxGridEvent&)’:
frmdeviceconfig.cpp:4991:67: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
             dlg.m_comboAction->SetClientData(idx, (void *)action->m_nCode); // Yes - pointer conversion
                                                                   ^
g++ -g -O2  -I/usr/lib/i386-linux-gnu/wx/include/gtk2-unicode-3.0 -I/usr/include/wx-3.0 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -DwxUSE_GUI=1 -DUSE_FOSSA -I../common -I../../common  -c -o frmscanfordevices.o frmscanfordevices.cpp
g++ -g -O2  -I/usr/lib/i386-linux-gnu/wx/include/gtk2-unicode-3.0 -I/usr/include/wx-3.0 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -DwxUSE_GUI=1 -DUSE_FOSSA -I../common -I../../common  -c -o dlgselectdaemoninterface.o dlgselectdaemoninterface.cpp
g++ -g -O2  -I/usr/lib/i386-linux-gnu/wx/include/gtk2-unicode-3.0 -I/usr/include/wx-3.0 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -DwxUSE_GUI=1 -DUSE_FOSSA -I../common -I../../common  -c -o frmmdfeditor.o frmmdfeditor.cpp
g++ -g -O2  -I/usr/lib/i386-linux-gnu/wx/include/gtk2-unicode-3.0 -I/usr/include/wx-3.0 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -DwxUSE_GUI=1 -DUSE_FOSSA -I../common -I../../common  -c -o dlgabout.o dlgabout.cpp
[... cut ...]
07:14:03: Debug: TCP/IP line: 0,20,1,2,1723905480,FF:FF:FF:FF:FF:FF:FF:FD:01:01:00:7F:00:02:00:01,2,16,5 
07:14:05: Debug: ev_handler: TCP/IP receive.
07:14:05: Debug: +OK

07:14:05: Debug: ev_handler: TCP/IP close.
07:14:05: Debug: clientTcpIpWorkerThread: Terminating.
07:14:05: Debug: ev_handler: TCP/IP close.
07:14:05: Debug: clientTcpIpWorkerThread: Terminating.

[1]+  Done                    ./vscpworks
]0;root@ubuntu: ~/vscp_software/src/vscp/vscpworksroot@ubuntu:~/vscp_software/src/vscp/vscpworks# exit

Script done on Mit 04 Feb 2015 07:14:13 CET

grodansparadis, 2015-02-04:
David,

OK then I know what the problem is. This was a real hard one to find. 
Will fix that today.  Thanks for your patience.  I can't count how many 
times I installed Ubuntu and then went through the vscp installl 
sequency yesterday. The worst thing with working alone is that there is 
no once office to run in to and say hurray.  You just sit there at your 
desk and take on the next challange.

Btw you have probably the same problem with the socketdriver then. Go to 
/str/vscp/drivers/level2/socketcan/linux and do the make process.    You 
can do "make install" of course so you get the file in the right place.

Thanks!

/Ake

waaslandwolf skrev den 2015-02-04 07:16:

> Ake,
>
> Just tried that, log is attached. This works, I can do "get interfaces",
> "test connection" and can see events in the monitor window. Doesn't core
> dump any more.
>
> David.
>
> > David,
> >
> > can you test one thing for me.
> >
> > On the system you built where vscpworks crashes
> >
> > go to /src/vscp/vscpworks
> > do "make clean"
> > do "make"
> > When built "./vscpworks"
> > Connect to server. Does it still crash?
> >
> > /Ake
> >
> > waaslandwolf skrev den 2015-02-03 16:36:
> > > Here's an example:
> > >
> > > 
> > > <reg page="0" offset="122">
> > > <name lang="en">PWM test register (TEST_PWM_OUT1)</name>
> > > <description lang="en">
> > > This register holds a PWM value in tens of percents (0
> > > - 10 for 0 to 100%).
> > > </description>
> > > <access>rw</access>
> > > </reg>
> > >
> > >
> > > This crashes VSCPworks on the % sign.
> > >
> > > Do mind that I am testing using an old Windows version of VSCPworks so
> > > this issue might already have been fixed. I'm just reporting any issue
> > > I find, trusting you'd be the judge of what is important to fix :-)
> > >
> > > David.
> > >
> > > > Can you send me a file that is a sample of this. Can't get it to 
> crash
> > > > here.
> > > >
> > > > waaslandwolf skrev den 2015-02-03 13:54:
> > > > >
> > > > > Ake,
[... 598 lines cut ...]
>
> 07:14:03: Debug: TCP/IP line: 
> 0,20,3,2,1723905472,FF:FF:FF:FF:FF:FF:FF:FD:01:01:00:7F:00:02:00:03,0,16,5 
>
> 07:14:03: Debug: ev_handler: TCP/IP receive.
> 07:14:03: Debug: 
> 0,20,1,2,1723905480,FF:FF:FF:FF:FF:FF:FF:FD:01:01:00:7F:00:02:00:01,2,16,5
>
> 07:14:03: Debug: TCP/IP line: 
> 0,20,1,2,1723905480,FF:FF:FF:FF:FF:FF:FF:FD:01:01:00:7F:00:02:00:01,2,16,5 
>
> 07:14:05: Debug: ev_handler: TCP/IP receive.
> 07:14:05: Debug: +OK
>
> 07:14:05: Debug: ev_handler: TCP/IP close.
> 07:14:05: Debug: clientTcpIpWorkerThread: Terminating.
> 07:14:05: Debug: ev_handler: TCP/IP close.
> 07:14:05: Debug: clientTcpIpWorkerThread: Terminating.
>
> [1]+ Done ./vscpworks
> ]0;root@ubuntu: 
> ~/vscp_software/src/vscp/vscpworksroot@ubuntu:~/vscp_software/src/vscp/vscpworks# 
> exit
>
> Script done on Mit 04 Feb 2015 07:14:13 CET