grodansparadis/vscp #71: vscphelper.cpp: Expression is always false because 'else if' condition matches previous condition at line 840.

BlueAndi, 2015-02-03:
Hi Ake,

vscphelper.cpp contains an expression at line 846 which is always false because 'else if' condition matches previous condition at line 840.

One condition should compare against something other than VSCP_CLASS1_MEASUREMENT, the question is, against what? :-)

Best regards
Andreas

grodansparadis, 2015-02-03:
I'm very unresponsive tonight Andy. Searching for the segfault people is 
reporting in. A hard one. But I will check as soon as I can.

Andreas Merkle skrev den 2015-02-03 20:28:

>
> Hi Ake,
>
> vscphelper.cpp contains an expression at line 846 which is always 
> false because 'else if' condition matches previous condition at line 840.
>
> One condition should compare against something other than 
> VSCP_CLASS1_MEASUREMENT, the question is, against what? :-)
>
> Best regards
> Andreas

BlueAndi, 2015-02-03:
Good hunting! :-)

grodansparadis, 2015-02-04:
Fixed! Thanks!

closed 2015-02-04 by grodansparadis