grodansparadis/vscp #75: Can't edit a DM at a page > 0

BlueAndi, 2015-02-05:
Hi Ake,

the function frmDeviceConfig::getPageFromCell() returns a wrong page in case the dm is at a page greater than 0. Thats because it looks like the format "<page>:<register>" is only partly considered.
Can you check this please?

Best regards
Andy

grodansparadis, 2015-02-05:
Dos this happen when you update a dm?

Andreas Merkle skrev den 2015-02-05 22:26:

>
> Hi Ake,
>
> the function frmDeviceConfig::getPageFromCell() returns a wrong page 
> in case the dm is at a page greater than 0. Thats because it looks 
> like the format ":" is only partly considered.
> Can you check this please?
>
> Best regards
> Andy

BlueAndi, 2015-02-05:
Yes

BlueAndi, 2015-02-05:
BTW if I change it locally similar to the getRegFromCell() via tokenizer .... it seems to work.

BlueAndi, 2015-02-05:
Hmm ... still doesn't work ...
vscpworks writes wrong data bytes in my decision matrix row, but I only changed the action id value.

2015-01-05 23:00:51 Rx: class=0x00, type=0x26, prio= 3, oAddr=0x00, -, num=5, data=01 00 01 00 02
2015-01-05 23:00:51 Tx: class=0x00, type=0x27, prio= 3, oAddr=0x01, -, num=5, data=00 00 01 00 02
2015-01-05 23:00:51 Rx: class=0x00, type=0x26, prio= 3, oAddr=0x00, -, num=5, data=01 00 01 01 C0
2015-01-05 23:00:51 Tx: class=0x00, type=0x27, prio= 3, oAddr=0x01, -, num=5, data=00 00 01 01 C0
2015-01-05 23:00:52 Rx: class=0x00, type=0x26, prio= 3, oAddr=0x00, -, num=5, data=01 00 01 02 02
2015-01-05 23:00:52 Tx: class=0x00, type=0x27, prio= 3, oAddr=0x01, -, num=5, data=00 00 01 02 02
2015-01-05 23:00:52 Rx: class=0x00, type=0x26, prio= 3, oAddr=0x00, -, num=5, data=01 00 01 03 C0
2015-01-05 23:00:52 Tx: class=0x00, type=0x27, prio= 3, oAddr=0x01, -, num=5, data=00 00 01 03 C0
2015-01-05 23:00:52 Rx: class=0x00, type=0x26, prio= 3, oAddr=0x00, -, num=5, data=01 00 01 04 02
2015-01-05 23:00:52 Tx: class=0x00, type=0x27, prio= 3, oAddr=0x01, -, num=5, data=00 00 01 04 02
2015-01-05 23:00:53 Rx: class=0x00, type=0x26, prio= 3, oAddr=0x00, -, num=5, data=01 00 01 05 C0
2015-01-05 23:00:53 Tx: class=0x00, type=0x27, prio= 3, oAddr=0x01, -, num=5, data=00 00 01 05 C0
2015-01-05 23:00:53 Rx: class=0x00, type=0x26, prio= 3, oAddr=0x00, -, num=5, data=01 00 01 06 02
2015-01-05 23:00:53 Tx: class=0x00, type=0x27, prio= 3, oAddr=0x01, -, num=5, data=00 00 01 06 02
2015-01-05 23:00:54 Rx: class=0x00, type=0x26, prio= 3, oAddr=0x00, -, num=5, data=01 00 01 07 C0
2015-01-05 23:00:54 Tx: class=0x00, type=0x27, prio= 3, oAddr=0x01, -, num=5, data=00 00 01 07 C0

grodansparadis, 2015-02-05:
Test if the fix helps.

Andreas Merkle skrev den 2015-02-05 23:08:

> Hmm ... still doesn't work ...
> vscpworks writes wrong data bytes in my decision matrix row, but I
> only changed the action id value.
>
> 2015-01-05 23:00:51 Rx: class=0x00, type=0x26, prio= 3, [...]

closed 2015-02-05 by grodansparadis

BlueAndi, 2015-02-05:
Bugfix confirmed! But vscpworks writes still wrong data ;-)
I change only the action id value and the dm row is complete overwritten with the first two dm row bytes repeately.

How the dm row was before editing:
02 C2 14 00 FF 01 FF 00

How it looks like after editing:
02 C0 02 C0 02 C0 02 C0

Should we open another issue?

grodansparadis, 2015-02-05:
Yes open another issue on that. I will have a look at that tomorrow.

/Ake

Andreas Merkle skrev den 2015-02-05 23:22:

> Bugfix confirmed! But vscpworks writes still wrong data ;-)
> I change only the action id value and the dm row is complete
> overwritten with the first two dm row bytes repeately.
>
> How the dm row was [...]

BlueAndi, 2015-02-05:
Ok, done. Thanks in advance.

grodansparadis, 2015-02-06:
Sorry got no time to fix this today.  Will get back to it on Monday.  
Have a nice weekend.

Andreas Merkle skrev den 2015-02-05 23:37:

>
> Ok, done. Thanks in advance.

grodansparadis, 2015-02-16:
Can you check if this is any better at your side now.

/Ake

Andreas Merkle skrev den 2015-02-05 23:37:

>
> Ok, done. Thanks in advance.

BlueAndi, 2015-02-16:
Not right now, because the msvc envirionment is on my business laptop at the company. :-)
But tomorrow I can check.

BTW regarding the problem I created a separate issue, because this one is closed. ;-)
https://github.com/grodansparadis/vscp_software/issues/76

grodansparadis, 2015-02-16:
Andreas Merkle skrev den 2015-02-16 21:42:
>
> Not right now, because the msvc envirionment is on my business laptop 
> at the company. :-)
> But tomorrow I can check.
>
OK. No hurry.  Test when you get a chance.

> BTW regarding the problem I created a separate issue, because this one 
> is closed. ;-)
> #76 <https://github.com/grodansparadis/vscp_software/issues/76>
>

Yes I remember now just took the firts mail n the issue I found..

/Ake

-- 
Be Hungry - Stay Foolish!
Åke Hedman, Grodans Paradis AB / Paradise of the Frog
Brattbergavägen 17, 82050 LOS, SWEDEN
Phone: +46 (0)840011835 CellPhone: +46 (0)730-533146
http://www.grodansparadis.com http://www.vscp.org

---
Detta e-postmeddelande har sökts igenom efter virus med antivirusprogram från Avast.
http://www.avast.com

commit e7efd8fe (2015-02-05): Fixes #75