grodansparadis/vscp #76: Wrong DM update with vscpworks

BlueAndi, 2015-02-05:
I change only the action id value and the dm row is complete overwritten with the first two dm row bytes repeately.

How the dm row was before editing:
02 C2 14 00 FF 01 FF 00

How it looks like after editing:
02 C0 02 C0 02 C0 02 C0

Trace:
2015-01-05 23:00:51 Rx: class=0x00, type=0x26, prio= 3, oAddr=0x00, -, num=5, data=01 00 01 00 02
2015-01-05 23:00:51 Tx: class=0x00, type=0x27, prio= 3, oAddr=0x01, -, num=5, data=00 00 01 00 02
2015-01-05 23:00:51 Rx: class=0x00, type=0x26, prio= 3, oAddr=0x00, -, num=5, data=01 00 01 01 C0
2015-01-05 23:00:51 Tx: class=0x00, type=0x27, prio= 3, oAddr=0x01, -, num=5, data=00 00 01 01 C0
2015-01-05 23:00:52 Rx: class=0x00, type=0x26, prio= 3, oAddr=0x00, -, num=5, data=01 00 01 02 02
2015-01-05 23:00:52 Tx: class=0x00, type=0x27, prio= 3, oAddr=0x01, -, num=5, data=00 00 01 02 02
2015-01-05 23:00:52 Rx: class=0x00, type=0x26, prio= 3, oAddr=0x00, -, num=5, data=01 00 01 03 C0
2015-01-05 23:00:52 Tx: class=0x00, type=0x27, prio= 3, oAddr=0x01, -, num=5, data=00 00 01 03 C0
2015-01-05 23:00:52 Rx: class=0x00, type=0x26, prio= 3, oAddr=0x00, -, num=5, data=01 00 01 04 02
2015-01-05 23:00:52 Tx: class=0x00, type=0x27, prio= 3, oAddr=0x01, -, num=5, data=00 00 01 04 02
2015-01-05 23:00:53 Rx: class=0x00, type=0x26, prio= 3, oAddr=0x00, -, num=5, data=01 00 01 05 C0
2015-01-05 23:00:53 Tx: class=0x00, type=0x27, prio= 3, oAddr=0x01, -, num=5, data=00 00 01 05 C0
2015-01-05 23:00:53 Rx: class=0x00, type=0x26, prio= 3, oAddr=0x00, -, num=5, data=01 00 01 06 02
2015-01-05 23:00:53 Tx: class=0x00, type=0x27, prio= 3, oAddr=0x01, -, num=5, data=00 00 01 06 02
2015-01-05 23:00:54 Rx: class=0x00, type=0x26, prio= 3, oAddr=0x00, -, num=5, data=01 00 01 07 C0
2015-01-05 23:00:54 Tx: class=0x00, type=0x27, prio= 3, oAddr=0x01, -, num=5, data=00 00 01 07 C0

BlueAndi, 2015-02-17:
Hi Ake, the DM is not shown now in vscpworks, but the DM was successful read.

Trace:
2015-02-17 20:02:07 Rx: class=0x00, type=0x25, prio= 3, oAddr=0x00, -, num=5, data=01 00 00 80 80
2015-02-17 20:02:07 Tx: class=0x00, type=0x27, prio= 3, oAddr=0x01, -, num=8, data=00 00 00 80 00 01 0A A0
2015-02-17 20:02:07 Tx: class=0x00, type=0x27, prio= 3, oAddr=0x01, -, num=8, data=01 00 00 84 00 00 00 00
2015-02-17 20:02:07 Tx: class=0x00, type=0x27, prio= 3, oAddr=0x01, -, num=8, data=02 00 00 88 00 00 00 00
2015-02-17 20:02:07 Tx: class=0x00, type=0x27, prio= 3, oAddr=0x01, -, num=8, data=03 00 00 8C 00 00 00 00
2015-02-17 20:02:07 Tx: class=0x00, type=0x27, prio= 3, oAddr=0x01, -, num=8, data=04 00 00 90 00 01 00 00
2015-02-17 20:02:07 Tx: class=0x00, type=0x27, prio= 3, oAddr=0x01, -, num=8, data=05 00 00 94 00 01 00 FF
2015-02-17 20:02:08 Tx: class=0x00, type=0x27, prio= 3, oAddr=0x01, -, num=8, data=06 00 00 98 08 02 00 00
2015-02-17 20:02:08 Tx: class=0x00, type=0x27, prio= 3, oAddr=0x01, -, num=8, data=07 00 00 9C 00 00 00 00
2015-02-17 20:02:08 Tx: class=0x00, type=0x27, prio= 3, oAddr=0x01, -, num=8, data=08 00 00 A0 00 00 00 00
2015-02-17 20:02:08 Tx: class=0x00, type=0x27, prio= 3, oAddr=0x01, -, num=8, data=09 00 00 A4 00 00 00 00
2015-02-17 20:02:08 Tx: class=0x00, type=0x27, prio= 3, oAddr=0x01, -, num=8, data=0A 00 00 A8 00 00 00 00
2015-02-17 20:02:08 Tx: class=0x00, type=0x27, prio= 3, oAddr=0x01, -, num=8, data=0B 00 00 AC 00 00 00 00
2015-02-17 20:02:08 Tx: class=0x00, type=0x27, prio= 3, oAddr=0x01, -, num=8, data=0C 00 00 B0 00 00 00 00
2015-02-17 20:02:08 Tx: class=0x00, type=0x27, prio= 3, oAddr=0x01, -, num=8, data=0D 00 00 B4 00 00 00 00
2015-02-17 20:02:08 Tx: class=0x00, type=0x27, prio= 3, oAddr=0x01, -, num=8, data=0E 00 00 B8 00 00 00 00
2015-02-17 20:02:08 Tx: class=0x00, type=0x27, prio= 3, oAddr=0x01, -, num=8, data=0F 00 00 BC 00 00 00 00
2015-02-17 20:02:09 Tx: class=0x00, type=0x27, prio= 3, oAddr=0x01, -, num=8, data=10 00 00 C0 00 00 00 00
2015-02-17 20:02:09 Tx: class=0x00, type=0x27, prio= 3, oAddr=0x01, -, num=8, data=11 00 00 C4 00 00 00 00
2015-02-17 20:02:09 Tx: class=0x00, type=0x27, prio= 3, oAddr=0x01, -, num=8, data=12 00 00 C8 00 00 00 00
2015-02-17 20:02:09 Tx: class=0x00, type=0x27, prio= 3, oAddr=0x01, -, num=8, data=13 00 00 CC 00 00 00 00
2015-02-17 20:02:09 Tx: class=0x00, type=0x27, prio= 3, oAddr=0x01, -, num=8, data=14 00 00 D0 00 00 00 00
2015-02-17 20:02:09 Tx: class=0x00, type=0x27, prio= 3, oAddr=0x01, -, num=8, data=15 00 00 D4 00 00 00 00
2015-02-17 20:02:09 Tx: class=0x00, type=0x27, prio= 3, oAddr=0x01, -, num=8, data=16 00 00 D8 00 00 00 00
2015-02-17 20:02:09 Tx: class=0x00, type=0x27, prio= 3, oAddr=0x01, -, num=8, data=17 00 00 DC 00 00 00 02
2015-02-17 20:02:10 Tx: class=0x00, type=0x27, prio= 3, oAddr=0x01, -, num=8, data=18 00 00 E0 77 77 77 2E
2015-02-17 20:02:10 Tx: class=0x00, type=0x27, prio= 3, oAddr=0x01, -, num=8, data=19 00 00 E4 62 6C 75 65
2015-02-17 20:02:10 Tx: class=0x00, type=0x27, prio= 3, oAddr=0x01, -, num=8, data=1A 00 00 E8 2D 61 6E 64
2015-02-17 20:02:10 Tx: class=0x00, type=0x27, prio= 3, oAddr=0x01, -, num=8, data=1B 00 00 EC 69 2E 64 65
2015-02-17 20:02:10 Tx: class=0x00, type=0x27, prio= 3, oAddr=0x01, -, num=8, data=1C 00 00 F0 2F 76 73 63
2015-02-17 20:02:10 Tx: class=0x00, type=0x27, prio= 3, oAddr=0x01, -, num=8, data=1D 00 00 F4 70 2F 65 78
2015-02-17 20:02:10 Tx: class=0x00, type=0x27, prio= 3, oAddr=0x01, -, num=8, data=1E 00 00 F8 70 30 32 2E
2015-02-17 20:02:10 Tx: class=0x00, type=0x27, prio= 3, oAddr=0x01, -, num=8, data=1F 00 00 FC 78 6D 6C 00
2015-02-17 20:02:11 Tx: class=0x41, type=0x06, prio= 3, oAddr=0x01, -, num=6, data=00 FF FF 82 6F 45
2015-02-17 20:02:11 Rx: class=0x00, type=0x25, prio= 3, oAddr=0x00, -, num=5, data=01 00 00 00 80
2015-02-17 20:02:11 Tx: class=0x00, type=0x27, prio= 3, oAddr=0x01, -, num=8, data=00 00 00 00 FF FF 00 00
2015-02-17 20:02:12 Tx: class=0x00, type=0x27, prio= 3, oAddr=0x01, -, num=8, data=01 00 00 04 00 00 00 00
2015-02-17 20:02:12 Tx: class=0x00, type=0x27, prio= 3, oAddr=0x01, -, num=8, data=02 00 00 08 00 00 00 00
2015-02-17 20:02:12 Tx: class=0x00, type=0x27, prio= 3, oAddr=0x01, -, num=8, data=03 00 00 0C 00 00 00 00
2015-02-17 20:02:12 Tx: class=0x00, type=0x27, prio= 3, oAddr=0x01, -, num=8, data=04 00 00 10 00 00 00 00
2015-02-17 20:02:12 Tx: class=0x00, type=0x27, prio= 3, oAddr=0x01, -, num=8, data=05 00 00 14 00 00 00 00
2015-02-17 20:02:12 Tx: class=0x00, type=0x27, prio= 3, oAddr=0x01, -, num=8, data=06 00 00 18 00 00 00 00
2015-02-17 20:02:12 Tx: class=0x00, type=0x27, prio= 3, oAddr=0x01, -, num=8, data=07 00 00 1C 00 00 00 00
2015-02-17 20:02:12 Tx: class=0x0A, type=0x06, prio= 3, oAddr=0x01, -, num=4, data=88 82 04 92
2015-02-17 20:02:12 Tx: class=0x00, type=0x27, prio= 3, oAddr=0x01, -, num=8, data=08 00 00 20 00 00 00 00
2015-02-17 20:02:12 Tx: class=0x00, type=0x27, prio= 3, oAddr=0x01, -, num=8, data=09 00 00 24 00 00 00 00
2015-02-17 20:02:13 Tx: class=0x00, type=0x27, prio= 3, oAddr=0x01, -, num=8, data=0A 00 00 28 00 00 00 00
2015-02-17 20:02:13 Tx: class=0x00, type=0x27, prio= 3, oAddr=0x01, -, num=8, data=0B 00 00 2C 00 00 00 00
2015-02-17 20:02:13 Tx: class=0x00, type=0x27, prio= 3, oAddr=0x01, -, num=8, data=0C 00 00 30 00 00 00 00
2015-02-17 20:02:13 Tx: class=0x00, type=0x27, prio= 3, oAddr=0x01, -, num=8, data=0D 00 00 34 00 00 00 00
[... cut ...]
2015-02-17 20:02:30 Tx: class=0x41, type=0x06, prio= 3, oAddr=0x01, -, num=6, data=00 FF FF 82 6F 45
2015-02-17 20:02:31 Tx: class=0x0A, type=0x06, prio= 3, oAddr=0x01, -, num=4, data=88 82 04 92
2015-02-17 20:02:36 Tx: class=0x41, type=0x06, prio= 3, oAddr=0x01, -, num=6, data=00 FF FF 82 6F 45
2015-02-17 20:02:36 Tx: class=0x0A, type=0x06, prio= 3, oAddr=0x01, -, num=4, data=88 82 04 92
2015-02-17 20:02:41 Tx: class=0x41, type=0x06, prio= 3, oAddr=0x01, -, num=6, data=00 FF FF 82 6F 45
2015-02-17 20:02:41 Tx: class=0x0A, type=0x06, prio= 3, oAddr=0x01, -, num=4, data=88 82 04 92

grodansparadis, 2015-02-17:
Hard to say what is wrong. Works here as far as I can see.

Andreas Merkle skrev den 2015-02-17 20:06:

> Hi Ake, the DM is not shown now in vscpworks, but the DM was
> successful read.
>
> Trace:
> 2015-02-17 20:02:07 Rx: class=0x00, type=0x25, prio= 3, oAddr=0x00, -,
> num=5, data=01 00 00 80 80
> 2015-02-17 [...]

BlueAndi, 2015-02-17:
Can I connect my node at your daemon?

BlueAndi, 2015-02-17:
The only difference is that the dm is at page 1, not 0.

grodansparadis, 2015-02-17:
Oh I see you mean the DM?  I found an error when I tested on to connect 
over a server. Will fix that.

Andreas Merkle skrev den 2015-02-17 20:06:

> Hi Ake, the DM is not shown now in vscpworks, but the DM was
> successful read.
>
> Trace:
> 2015-02-17 20:02:07 Rx: class=0x00, type=0x25, prio= 3, oAddr=0x00, -,
> num=5, data=01 00 00 80 80
> 2015-02-17 [...]

grodansparadis, 2015-02-17:
No public avialable yet but I will set up one in Digital Ocean when I 
get some more time to do something more then fixing things.

Andreas Merkle skrev den 2015-02-17 20:12:

>
> Can I connect my node at your daemon?

grodansparadis, 2015-02-17:
I have checked in a new version that should solve that problem.

Andreas Merkle skrev den 2015-02-17 20:06:

> Hi Ake, the DM is not shown now in vscpworks, but the DM was
> successful read.
>
> Trace:
> 2015-02-17 20:02:07 Rx: class=0x00, type=0x25, prio= 3, oAddr=0x00, -,
> num=5, data=01 00 00 80 80
> 2015-02-17 [...]

BlueAndi, 2015-02-17:
Fine, DM is back again. :-) Now I will try to write and read back.

BlueAndi, 2015-02-17:
Bug is solved!! Thanks.

closed 2015-02-17 by BlueAndi

grodansparadis, 2015-02-17:
Thanks to you to.  You do valuable reporting. I can't say how valuable 
your work is.

Andreas Merkle skrev den 2015-02-17 20:44:

>
> Bug is solved!! Thanks.

BlueAndi, 2015-02-17:
You are absolute welcome! This is the one way I can contribute and the funny thing is, all done with a simulated node on the pc and the advantage of a different code base.