grodansparadis/vscp #79: Uninitialized variables in bootdevice_pic1.cpp

BlueAndi, 2015-02-10:
Hi Ake,

the following variables are not initialized at line 654 ... in case of the USE_TCPIP_INTERFACE
* guid3
* guid5
* guid7
* pageSelectMsb
* pageSelectLsb

In case of the USE_DLL_INTERFACE they are initialized by functions.

Best regards
Andreas

grodansparadis, 2015-02-10:
I am working with this now. So there are a lot of issues here.

/Ake

Andreas Merkle skrev den 2015-02-10 22:50:

>
> Hi Ake,
>
> the following variables are not initialized at line 654 ... in case of 
> the USE_TCPIP_INTERFACE
>
>   * guid3
>   * guid5
>   * guid7
>   * pageSelectMsb
>   * pageSelectLsb
>
> In case of the USE_DLL_INTERFACE they are initialized by functions.
>
> Best regards
> Andreas

closed 2015-02-10 by grodansparadis

BlueAndi, 2015-02-10:
Ok, didn't know that its pending work. :-)

grodansparadis, 2015-02-10:
I am rewriting this stuff so it is in a sort of intrim state.  No chance 
for you to know.  It takes longer then expected btw and that is why the 
changes I promise you takes more time.

Andreas Merkle skrev den 2015-02-10 22:57:

>
> Ok, didn't know that its pending work. :-)

BlueAndi, 2015-02-10:
BTW regarding your roadmap ... the pc node is ready for the standard vscp functionality. Only the xml format file for the decision matrix next generation is missing right now. Any wish or do you have more detail about your ideas?

And it looks like that we will get a openHAB binding ... great. So another point of your list. Looks promising.

BlueAndi, 2015-02-10:
And I forgot ... good night! ;-)

grodansparadis, 2015-02-10:
Andreas Merkle skrev den 2015-02-10 23:07:
>
> BTW regarding your roadmap ... the pc node is ready for the standard 
> vscp functionality. Only the xml format file for the decision matrix 
> next generation is missing right now. Any wish or do you have more 
> detail about your ideas?
>
It would be great if it was possible to react on switches, sliders (set 
level) and if it could be setup to send some type of series of 
temperature events.  With that it wold be a perfect counterpart for the 
websocket demos and that package could get live after an installation.

> And it looks like that we will get a openHAB binding ... great. So 
> another point of your list. Looks promising.
>

That is great. Openremote is on the list to. But it is a bit easier I think.

/Ake

-- 
Be Hungry - Stay Foolish!
Åke Hedman, Grodans Paradis AB / Paradise of the Frog
Brattbergavägen 17, 82050 LOS, SWEDEN
Phone: +46 (0)840011835 CellPhone: +46 (0)730-533146
http://www.grodansparadis.com http://www.vscp.org

---
Detta e-postmeddelande har sökts igenom efter virus med antivirusprogram från Avast.
http://www.avast.com

grodansparadis, 2015-02-10:
Same to you Andreas.  I got Gabors flue I think so I will give up for 
today as well.

Andreas Merkle skrev den 2015-02-10 23:09:

>
> And I forgot ... good night! ;-)