grodansparadis/vscp #81: Cannot open include file "afxres.h"

BlueAndi, 2015-02-12:
Hi Ake,

I get the error message above if I try to compile the vscphelper. Do you really need it?

Best regards
Andreas

grodansparadis, 2015-02-12:
Not the lib but the helpertest does. But both compiles fine here.

Andreas Merkle skrev den 2015-02-12 21:11:

>
> Hi Ake,
>
> I get the error message above if I try to compile the vscphelper. Do 
> you really need it?
>
> Best regards
> Andreas

BlueAndi, 2015-02-12:
Because you have MFC installed and I have the express edition without. :-)

grodansparadis, 2015-02-12:
no mfc there so it can go if precompiled header is turned of.  Gone now.

Andreas Merkle skrev den 2015-02-12 21:29:

>
> Because you have MFC installed and I have the express edition without. :-)

closed 2015-02-12 by grodansparadis

BlueAndi, 2015-02-12:
Thanks for the fast support ... as always. :-)

BlueAndi, 2015-03-30:
afx stuff still there:
src/vscp/helperlib/windows/vscphelper.rc
```
#include "afxres.h"
```
and
```
#include "afxres.rc"
```

commit ea45942f (2015-02-12): Fixes #81