grodansparadis/vscp #83: Question about vscphelperlib dll version function

BlueAndi, 2015-02-12:
Why does the vscphlp_getDLLVersion() needs an established connection? It shall only return the version constants of the dll, isn't it?

How about the vscphlp_getVersion() function?

grodansparadis, 2015-02-12:
No need at all but it looks good... ;-)))))

Andreas Merkle skrev den 2015-02-12 21:43:

>
> Why does the vscphlp_getDLLVersion() needs an established connection? 
> It shall only return the version constants of the dll, isn't it?
>
> How about the vscphlp_getVersion() function?

BlueAndi, 2015-02-12:
*LOL* ... copy&paste king ;-)

grodansparadis, 2015-02-12:
;-)  Consider it gone, soon...

Andreas Merkle skrev den 2015-02-12 22:14:

>
> /LOL/ ... copy&paste king ;-)

BlueAndi, 2015-02-12:
okeydokey

BlueAndi, 2015-02-12:
BTW does vscpworks should display a temperature if it receives one?

grodansparadis, 2015-02-12:
Yes at least for the the normalized and the floating point values. 
Hopefully also the string variant.  The websocket event sample can send 
some temperatures.

I am looking into the page problem now at last  so hopefully it is 
resolved until tomorrow.  That is a programmers tomorrow == soon btw.

/Ake

Andreas Merkle skrev den 2015-02-12 22:17:

>
> BTW does vscpworks should display a temperature if it receives one?

BlueAndi, 2015-02-12:
At the moment my node sends CLASS_MEASUREZONE temperature events with normalized values, maybe thats the problem ... hmm ...

Isn't it more dark than bright at your side at the moment? So you have 23 hours before tomorrow :-)

grodansparadis, 2015-02-12:
Andreas Merkle skrev den 2015-02-12 22:26:
>
> At the moment my node sends CLASS_MEASUREZONE temperature events with 
> normalized values, maybe thats the problem ... hmm ...
>
It can be CLASS_MEASUREZONE that is the problem. Pretty new. Try 
CLASS_MEASUREMENT

> Isn't it more dark than bright at your side at the moment? So you have 
> 23 hours before tomorrow :-)
>
;-)

-- 
Be Hungry - Stay Foolish!
Åke Hedman, Grodans Paradis AB / Paradise of the Frog
Brattbergavägen 17, 82050 LOS, SWEDEN
Phone: +46 (0)840011835 CellPhone: +46 (0)730-533146
http://www.grodansparadis.com http://www.vscp.org

---
Detta e-postmeddelande har sökts igenom efter virus med antivirusprogram från Avast.
http://www.avast.com

grodansparadis, 2015-02-17:
I checked this. The handle is actually needed as the ´driver object can be different for different clients. So this will stay.

closed 2015-02-17 by grodansparadis