grodansparadis/vscp #85: Normalized values are shown incorrect in vscpworks

BlueAndi, 2015-02-12:
Example:
Payload: 88 81 01 07
Shown 2630.000000 C
Expected 26.30000000 C

grodansparadis, 2015-02-12:
The correct format is 88 01 01 07 to get 26.3 C   I get confused when I 
see this and have to go to the spec and see that it's explained the 
wrong way around there for some reason. Normally I would say that the 
spec is always right but all code I know of is the other way around.  Or 
maybe I'm just tired now and mess it all up. Will check tomorrow and 
come back. This was bad.

Andreas Merkle skrev den 2015-02-12 23:07:

>
> Example:
> Payload: 88 81 01 07
> Shown 2630.000000 C
> Expected 26.30000000 C

BlueAndi, 2015-02-12:
Oups ... hear you tomorrow ... I am tired too and my bed asks for me.

grodansparadis, 2015-02-12:
Same here. Sweet dreams.

Andreas Merkle skrev den 2015-02-12 23:53:

>
> Oups ... hear you tomorrow ... I am tired too and my bed asks for me.

closed 2015-02-17 by grodansparadis