grodansparadis/vscp #86: Segmentation faults in vscphelper

BlueAndi, 2015-02-17:
From time to time I get segmentation faults in the vscphelper lib (last official version) on linux and windows.

gdb output on linux:
Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0xb6affb40 (LWP 3157)]
ns_out (nc=0x0, buf=0xb4826ff0, len=156130) at ../../../../../common/net_skeleton.c:87
87	  if (nc->flags & NSF_UDP) {

grodansparadis, 2015-02-17:
Something strange there, net_skeleton.c is not used in vscp_helper 
anymore. Or should at least not be.

Andreas Merkle skrev den 2015-02-17 20:17:

> From time to time I get segmentation faults in the vscphelper lib
> (last official version) on linux and windows.
>
> gdb output on linux:
> Program received signal SIGSEGV, Segmentation fault.
> [Switching [...]

grodansparadis, 2015-02-17:
Try ./configure to generate correct makefile and build

Andreas Merkle skrev den 2015-02-17 20:17:

> From time to time I get segmentation faults in the vscphelper lib
> (last official version) on linux and windows.
>
> gdb output on linux:
> Program received signal SIGSEGV, Segmentation fault.
> [Switching [...]

BlueAndi, 2015-02-17:
I don't use the git head, I use the official released version ... so the problem is gone by removing net_skeleton ... I will build a new one for windows and give it a try there ... after I will do it for linux.

grodansparadis, 2015-02-17:
The release version have a couple of errors so I recommend going to the 
head.

btw my fix upload is on the way. Apparently some one else here is using 
all bandwidth at the moment.

Andreas Merkle skrev den 2015-02-17 20:30:

>
> I don't use the git head, I use the official released version ... so 
> the problem is gone by removing net_skeleton ... I will build a new 
> one for windows and give it a try there ... after I will do it for linux.

BlueAndi, 2015-02-17:
Ok, so lets close the issue and I will try later with a new lib.

BlueAndi, 2015-03-01:
I tested yesterday evening again with the current head and the segmentation fault was still there, but less often.

> Program received signal SIGSEGV, Segmentation fault.
> [Switching to Thread 0xb57ffb40 (LWP 9833)]
> __memmove_ia32 () at ../sysdeps/i386/i686/multiarch/../memmove.S:103
> 103	../sysdeps/i386/i686/multiarch/../memmove.S: Datei oder Verzeichnis nicht gefunden.
> (gdb) where
> #0  __memmove_ia32 () at ../sysdeps/i386/i686/multiarch/../memmove.S:103
> #1  0xb7f86c59 in iobuf_insert () from /home/amerkle/Projekte/github/vscp-framework/examples/pc/bin/../lib/linux/libvscphelper.so
> #2  0xb7f86a9f in iobuf_append () from /home/amerkle/Projekte/github/vscp-framework/examples/pc/bin/../lib/linux/libvscphelper.so
> #3  0xb7f86f1d in ns_out () from /home/amerkle/Projekte/github/vscp-framework/examples/pc/bin/../lib/linux/libvscphelper.so
> #4  0xb7f87d9b in ns_send () from /home/amerkle/Projekte/github/vscp-framework/examples/pc/bin/../lib/linux/libvscphelper.so
> #5  0xb7f6cfab in VscpRemoteTcpIf::doCmdSendEx (this=0x8084770, pEvent=pEvent@entry=0xb57feed4)
>     at ../../common/vscpremotetcpif.cpp:711
> #6  0xb7f4d710 in vscphlp_sendEventEx (handle=handle@entry=1681, pEvent=pEvent@entry=0xb57feed4) at ../vscphelperlib.cpp:346
> #7  0x0804bcfe in vscp_tp_adapter_writeMessage (msg=msg@entry=0xb57ff13a) at vscp_user/vscp_tp_adapter.c:286
> #8  0x0804f5a7 in vscp_transport_writeMessage (msg=msg@entry=0xb57ff13a) at ../../vscp/vscp_transport.c:205
> #9  0x0804ecb7 in vscp_core_stateActive () at ../../vscp/vscp_core.c:1048
> #10 vscp_core_process () at ../../vscp/vscp_core.c:467
> #11 0x0804ad95 in vscp_thread_frameworkThread (par=0x80553e0 <vscp_thread_frameworkThrdData>) at vscp_thread.c:269
> #12 0xb7fadf70 in start_thread (arg=0xb57ffb40) at pthread_create.c:312
> #13 0xb7e4b50e in clone () at ../sysdeps/unix/sysv/linux/i386/clone.S:129
> (gdb) list
> 98	in ../sysdeps/i386/i686/multiarch/../memmove.S
> (gdb) frame 5
> #5  0xb7f6cfab in VscpRemoteTcpIf::doCmdSendEx (this=0x8084770, pEvent=pEvent@entry=0xb57feed4)
>     at ../../common/vscpremotetcpif.cpp:711
> 711	                    strBuf.length() );
> (gdb) print strBuf
> $1 = {static npos = 4294967295, m_impl = {static npos = 4294967295, 
>     _M_dataplus = {<std::allocator<wchar_t>> = {<__gnu_cxx::new_allocator<wchar_t>> = {<No data fields>}, <No data fields>}, 
>       _M_p = 0xb3a00014 L"SEND 0,0,0,0,350224384,00:00:00:00:00:00:00:00:00:00:00:00:01:03:00:00,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0"...}}, m_convertedToChar = {
>     m_str = 0xb464df88 "SEND 0,0,0,0,350224384,00:00:00:00:00:00:00:00:00:00:00:00:01:03:00:00,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0"..., m_len = 153789}}

BlueAndi, 2015-03-01:
The event in the strBuf is bad. Its destroyed during the way through the helper lib.

BlueAndi, 2015-03-01:
Hmm ... after running over an hour I got now a out of bounce error on windows:

warning: c:\wxwidgets-3.0.2\include\wx\arrstr.h(178): assert "nIndex < m_nCount" failed in wxArrayString::Item(): wxArra
yString: index out of bounds [in thread 1460]

But this seems to be a different bug.

Sorry, no callstack this time.

grodansparadis, 2015-03-01:
Strange. I haven't seen this a single time since the fix.  Do you have a 
way to repeat it?

Andreas Merkle skrev den 2015-03-01 11:22:

>
> I tested yesterday evening again with the current head and the 
> segmentation fault was still there, but less often.
>
>     Program received signal SIGSEGV, Segmentation fault.
>     [Switching to Thread 0xb57ffb40 (LWP 9833)]
>     __memmove_ia32 () at ../sysdeps/i386/i686/multiarch/../memmove.S:103
>     103 ../sysdeps/i386/i686/multiarch/../memmove.S: Datei oder
>     Verzeichnis nicht gefunden.
>     (gdb) where
>     #0 __memmove_ia32 () at
>     ../sysdeps/i386/i686/multiarch/../memmove.S:103
>     #1 <https://github.com/grodansparadis/vscp_software/issues/1>
>     0xb7f86c59 in iobuf_insert () from
>     /home/amerkle/Projekte/github/vscp-framework/examples/pc/bin/../lib/linux/libvscphelper.so
>     #2 <https://github.com/grodansparadis/vscp_software/issues/2>
>     0xb7f86a9f in iobuf_append () from
>     /home/amerkle/Projekte/github/vscp-framework/examples/pc/bin/../lib/linux/libvscphelper.so
>     #3 <https://github.com/grodansparadis/vscp_software/issues/3>
>     0xb7f86f1d in ns_out () from
>     /home/amerkle/Projekte/github/vscp-framework/examples/pc/bin/../lib/linux/libvscphelper.so
>     #4 <https://github.com/grodansparadis/vscp_software/issues/4>
>     0xb7f87d9b in ns_send () from
>     /home/amerkle/Projekte/github/vscp-framework/examples/pc/bin/../lib/linux/libvscphelper.so
>     #5 <https://github.com/grodansparadis/vscp_software/issues/5>
>     0xb7f6cfab in VscpRemoteTcpIf::doCmdSendEx (this=0x8084770,
>     pEvent=pEvent@entry=0xb57feed4)
>     at ../../common/vscpremotetcpif.cpp:711
>     #6 <https://github.com/grodansparadis/vscp_software/issues/6>
>     0xb7f4d710 in vscphlp_sendEventEx (handle=handle@entry=1681,
>     pEvent=pEvent@entry=0xb57feed4) at ../vscphelperlib.cpp:346
>     #7 <https://github.com/grodansparadis/vscp_software/issues/7>
>     0x0804bcfe in vscp_tp_adapter_writeMessage
>     (msg=msg@entry=0xb57ff13a) at vscp_user/vscp_tp_adapter.c:286
>     #8 <https://github.com/grodansparadis/vscp_software/issues/8>
>     0x0804f5a7 in vscp_transport_writeMessage
>     (msg=msg@entry=0xb57ff13a) at ../../vscp/vscp_transport.c:205
>     #9 <https://github.com/grodansparadis/vscp_software/issues/9>
>     0x0804ecb7 in vscp_core_stateActive () at ../../vscp/vscp_core.c:1048
>     #10 <https://github.com/grodansparadis/vscp_software/issues/10>
>     vscp_core_process () at ../../vscp/vscp_core.c:467
>     #11 <https://github.com/grodansparadis/vscp_software/issues/11>
>     0x0804ad95 in vscp_thread_frameworkThread (par=0x80553e0 ) at
>     vscp_thread.c:269
>     #12 <https://github.com/grodansparadis/vscp_software/issues/12>
>     0xb7fadf70 in start_thread (arg=0xb57ffb40) at pthread_create.c:312
>     #13 <https://github.com/grodansparadis/vscp_software/issues/13>
>     0xb7e4b50e in clone () at ../sysdeps/unix/sysv/linux/i386/clone.S:129
>     (gdb) list
>     98 in ../sysdeps/i386/i686/multiarch/../memmove.S
>     (gdb) frame 5
>     #5 <https://github.com/grodansparadis/vscp_software/issues/5>
>     0xb7f6cfab in VscpRemoteTcpIf::doCmdSendEx (this=0x8084770,
>     pEvent=pEvent@entry=0xb57feed4)
>     at ../../common/vscpremotetcpif.cpp:711
>     711 strBuf.length() );
>     (gdb) print strBuf
>     $1 = {static npos = 4294967295, m_impl = {static npos = 4294967295,
>     _M_dataplus = {std::allocator<wchar_t> =
>     {<__gnu_cxx::new_allocator> = {}, },
>     _M_p = 0xb3a00014 L"SEND
>     0,0,0,0,350224384,00:00:00:00:00:00:00:00:00:00:00:00:01:03:00:00,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0"...}},
>     m_convertedToChar = {
>     m_str = 0xb464df88 "SEND
>     0,0,0,0,350224384,00:00:00:00:00:00:00:00:00:00:00:00:01:03:00:00,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0"...,
>     m_len = 153789}}

BlueAndi, 2015-03-01:
Not really a way ... only running the pc node long enough. I try to hunt this the whole day now and since I use a gdb frontend (a hour ago) it never happen again.

The event is always bad at dll level, one step back to the application where vscphlp_sendEventEx() is called, the event looks perfect. Absolute no glue at the moment how this can happen.

grodansparadis, 2015-03-01:
Hi,

You are on Linux or Windows?   I don't think this is related with the 
last problem. I will try to set up a test for it if you give me your 
environment but can't really spend much time on it until in a couple of 
weeks.

/Ake

Andreas Merkle skrev den 2015-03-01 16:50:

> Not really a way ... only running the pc node long enough. I try to
> hunt this the whole day now and since I use a gdb frontend (a hour
> ago) it never happen again.
>
> The event is always bad at dll [...]

grodansparadis, 2015-03-01:
In libvscphelper, vespworks, the daemon?

Andreas Merkle skrev den 2015-03-01 14:50:

> Hmm ... after running over an hour I got now a out of bounce error on
> windows:
>
> warning: c:\wxwidgets-3.0.2\include\wx\arrstr.h(178): assert "nIndex <
> m_nCount" failed in wxArrayString::Item(): [...]

BlueAndi, 2015-03-01:
I am at both sides, but with different bugs. Both bugs happen in the libvscphelper.
Don't spend too much time for it. Maybe we keep it open and test it just a minute before the release again. I will write here any further info I get during debugging.

grodansparadis, 2015-09-02:
Still a problem?

BlueAndi, 2015-09-02:
Will check it ...

BlueAndi, 2015-09-02:
Can't reproduce it right now, except #128 ... lets close it, until it is back or hopefully never will come again.

closed 2015-09-02 by BlueAndi

grodansparadis, 2015-09-02:
Thumbs held...

Den 2015-09-02 kl. 21:46, skrev Andreas Merkle:

>
> Can't reproduce it right now, except #128 
> <https://github.com/grodansparadis/vscp_software/issues/128> ... lets 
> close it, until it is back or hopefully never will come again.