grodansparadis/vscp #96: Protocol events with types greater than 39 are unknown by vscpworks

BlueAndi, 2015-02-27:
See example:
[image]

BlueAndi, 2015-02-28:
As I see all strings are hardcoded in the source.
How about keeping them external in a xml file? They could be easy extended for everyone, e. g. someone defined own labratory or local class types.

Shall I make a proposal for xml format?

grodansparadis, 2015-03-01:
That is a good suggestion.  Do that.

Andreas Merkle skrev den 2015-02-28 20:05:

>
> As I see all strings are hardcoded in the source.
> How about keeping them external in a xml file? They could be easy 
> extended for everyone, e. g. someone defined own labratory or local 
> class types.
>
> Shall I make a proposal for xml format?

grodansparadis, 2016-05-18:
Will be in SQLite table

closed 2016-05-18 by grodansparadis