grodansparadis/vscp #99: Daemon memory access error

BlueAndi, 2015-03-07:
Hi Ake,

I pulled today the vscp_software and compiled it on the raspberry pi again with "make clean", "make" and "sudo make install".

The daemon starts, but if I connect with the pc node, the daemon throws an memory access error.

Best regards
Andreas

BlueAndi, 2015-03-07:
Note that the raspian os is update to date, which means "sudo apt-get upgrade" shows no packages.

grodansparadis, 2015-03-07:
Connect over tcp/ip i/f?  Just tried, works here.

Andreas Merkle skrev den 2015-03-07 23:40:

>
> Hi Ake,
>
> I pulled today the vscp_software and compiled it on the raspberry pi 
> again with "make clean", "make" and "sudo make install".
>
> The daemon starts, but if I connect with the pc node, the daemon 
> throws an memory access error.
>
> Best regards
> Andreas

BlueAndi, 2015-03-08:
Yes, via tcp/ip. At the moment I can only reproduce it with the pc node on linux. pc node on windows or with vscpworks and etc. works well.

grodansparadis, 2015-03-08:
So

pc mode <-> vscp daemon crashes if both are on Linux?
vscpworks<->vscp daemon on Linux works or not?

/Ake

Andreas Merkle skrev den 2015-03-08 12:51:

>
> Yes, via tcp/ip. At the moment I can only reproduce it with the pc 
> node on linux. pc node on windows or with vscpworks and etc. works well.

BlueAndi, 2015-03-08:
pc node (ubuntu linux) <-> vscp daemon (raspbian linux) crashs
all other variantes are working

grodansparadis, 2015-03-09:
btw Did you do a ./configure also before "make clean"?  The fix I did 
for this problem needs new Makefiles.

Andreas Merkle skrev den 2015-03-07 23:40:

>
> Hi Ake,
>
> I pulled today the vscp_software and compiled it on the raspberry pi 
> again with "make clean", "make" and "sudo make install".
>
> The daemon starts, but if I connect with the pc node, the daemon 
> throws an memory access error.
>
> Best regards
> Andreas

BlueAndi, 2015-03-09:
Hmm ... as far as I remember ... yes I did it. But to be sure, I will compile it again.

grodansparadis, 2015-03-09:
I am unable to repeat here anyway.

Andreas Merkle skrev den 2015-03-09 20:12:

>
> Hmm ... as far as I remember ... yes I did it. But to be sure, I will 
> compile it again.

BlueAndi, 2015-03-09:
I believe you ... hmm ... I will start the compilation in a few minutes ... hopefully I can test it again just a minute before I go to bed. ;-)

BlueAndi, 2015-03-09:
And you tried with my pc node under linux?

grodansparadis, 2015-03-09:
No just works. But it's the same code.

Did you talk any more with Iian btw?

/Ake

Andreas Merkle skrev den 2015-03-09 20:35:

>
> And you tried with my pc node under linux?

BlueAndi, 2015-03-09:
Please try with the pc node, because with vscpworks it worked for me too.

After my first answer he asked me again whether I could implement the framework on a arm m0+ or m4. I wrote that I am intersted, but its not possible right now, maybe we can talk at the end of the year again.

Priority has my own project at the moment ... I just wait for the last parts to solder it. If the prototype works as expected, I will go in mass production. ;-)

grodansparadis, 2015-03-09:
Andreas Merkle skrev den 2015-03-09 21:45:

>
> Please try with the pc node, because with vscpworks it worked for me too.
OK will be later then.

> After my first answer he asked me again whether I could implement the 
> framework on a arm m0+ or m4. I wrote that I am intersted, but its not 
> possible right now, maybe we can talk at the end of the year again.
OK

> Priority has my own project at the moment ... I just wait for the last 
> parts to solder it. If the prototype works as expected, I will go in 
> mass production. ;-)

;-)

BlueAndi, 2015-03-15:
I compiled the daemon again on the raspberry pi with the latest head revision. Called "./configure" too.
But still the same problem appears. Here is now a trace of the daemon on the raspberry pi, made with gdb:

Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0xb4baa2b0 (LWP 16905)]
0xb66f8560 in wxStringBase::find_first_not_of(wchar_t const*, unsigned int) const () from /usr/lib/arm-linux-gnueabihf/libwx_baseu-2.8.so.0
(gdb) backtrace
#0  0xb66f8560 in wxStringBase::find_first_not_of(wchar_t const*, unsigned int) const () from /usr/lib/arm-linux-gnueabihf/libwx_baseu-2.8.so.0
#1  0xb67035e4 in wxStringTokenizer::HasMoreTokens() const () from /usr/lib/arm-linux-gnueabihf/libwx_baseu-2.8.so.0
#2  0x00033210 in VSCPClientThread::handleClientSend (this=0x14a8e8, conn=0xb4ba691c, pCtrlObject=0x140418)
    at ../../common/tcpipclientthread.cpp:776
#3  0x0003c4b4 in VSCPClientThread::CommandHandler (this=0x150ad8, conn=0x149c10, pCtrlObject=0x140418, strCommand=...)
    at ../../common/tcpipclientthread.cpp:302
#4  0x0003d174 in VSCPClientThread::ev_handler (conn=0x149c10, ev=<optimized out>, pUser=<optimized out>)
    at ../../common/tcpipclientthread.cpp:222
#5  0x0009feb0 in ns_call ()
#6  0x000a0fd4 in ns_read_from_socket ()
#7  0x000a17f8 in ns_mgr_poll ()
#8  0x00032b60 in VSCPClientThread::Entry (this=0x150ad8) at ../../common/tcpipclientthread.cpp:108
#9  0xb6728810 in wxThreadInternal::PthreadStart(wxThread*) () from /usr/lib/arm-linux-gnueabihf/libwx_baseu-2.8.so.0
#10 0xb6d7dc00 in start_thread () from /lib/arm-linux-gnueabihf/libpthread.so.0
#11 0xb648e348 in ?? () from /lib/arm-linux-gnueabihf/libc.so.6
#12 0xb648e348 in ?? () from /lib/arm-linux-gnueabihf/libc.so.6
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

BlueAndi, 2015-03-15:
Event of frame 2 is invalid

```
(gdb) print event
$8 = {crc = 1, pdata = 0x804f5b7 <Address 0x804f5b7 out of bounds>, obid = 3019886874, timestamp = 2015, head = 492, vscp_class = 0, 
  vscp_type = 20480, GUID = "o\267\200\207o\267|\321\377\263\001\000\000\000", <incomplete sequence \356>, sizeData = 563}
```

grodansparadis, 2015-03-15:
Same here. Will have stacked but will put priority to this one when I an 
on  it again.  Looks like data size that is not set somewhere. Strange 
if it does not happen on all Linuxes. though.

/Ake

Andreas Merkle skrev den 2015-03-15 12:39:

>
> I compiled the daemon again on the raspberry pi with the latest head 
> revision. Called "./configure" too.
> But still the same problem appears. Here is now a trace of the daemon 
> on the raspberry pi, made with gdb:
>
> Program received signal SIGSEGV, Segmentation fault.
> [Switching to Thread 0xb4baa2b0 (LWP 16905)]
> 0xb66f8560 in wxStringBase::find_first_not_of(wchar_t const/, unsigned 
> int) const () from /usr/lib/arm-linux-gnueabihf/libwx_baseu-2.8.so.0
> (gdb) backtrace
> #0 0xb66f8560 in wxStringBase::find_first_not_of(wchar_t const/, 
> unsigned int) const () from 
> /usr/lib/arm-linux-gnueabihf/libwx_baseu-2.8.so.0
> #1 <https://github.com/grodansparadis/vscp_software/issues/1> 
> 0xb67035e4 in wxStringTokenizer::HasMoreTokens() const () from 
> /usr/lib/arm-linux-gnueabihf/libwx_baseu-2.8.so.0
> #2 <https://github.com/grodansparadis/vscp_software/issues/2> 
> 0x00033210 in VSCPClientThread::handleClientSend (this=0x14a8e8, 
> conn=0xb4ba691c, pCtrlObject=0x140418)
> at ../../common/tcpipclientthread.cpp:776
> #3 <https://github.com/grodansparadis/vscp_software/issues/3> 
> 0x0003c4b4 in VSCPClientThread::CommandHandler (this=0x150ad8, 
> conn=0x149c10, pCtrlObject=0x140418, strCommand=...)
> at ../../common/tcpipclientthread.cpp:302
> #4 <https://github.com/grodansparadis/vscp_software/issues/4> 
> 0x0003d174 in VSCPClientThread::ev_handler (conn=0x149c10, ev=, pUser=)
> at ../../common/tcpipclientthread.cpp:222
> #5 <https://github.com/grodansparadis/vscp_software/issues/5> 
> 0x0009feb0 in ns_call ()
> #6 <https://github.com/grodansparadis/vscp_software/issues/6> 
> 0x000a0fd4 in ns_read_from_socket ()
> #7 <https://github.com/grodansparadis/vscp_software/issues/7> 
> 0x000a17f8 in ns_mgr_poll ()
> #8 <https://github.com/grodansparadis/vscp_software/issues/8> 
> 0x00032b60 in VSCPClientThread::Entry (this=0x150ad8) at 
> ../../common/tcpipclientthread.cpp:108
> #9 <https://github.com/grodansparadis/vscp_software/issues/9> 
> 0xb6728810 in wxThreadInternal::PthreadStart(wxThread*) () from 
> /usr/lib/arm-linux-gnueabihf/libwx_baseu-2.8.so.0
> #10 <https://github.com/grodansparadis/vscp_software/issues/10> 
> 0xb6d7dc00 in start_thread () from 
> /lib/arm-linux-gnueabihf/libpthread.so.0
> #11 <https://github.com/grodansparadis/vscp_software/issues/11> 
> 0xb648e348 in ?? () from /lib/arm-linux-gnueabihf/libc.so.6
> #12 <https://github.com/grodansparadis/vscp_software/issues/12> 
> 0xb648e348 in ?? () from /lib/arm-linux-gnueabihf/libc.so.6
> Backtrace stopped: previous frame identical to this frame (corrupt stack?)

BlueAndi, 2015-03-15:
Note that the conn address is wrong in frame 2, in frame 3 its correct.

BlueAndi, 2015-03-15:
´´´
(gdb) print ((CClientItem*)conn->user_data)->m_currentCommand.m_pchData
$24 = (
    wxChar *) 0x163e3c L"SEND 0,0,0,0,174063616,00:00:00:00:00:00:00:00:00:00:00:00:01:04:00:81,170,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,62,154,209,157,59,8,8,24,55,8,8,66,32,236,183,26,241,95,180,1,0,0,0,254,255,255,255,1,0,0,0,183,245,4,8,26,241,95,180,223,7,0,0,236,1,0,0,0,160,240,183,128,215,240,183,124,241,95,180,1,0,0,0,64,238,4,8,26,241,95,180,2,0,0,0,128,0,0,3,26,241,95,180,1,0,0,0,254,255,255,255,1,0,0,0,15,0,5,8,26,241,95,180,254,255,255,255,37,241,95,180,7,0,0,0,0,0,0,0,0,0,0,0,0,0,3,10,0,6,1,0,0,0,0,4,129,130,5,170,3,0,0,0,0,62,154,209,193,83,5,8,1,0,0,0,128,215,240,183,121,83,5,8,0,0,0,0,170,5,0,0,3,0,0,0,36,170,4,8,0,0,0,0,1,0,0,0,170,5,0,0,254,255,255,255,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,23,240,5,85,0,0,0,0,0,0,0,0,0,0,0,0,0,0,252,183,0,0,0,0,64,251,95,180,104,242,95,180,112,223,250,183,121,83,5,8,2,0,0,0,2,0,0,0,2,0,0,0,208,241,95,180,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,252,183,64,251,95,180,0,15,61,0,104,242,95,180,225,137,62,47,230,151,99,101,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,\030\004\024\060,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,160,222,250,183,0,0,0,0,0,15,61,0,0,0,0,0,238,187,228,183,64,251,95,180,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0"
´´´
This is the command.

BlueAndi, 2015-03-15:
Ake, should the this pointer point to the same object in frame 2 and 3?

grodansparadis, 2015-03-15:
When you talk about frames here do you mean frame==event?  If so yes conn should be the same. It's allocated when the connection is established.

BlueAndi, 2015-03-15:
gdb frame ... see in the trace above, marked with # and the frame number.

BlueAndi, 2015-03-15:
I checked the source and I think the CommandHandler() should call the handleClientSend() of the same object.

grodansparadis, 2015-03-15:
And it's not now?

BlueAndi, 2015-03-16:
No, its not:
```
VSCPClientThread::handleClientSend (this=0x14a8e8 ...
VSCPClientThread::CommandHandler (this=0x150ad8 ...
```
Meanwhile I made the VSCPClientThread class a little bit more robust and the stack overflow in handleClientSend() doesn't happen now. Anyway the problem still exists, but without segmentation fault.
I will continue the investigation, maybe without the stack overflow I see now more ...

BlueAndi, 2015-03-17:
Current backtrace ... I don't get it :-( ... interesting is that  frame 0 this has the same address as frame 1 conn !?

```
#0  VSCPClientThread::handleClientSend (this=0x147db0, conn=0x0, pCtrlObject=0x140418) at ../../common/tcpipclientthread.cpp:848
#1  0x0003bee4 in VSCPClientThread::CommandHandler (this=0x14bdf0, conn=0x147db0, pCtrlObject=0x140418, strCommand=...)
    at ../../common/tcpipclientthread.cpp:348
#2  0x0003d18c in VSCPClientThread::ev_handler (conn=0x147db0, ev=<optimized out>, pUser=<optimized out>)
    at ../../common/tcpipclientthread.cpp:262
#3  0x0009fe30 in ns_call ()
#4  0x000a0f54 in ns_read_from_socket ()
#5  0x000a1778 in ns_mgr_poll ()
#6  0x00032b88 in VSCPClientThread::Entry (this=0x14bdf0) at ../../common/tcpipclientthread.cpp:108
#7  0xb6728810 in wxThreadInternal::PthreadStart(wxThread*) () from /usr/lib/arm-linux-gnueabihf/libwx_baseu-2.8.so.0
#8  0xb6d7dc00 in start_thread () from /lib/arm-linux-gnueabihf/libpthread.so.0
#9  0xb648e348 in ?? () from /lib/arm-linux-gnueabihf/libc.so.6
#10 0xb648e348 in ?? () from /lib/arm-linux-gnueabihf/libc.so.6
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

BlueAndi, 2015-03-17:
Ake, what is the difference between:
pClientItem->m_currentCommand
and
pClientItem->m_currentCommandUC
?

BlueAndi, 2015-03-17:
Because VSCPClientThread::CommandHandler() checks for pClientItem->m_currentCommandUC and calls VSCPClientThread::handleClientSend() which uses pClientItem->m_currentCommand

BlueAndi, 2015-03-17:
Sry ... I found it ... UC = UpperCase

grodansparadis, 2015-03-17:
Impossible for me to go into detail of this, I'm deep into other things, 
and will be for this week.  In a previous dump you  had an event with a 
length that was wrong and that will definitely kill the stack when it is 
deleted. Did you isolate that cause.

How does the code that cause this look like. I mean is it repeatable or 
is it happening  just occasionally.

/Ake

Andreas Merkle skrev den 2015-03-17 22:06:

> Current backtrace ... I don't get it :-( ... interesting is that frame
> 0 this has the same address as frame 1 conn !?
>
> |#0  VSCPClientThread::handleClientSend (this=0x147db0, conn=0x0, [...]

BlueAndi, 2015-03-17:
I made the code more robust, anyway I can reproduce it 100%, but now I know where to set the breakpoint.

The event, which causes the situation looks now:
```
(gdb) print strCmdGo
$8 = {<wxStringBase> = {static npos = 4294967295, 
    m_pchData = 0x151584 L"SEND 0,0,0,0,0,00:00:00:00:00:00:00:00:00:00:00:00:01:01:00:01,\r"}, <No data fields>}

grodansparadis, 2015-03-17:
The little comma at the end looks a little annoying.

And "I made it more robust" is really programmers hubris. Most 
programmers say they make other programmers code more robust. All 
fools.  ;-)

Andreas Merkle skrev den 2015-03-17 23:20:

> I made the code more robust, anyway I can reproduce it 100%, but now I
> know where to set the breakpoint.
>
> The event, which causes the situation looks now:
>
> |(gdb) print strCmdGo
> $8 = [...]

BlueAndi, 2015-03-17:
;-)

The following code produces that:
```
			// Check if command already in buffer
			while ( wxNOT_FOUND != ( pos4lf = pClientItem->m_readBuffer.Find ( (const char)0x0a ) ) ) {
				wxString strCmdGo = pClientItem->m_readBuffer.Mid( 0, pos4lf );
				pCtrlObject->getTCPIPServer()->CommandHandler( conn, 
										pCtrlObject, 
										strCmdGo );
				pClientItem->m_readBuffer = pClientItem->m_readBuffer.Right( pClientItem->m_readBuffer.Length()-pos4lf-1 );
			}
```
But more maybe tomorrow ... good night!

grodansparadis, 2015-03-18:
The strangest thing with this is that it works on a standard Linux 
machine.  That suggest something else that a logic error IMO. Are the 
setup really the same? Same drivers active on both platforms? Have you 
tried to disable all drivers?

Andreas Merkle skrev den 2015-03-17 22:06:

> Current backtrace ... I don't get it :-( ... interesting is that frame
> 0 this has the same address as frame 1 conn !?
>
> |#0  VSCPClientThread::handleClientSend (this=0x147db0, conn=0x0, [...]

BlueAndi, 2015-03-20:
> The little comma at the end looks a little annoying.

I found it!
If a event has no data, this will happen in the doCmdSend() methods.

I fixed it ... but the problem still appears *grrr*.
And yes, I use everywhere the current head revision of vscp_software.

grodansparadis, 2015-03-20:
Yes I guessed that. How was it with drivers. What drivers do you have 
active?

Andreas Merkle skrev den 2015-03-20 21:40:

>
>     The little comma at the end looks a little annoying.
>
> I found it!
> If a event has no data, this will happen in the doCmdSend() methods.
>
> I fixed it ... but the problem still appears /grrr/.
> And yes, I use everywhere the current head revision of vscp_software.

BlueAndi, 2015-04-17:
Fixed by removing -fpack-struct in the makefile. Thanks again for your support!

closed 2015-04-17 by BlueAndi