grodansparadis/vscp #101: Return value of canXXXToCanal not used.

BlueAndi, 2015-03-12:
Hi Ake,

in all canXXXToCanal functions the rv variable is never set to true and the function returns always false.

Best regards
Andreas

grodansparadis, 2015-03-12:
Where are they?

BlueAndi, 2015-03-12:
level1\rs485\common\rs485obj.cpp
level1\rs485\win32\test\rs485drv.cpp
level1\can232\win32\Test\CAN232Drv.cpp
level1\can232\common\can232obj.cpp
level1\can232\linux\can232obj.cpp

grodansparadis, 2015-03-15:
OK will stack this up  as for now. Busy with other things at the moment.

/Ake

Andreas Merkle skrev den 2015-03-12 22:51:

>
> level1\rs485\common\rs485obj.cpp
> level1\rs485\win32\test\rs485drv.cpp
> level1\can232\win32\Test\CAN232Drv.cpp
> level1\can232\common\can232obj.cpp
> level1\can232\linux\can232obj.cpp

commit 72699897 (2015-09-18): Fixed #101
closed 2015-09-18 by grodansparadis