grodansparadis/vscp #121: Temperature event value with 1.0.1

mamac, 2015-04-13:
Hello

Using VSCP 1.0.1 Neon.

Since I upgraded to that version, temperature values received by the daemon are multiplied by 100.
Screenshot below should show a temperature of 20.60 (2060 received):
[image]

Thank you,
JM

mamac, 2015-04-13:
Humidity doesn't appear to be affected:
[image]

closed 2015-04-13 by mamac

grodansparadis, 2015-04-13:
Strange. Nothing changed there AFAIK.  Looks like this sample uses the 
old coding. Remember the change we did for some time ago?  Yes this is 
2060 degrees.

/Ake

mamac skrev den 2015-04-13 21:11:

>
> Hello
>
> Using VSCP 1.0.1 Neon.
>
> Since I upgraded to that version, temperature values received by the 
> daemon are multiplied by 100.
> Screenshot below should show a temperature of 20.60 (2060 received):
> image 
> <https://cloud.githubusercontent.com/assets/7148802/7123288/779e20d0-e221-11e4-95cd-5d684d3e1adf.png>
>
> Thank you,
> JM

tmhfr, 2015-04-14:
Strange indeed.

Something I don't get, 0xCE is 206, and we have decimal position set to 1, so the result should be 20.6, shouldn't it?

At least it was like this before the upgrade and I didn't change anything on the firmware of my nodes.

You can see on my temp charts when the switch happened last night: http://www.lljm.net:8081/fr/tempvisual/2015-04-13,2015-04-14,FF:FF:FF:FF:FF:FF:FF:FC:01:01:01:01:01:01:00:01

Thank you,
JM

grodansparadis, 2015-04-14:
There was a warning posted some time ago (See here 
https://groups.google.com/forum/#!topic/vscp/EV7t0jQ2koE 
<https://groups.google.com/forum/#%21topic/vscp/EV7t0jQ2koE>) that my 
code used bit 7 the wrong way to what is said in the specification.

0x01 0xCE

206 * 10^1 = 2060

the fault I had done was doing it the other way around

206 * 10^-1 = 20.60

This code was changed in vscpworks and in the web lib.  The description 
in the spec. is correct 
http://www.vscp.org/docs/vscpspec/doku.php?id=data_coding

/Ake

Jean-Marc Beaune skrev den 2015-04-14 08:10:

> Strange indeed.
>
> Something I don't get, 0xCE is 206, and we have decimal position set
> to 1, so the result should be 20.6, shouldn't it?
>
> At least it was like this before the upgrade and I didn't [...]

grodansparadis, 2015-04-14:
Btw Very nice page you have created!

Jean-Marc Beaune skrev den 2015-04-14 08:10:

> Strange indeed.
>
> Something I don't get, 0xCE is 206, and we have decimal position set
> to 1, so the result should be 20.6, shouldn't it?
>
> At least it was like this before the upgrade and I didn't [...]

mamac, 2015-04-14:
Well... I just missed this communication, I though it wouldn't affect me...
I'll update my nodes then.

Thank you,
JM

grodansparadis, 2015-04-14:
Hi JM,

all together my fault. I am sorry if it cause you a lot of problems.  
But the spec is right even when I read it wrong.

/Ake

mamac skrev den 2015-04-14 10:36:

> Well... I just missed this communication, I though it wouldn't affect 
> me...
> I'll update my nodes then.
>
> Thank you,
> JM
>
>
> > There was a warning posted some time ago (See here
> > https://groups.google.com/forum/#!topic/vscp/EV7t0jQ2koE
> > <https://groups.google.com/forum/#%21topic/vscp/EV7t0jQ2koE>) that my
> > code used bit 7 the wrong way to what is said in the specification.
> >
> > 0x01 0xCE
> >
> > 206 * 10^1 = 2060
> >
> > the fault I had done was doing it the other way around
> >
> > 206 * 10^-1 = 20.60
> >
> > This code was changed in vscpworks and in the web lib. The description
> > in the spec. is correct
> > http://www.vscp.org/docs/vscpspec/doku.php?id=data_coding
> >
> > /Ake
> >
> > Jean-Marc Beaune skrev den 2015-04-14 08:10:
> > >
> > > Strange indeed.
> > >
> > > Something I don't get, 0xCE is 206, and we have decimal position set
> > > to 1, so the result should be 20.6, shouldn't it?
> > >
> > > At least it was like this before the upgrade and I didn't change
> > > anything on the firmware of my nodes.
> > >
> > > You can see on my temp charts when the switch happened last night:
> > >
> > 
> http://www.lljm.net:8081/fr/tempvisual/2015-04-13,2015-04-14,FF:FF:FF:FF:FF:FF:FF:FC:01:01:01:01:01:01:00:01
> > >
> > > Thank you,
> > > JM
> > >
> > > <
> > 
> https://github.com/grodansparadis/vscp_software/issues/121#issuecomment-92626697
> > >
> >
> > Be Hungry - Stay Foolish!
> > Åke Hedman, Grodans Paradis AB / Paradise of the Frog
> > Brattbergavägen 17, 82050 LOS, SWEDEN
> > Phone: +46 (0)840011835 CellPhone: +46 (0)730-533146
> > http://www.grodansparadis.com http://www.vscp.org
> >
> >
> >
> > ---
> > Detta e-postmeddelande har sökts igenom efter virus med antivirusprogram
> > från Avast.
> > http://www.avast.com

mamac, 2015-04-14:
Not a big deal at all.
I only have 5 temperature nodes installed in my house so far.
Just read the spec again and understood I only had to change the value of the normalizer byte.
Tested it and it works fine, so everything is back to normal now!
Thank you ;-)

grodansparadis, 2015-04-14:
Good!

mamac skrev den 2015-04-14 22:19:

>
> Not a big deal at all.
> I only have 5 temperature nodes installed in my house so far.
> Just read the spec again and understood I only had to change the value 
> of the normalizer byte.
> Tested it and it works fine, so everything is back to normal now!
> Thank you ;-)