grodansparadis/vscp #122: Not possible to allow everyone to access the daemon remotely

mamac, 2015-04-17:
Hello

I have the setup below in vscpd.conf:

    <remoteuser>
        <user>
            <name>admin</name>
            <password>d50c3180375c27927c22e42a379c3f67</password>
            <privilege>admin</privilege>
            <allowfrom>*</allowfrom>
        </user>
        <user>
            <name>user</name>
            <password>ee11cbb19052e40b07aac0ca060c23ee</password>
            <privilege>admin</privilege>
            <allowfrom>*</allowfrom>
        </user>
    </remoteuser>
But the client has an access denied message while connecting to the daemon via websocket.

Thank you,
JM

grodansparadis, 2015-04-18:
Hmmm... Strange looks like it works here. I will have a check again.

/Ake

mamac skrev den 2015-04-17 19:09:

> Hello
>
> I have the setup below in vscpd.conf:
>
> |<remoteuser>
>      <user>
>          <name>admin</name>
>          <password>d50c3180375c27927c22e42a379c3f67</password> [...]

grodansparadis, 2015-09-16:
This should be fixed now.  Works on demo.vscp.org  Open again if not

closed 2015-09-16 by grodansparadis