grodansparadis/vscp #127: vscpl1_usb2candrv.dll not able to open more than one interface

datenheim, 2015-04-28:
System: Windows 7-64, latest release of vscpd/works, USB2CAN interface with firmware 1.5 (latest)

Neither the usb2can.dll provided by 8devices nor the vscpl1_usb2candrv.dll provided in:
https://github.com/grodansparadis/vscp_software/tree/master/src/vscp/drivers/level1/usb2can

is able to open more than one CAN interface at any one time. This seems to be known problem, as in file
https://github.com/grodansparadis/vscp_software/blob/master/src/vscp/drivers/level1/usb2can/usb2can/common/dlldrvobj.h
line 41 the maximum number of opened drivers is limited to 1. Changing this constant from 1 to 256 does not change the behaviour - still only one channel can be opened, further calls to CanalOpen() will not do anything but return an error.

Furthermore the handle returned by CanalOpen() is always 1681 for any number of usb2can-interfaces on the same machine.

That said, it seems that in file
https://github.com/grodansparadis/vscp_software/blob/master/src/vscp/drivers/level1/usb2can/usb2can/common/dlldrvobj.cpp
in line 178 we get always 1681 as handle. This may be the case because the break in line 179 always breaks the for-loop counting through all interfaces.

Sadly even with removing the break (which seems to be obvious) the dll is still not able to open multiple interfaces - but the returned handle now changes, at least sometimes, to values of 1681, 1682, 1683 and 1684 (and probably more) - but still only one usb2can is delivering data.

Reproduction:
1. have two usb2can-interfaces with different serial numbers,
    having at least one node running on each interface makes sense,
    but is not required to produce the error

2. configure vscpd to have those two interfaces enabled and supported

3. start vscpd and watch the debug output

I always got:
```shell
D:\sglux\vscp>vscpd -v
2015-04-27 17:00:50 - Reading configuration from [C:\ProgramData/vscp/vscpd.conf]
2015-04-27 17:00:50 - Level I driver added. - [ D:\sglux\usb2can-dll-versionen\vscpl1_usb2candrv-open255-olesch.dll ]
2015-04-27 17:00:50 - Level I driver added. - [ D:\sglux\usb2can-dll-versionen\vscpl1_usb2candrv-open255-olesch.dll ]
2015-04-27 17:00:50 - VSCP Daemon started
Version: 1.0.0.43 Neon
Copyright (C) 2000-2014, Grodans Paradis AB, http://www.paradiseofthefrog.com
2015-04-27 17:00:50 - Log Level=1
2015-04-27 17:00:50 - DM enabled.
2015-04-27 17:00:50 - Variables enabled.
2015-04-27 17:00:50 - TCP/IP interface enabled.
2015-04-27 17:00:50 - WebServer interface active.
2015-04-27 17:00:50 - Level I drivers enabled.
2015-04-27 17:00:50 - USB2CAN_206 Started at 2015-04-27 17:00:50
2015-04-27 17:00:50 - USB2CAN_221 Started at 2015-04-27 17:00:50
2015-04-27 17:00:50 - TCP Client: Thread started.
2015-04-27 17:00:50 - Loading level I driver: USB2CAN_221
2015-04-27 17:00:50 - Loading level I driver: USB2CAN_206
2015-04-27 17:00:50 - Failed to open driver. Will not use it!
        [ USB2CAN_206 ]
2015-04-27 17:00:50 - Driver USB2CAN_221 opended.
```
Assumption: the strange thing is that always the same handle is returned by CanalOpen() - cause unknown so far.

grodansparadis, 2015-04-28:
OK I will test as soon as I get some time to sit with it.  Great report!

Stefan Langer skrev den 2015-04-28 16:16:

>
> System: Windows 7-64, latest release of vscpd/works, USB2CAN interface 
> with firmware 1.5 (latest)
>
> Neither the usb2can.dll provided by 8devices nor the 
> vscpl1_usb2candrv.dll provided in:
> https://github.com/grodansparadis/vscp_software/tree/master/src/vscp/drivers/level1/usb2can
>
> is able to open more than one CAN interface at any one time. This 
> seems to be known problem, as in file
> https://github.com/grodansparadis/vscp_software/blob/master/src/vscp/drivers/level1/usb2can/usb2can/common/dlldrvobj.h
> line 41 the maximum number of opened drivers is limited to 1. Changing 
> this constant from 1 to 256 does not change the behaviour - still only 
> one channel can be opened, further calls to CanalOpen() will not do 
> anything but return an error.
>
> Furthermore the handle returned by CanalOpen() is always 1681 for any 
> number of usb2can-interfaces on the same machine.
>
> That said, it seems that in file
> https://github.com/grodansparadis/vscp_software/blob/master/src/vscp/drivers/level1/usb2can/usb2can/common/dlldrvobj.cpp
> in line 178 we get always 1681 as handle. This may be the case because 
> the break in line 179 always breaks the for-loop counting through all 
> interfaces.
>
> Sadly even with removing the break (which seems to be obvious) the dll 
> is still not able to open multiple interfaces - but the returned 
> handle now changes, at least sometimes, to values of 1681, 1682, 1683 
> and 1684 (and probably more) - but still only one usb2can is 
> delivering data.
>
> Reproduction:
> 1. have two usb2can-interfaces with different serial numbers,
> having at least one node running on each interface makes sense,
> but is not required to produce the error
>
> 1.
>
>     configure vscpd to have those two interfaces enabled and supported
>
> 2.
>
>     start vscpd and watch the debug output
>
> I always got:
>
> D:\sglux\vscp>vscpd -v
> 2015-04-27 17:00:50 - Reading configuration from [C:\ProgramData/vscp/vscpd.conf]
> 2015-04-27 17:00:50 - Level I driver added. - [ D:\sglux\usb2can-dll-versionen\vscpl1_usb2candrv-open255-olesch.dll ]
> 2015-04-27 17:00:50 - Level I driver added. - [ D:\sglux\usb2can-dll-versionen\vscpl1_usb2candrv-open255-olesch.dll ]
> 2015-04-27 17:00:50 - VSCP Daemon started
> Version: 1.0.0.43 Neon
> Copyright (C) 2000-2014, Grodans Paradis AB, http://www.paradiseofthefrog.com
> 2015-04-27 17:00:50 - Log Level=1
> 2015-04-27 17:00:50 - DM enabled.
> 2015-04-27 17:00:50 - Variables enabled.
> 2015-04-27 17:00:50 - TCP/IP interface enabled.
> 2015-04-27 17:00:50 - WebServer interface active.
> 2015-04-27 17:00:50 - Level I drivers enabled.
> 2015-04-27 17:00:50 - USB2CAN_206 Started at 2015-04-27 17:00:50
> 2015-04-27 17:00:50 - USB2CAN_221 Started at 2015-04-27 17:00:50
> 2015-04-27 17:00:50 - TCP Client: Thread started.
> 2015-04-27 17:00:50 - Loading level I driver: USB2CAN_221
> 2015-04-27 17:00:50 - Loading level I driver: USB2CAN_206
> 2015-04-27 17:00:50 - Failed to open driver. Will not use it!
>          [ USB2CAN_206 ]
> 2015-04-27 17:00:50 - Driver USB2CAN_221 opended.
>
> Assumption: the strange thing is that always the same handle is 
> returned by CanalOpen() - cause unknown so far.

BlueAndi, 2015-04-28:
Hope its ok for both of you that I am writing something to the problem,  because some days ago I took a look at that area.

1. CANAL_USB2CAN_DRIVER_MAX_OPEN has to be set greater than 1, because the dll driver limits with that the max. number of usb2can objects.

2. In my eyes the "break" in https://github.com/grodansparadis/vscp_software/blob/master/src/vscp/drivers/level1/usb2can/usb2can/common/dlldrvobj.cpp you mentioned Stefan, is necessary. Because the CDllDrvObj::addDriver method searchs for a empty slot and if a empty slot is found, it will assign the usb2can object to this slot and break out of the loop.

If you really change only the CANAL_USB2CAN_DRIVER_MAX_OPEN to e. g. 10, then clean everything and compile it again, it should run.

Your log shows "Failed to open driver. Will not use it!", which is thrown only in case that the handle is lower or equal than 0, not 1681! So it is strange that you see always 1681?! But it would match in case the CANAL_USB2CAN_DRIVER_MAX_OPEN is set to 1.

Please add a printf after line 97, after line 104 and after line 106. in CanalOpen (https://github.com/grodansparadis/vscp_software/blob/master/src/vscp/drivers/level1/usb2can/usb2can/win32/usb2candrv.cpp) and we should see how it behave.

grodansparadis, 2015-04-28:
You are welcome to take the task on Andreas. I have plenty of other 
stuff on my desk anyway. ;-)

Andreas Merkle skrev den 2015-04-28 20:57:

> Hope its ok for both of you that I am writing something to the
> problem, because some days ago I took a look at that area.
>
> 1.
>
>     CANAL_USB2CAN_DRIVER_MAX_OPEN has to be set greater than 1, [...]

BlueAndi, 2015-04-28:
I will support as much as I can, but you know, I only have one usb2can adapter right now. So I am not able to test it, only to make proposals how we could get closer to the problem. Stefan has to be the test roboter. ;-)

datenheim, 2015-04-29:
Andreas, thanks for jumping in - you are welcome!

I will test your suggestions this afternoon.

Depending where your homebase is located I may also leave a second interface to you...

Thank you, regards!

BlueAndi, 2015-05-05:
Stefan lets see the resulsts first, maybe we can surround the problem together.

datenheim, 2015-05-06:
Andreas, with your suggestions we are one step forward. Only change done was in dlldrvobj.h:
```c++
#define CANAL_USB2CAN_DRIVER_MAX_OPEN	32
```
Opening multiple usb2can now works and different handles are returned. So far I have checked with vscpd 1.0.0.4.3 opening two usb2can (firmware 1.5, driver from 8devices website), one node connected to each usb2can, and vscpworks 1.0.0.4.3 connected the local deamon under windows 7-64, 
```
>vscpd -v -d 3 -c D:\Software\Technik\VSCP\vscpd-dev-dll.conf

2015-05-06 14:20:22 - Reading configuration from [D:\Software\Technik\VSCP\vscpd-dev-dll.conf]
2015-05-06 14:20:22 - Level I driver added. - [ D:\Repository\vscp_software_git\src\vscp\drivers\level1\usb2can\usb2can\Debug\vscpl1_usb2candrv.dll ]
2015-05-06 14:20:22 - Level I driver added. - [ D:\Repository\vscp_software_git\src\vscp\drivers\level1\usb2can\usb2can\Debug\vscpl1_usb2candrv.dll ]
2015-05-06 14:20:22 - VSCP Daemon started
Version: 1.0.0.43 Neon
Copyright (C) 2000-2014, Grodans Paradis AB, http://www.paradiseofthefrog.com
2015-05-06 14:20:22 - Log Level=1
2015-05-06 14:20:22 - DM disabled.
2015-05-06 14:20:22 - Variables enabled.
2015-05-06 14:20:22 - TCP/IP interface enabled.
2015-05-06 14:20:22 - WebServer interface active.
2015-05-06 14:20:22 - Level I drivers enabled.
2015-05-06 14:20:22 - USB2CAN_221_dev_dll Started at 2015-05-06 14:20:22
2015-05-06 14:20:22 - USB2CAN_223_dev_dll Started at 2015-05-06 14:20:22
2015-05-06 14:20:22 - TCP Client: Thread started.
2015-05-06 14:20:22 - Loading level I driver: USB2CAN_221_dev_dll
2015-05-06 14:20:22 - Loading level I driver: USB2CAN_223_dev_dll
H3: 1682
H3: 1681
2015-05-06 14:20:22 - Driver USB2CAN_223_dev_dll opended.
2015-05-06 14:20:22 - Driver USB2CAN_221_dev_dll opended.
```
That said, It looks promising.

But there is another issue: Running the above mentioned setup on should be able to connect and disconnect nodes on either usb2can at any time. Data should arrive as soon as the nodes are connected.

If I have everything connected, start the deamon and works, I can see the data of all nodes.
But If I not disconnect a node on either of the interfaces, the reaction upon reconnect of that node to the same usb2can differs. If I disconnect and reconnect the node on the first opened usb2can (as listed in deamon log, see above) data will appear as expected.
Doing the same with the node on the second usb2can NO EVENTS are displayed in works after reconnecting that node.
I have to restart the daemon to again get data from both nodes. Strange, can't tell if it's a daemon-, usb2can-driver or usb2can-canal issue. This is reproducible.
Sometimes after restarting the daemon the usb2can interfaces are opened in another order (strange!) and the node-reconnect-issue also changes accordingly, e.g. node reconnects on the first opened usb2can is working and not working on the ones opened later.

Next I will test with my own applications and with more than two usb2can and more nodes.

grodansparadis, 2015-05-06:
I have a report that the first event is always lost. Can possibly be related to that. Some semaphore or mutex that are not initialized correctly. But connecting and disconnecting nodes should not have anything to do with that on the other hand. But I still guess on a mutex problem in the driver as it sounds.

datenheim, 2015-05-07:
Today we were able to successfully open two USB2CAN with vscpd as well as our own software (labview), got different handles and could receive data from the nodes.

Next tasks:
* test with more than 2 usb2can and more nodes
* node-reconnect with own software
* investigate the CEnum::MAX_USB_DEVICES impact as Andreas noted

and I will report here.

BlueAndi, 2015-05-07:
Stupid question Stefan, but who is "we"?

sglux, 2017-02-08:
we was me and "my" student working with a bunch of vscp nodes on a sensor project.

grodansparadis, 2017-02-08:
How is status on this issue?

datenheim, 2017-02-08:
In fact BlueAndi's modifications allowed to compile a DLL that proved to work with 2x USB2CAN and a couple (not more than 4) nodes. I have not tested beyond this.
So the status in detail is:

- The original DLL from 8devices and also the older usb2can.dll from vscp repository works with only single USB2CAN.
- This DLL also runs under XP up to Win7 32 Bit, but can be troublesome with 64 Bit and newer windows.
- Part of the problem is the USB driver provided by 8devices. The current driver runs fine up to Win10 for both 32 and 64 bit, but they still provide the old usb2can.DLL supporting only 1 USB2CAN.
- The modified DLL (not sure if vscp code base contains that change today) works with at least two USB2CAN
- The modified DLL needs testing with >3 USB2CAn and >4 nodes in total

grodansparadis, 2017-02-08:
OK I close this for now then as it is more of an 8Devices issue.

closed 2017-02-08 by grodansparadis