grodansparadis/vscp #128: Can't receive any event from demoserver

BlueAndi, 2015-04-29:
I connected with the simulated node on windows (vscp helper lib from Neon release 1.0.1) to the demoserver.

Every event the pc example sends to the demoserver, is received successful (can be seen in the websocket vscpws_event demo).

But every event from the simulation driver can't be received by the simulated node, which means the function vscphlp_receiveEventEx() returns always VSCP_ERROR_PARAMETER.

BlueAndi, 2015-04-29:
I connected now several times and it happens in about 75%.
It may help to know that tests with my VSCP daemon (v1.0.0) on a raspberry pi are 100% successful at anytime.

grodansparadis, 2015-04-29:
Have you tested to change (increase) the timings?

Andreas Merkle skrev den 2015-04-29 22:58:

>
> I connected now several times and it happens in about 75%.
> It may help to know that tests with my VSCP daemon (v1.0.0) on a 
> raspberry pi are 100% successful at anytime.

BlueAndi, 2015-04-29:
Response timeout is set to 6s. But as I see, it happens right after the connection establishment.

See the command flow here:

1. vscphlp_newSession
2. vscphlp_setResponseTimeout
3. vscphlp_open
4. vscphlp_noop
5. vscphlp_setFilter
6. vscphlp_clearDaemonEventQueue
7. vscphlp_getVersion fails, because it returns 0.0.0 in case it doesn't work

Strange ...

grodansparadis, 2015-04-29:
I tested in VSCP Works and it seem to receive the events.

BlueAndi, 2015-04-29:
I see them in VSCP Works v1.0.1.2 too. But the simulated node uses the same VSCP helper library dll and it fails in most of the cases after this straight forward flow, shown above.

Maybe time again to go to bed and check it the next days again ...

BlueAndi, 2015-04-29:
Good case:
```
2015-04-29 23:45:34 Connecting ...
2015-04-29 23:45:35 vscp_tp_adapter.c (546) - Info: Connected.
2015-04-29 23:45:36 Remote VSCP daemon v1.0.1
2015-04-29 23:45:36 Used VSCP dll version 0x00000003
2015-04-29 23:45:36 Vendor of driver: Grodans Paradis AB, Sweden
2015-04-29 23:45:36 Connection successful.
2015-04-29 23:45:36 Rx: class=0x0A, type=0x06, prio= 3, oAddr=0x00, -, num=4, data=8A 82 08 1E
2015-04-29 23:45:36 Lamp on
2015-04-29 23:45:36 Tx: class=0x00, type=0x02, prio= 0, oAddr=0x01, -, num=1, data=01 New Node Online
2015-04-29 23:45:38 Tx: class=0x14, type=0x09, prio= 7, oAddr=0x01, -, num=3, data=00 FF FF
2015-04-29 23:45:38 Rx: class=0x0A, type=0x06, prio= 3, oAddr=0x00, -, num=4, data=8A 82 08 13
2015-04-29 23:45:40 Tx: class=0x14, type=0x09, prio= 7, oAddr=0x01, -, num=3, data=00 FF FF
2015-04-29 23:45:41 Rx: class=0x0A, type=0x06, prio= 3, oAddr=0x00, -, num=4, data=8A 82 08 06
2015-04-29 23:45:42 Tx: class=0x14, type=0x09, prio= 7, oAddr=0x01, -, num=3, data=00 FF FF
2015-04-29 23:45:43 Rx: class=0x0A, type=0x06, prio= 3, oAddr=0x00, -, num=4, data=8A 82 08 03
```

Bad case:
```
2015-04-29 23:45:26 Connecting ...
2015-04-29 23:45:27 vscp_tp_adapter.c (546) - Info: Connected.
2015-04-29 23:45:27 Remote VSCP daemon v0.0.0
2015-04-29 23:45:27 Used VSCP dll version 0x00000003
2015-04-29 23:45:27 Vendor of driver: Grodans Paradis AB, Sweden
2015-04-29 23:45:27 Connection successful.
2015-04-29 23:45:27 Lamp on
2015-04-29 23:45:27 Tx: class=0x00, type=0x02, prio= 0, oAddr=0x01, -, num=1, data=01 New Node Online
2015-04-29 23:45:28 vscp_tp_adapter.c (277) - Warning: Couldn't receive event:  34
2015-04-29 23:45:28 vscp_tp_adapter.c (278) - Warning: vscphlp_receiveEventEx failed:  A parameter is invalid
2015-04-29 23:45:29 vscp_tp_adapter.c (277) - Warning: Couldn't receive event:  34
2015-04-29 23:45:29 vscp_tp_adapter.c (278) - Warning: vscphlp_receiveEventEx failed:  A parameter is invalid
2015-04-29 23:45:29 vscp_tp_adapter.c (277) - Warning: Couldn't receive event:  34
2015-04-29 23:45:29 vscp_tp_adapter.c (278) - Warning: vscphlp_receiveEventEx failed:  A parameter is invalid
2015-04-29 23:45:30 Tx: class=0x14, type=0x09, prio= 7, oAddr=0x01, -, num=3, data=00 FF FF
```

BlueAndi, 2015-05-01:
I followed the problem now via VscpRemoteTcpIf::doCmdVersion and found out, that
```
if ( m_inputStrArray.Count() < 2 ) return VSCP_ERROR_ERROR;
```
returns with that error.
It may lead into the direction where the root cause is.

BlueAndi, 2015-05-01:
VscpRemoteTcpIf::doCmdReceiveEx fails, because of
```
if ( !getEventFromLine( strLine, pEvent ) ) return VSCP_ERROR_PARAMETER;
```

BlueAndi, 2015-05-05:
Hi Ake, here you see the problem with debug logging:

Good case:
```
[6660] doCmdVersion: VERS 
[6660] 1,0,1  
[6660] +OK - Success.  
[6660] checkReturnValue: Command success! 
[6660] doCmdVersion: getInputQueueCount() 
```

Bad case:
```
[7152] doCmdVersion: VERS 
[7152] +OK - All events cleared.  
[7152] checkReturnValue: Command success! 
[7152] doCmdVersion: getInputQueueCount() 
```

grodansparadis, 2015-05-05:
OK. Hope I can have a look on Friday or in the weekend to see what this 
is.  Thanks for input.

Andreas Merkle skrev den 2015-05-05 21:34:

> Hi Ake, here you see the problem with debug logging:
>
> Good case:
>
> |[6660] doCmdVersion: VERS
> [6660] 1,0,1
> [6660] +OK - Success.
> [6660] checkReturnValue: Command success!
> [6660] doCmdVersion: [...]

BlueAndi, 2015-05-06:
I extended the log, which is more helpful than my limited log before:

Good case:
```
[1724] ============================================================ 
[1724] Connect in progress with server 104.236.235.81:9598 
[1724] ============================================================ 
[1724] Checking server response 
[1724] +OK Welcome to the VSCP daemon.  
[1724] checkReturnValue: Command success! 
[1724] +OK found from server. 
[1724] doCommand: USER admin  
[1724]  
[1724] +OK - Success.  
[1724] checkReturnValue: Command success! 
[1724] Username OK 
[1724] doCommand: PASS secret  
[1724]  
[1724] +OK - User name accepted, password please  
[1724] checkReturnValue: Command success! 
[1724] Password OK 
[1724] Successful log in to VSCP server 
[1724] doCommand: NOOP  
[1724]  
[1724] +OK - Success.  
[1724] checkReturnValue: Command success! 
[1724] doCommand: SFLT 0,0,0,00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00  
[1724]  
[1724] +OK - Success.  
[1724] checkReturnValue: Command success! 
[1724] doCommand: SMSK 0,64512,0,00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00  
[1724]  
[1724] +OK - Success.  
[1724] checkReturnValue: Command success! 
[1724] doCommand: CLRA  
[1724]  
[1724] +OK - Success.  
[1724] checkReturnValue: Command success! 
[1724] doCommand: VERS  
[1724]  
[1724] +OK - All events cleared.  
[1724] checkReturnValue: Command success! 
[1724] clientTcpIpWorkerThread: Starting. 
[1724] clientTcpIpWorkerThread: Before loop. 
[1724] ev_handler: TCP/IP connect OK. 
[1724] ev_handler: TCP/IP receive. 
[1724] +OK Welcome to the VSCP daemon.  
[1724] +OK Version: 1.0.1.7 Neon  
[1724] +OK Copyright (C) 2000-2015, Grodans Paradis AB, http://www.paradiseofthefrog.com  
[1724] +OK - Success.  
[1724]  
[1724] TCP/IP line: +OK Welcome to the VSCP daemon.   
[1724] TCP/IP line: +OK Version: 1.0.1.7 Neon   
[1724] TCP/IP line: +OK Copyright (C) 2000-2015, Grodans Paradis AB, http://www.paradiseofthefrog.com   
[1724] TCP/IP line: +OK - Success.   
[1724] ev_handler: TCP/IP receive. 
[1724] +OK - Success.  
[1724]  
[1724] TCP/IP line: +OK - Success.   
[1724] ev_handler: TCP/IP receive. 
[1724] +OK - User name accepted, password please  
[1724]  
[1724] TCP/IP line: +OK - User name accepted, password please   
[1724] ev_handler: TCP/IP receive. 
[1724] +OK - Success.  
[1724]  
[1724] TCP/IP line: +OK - Success.   
[1724] ev_handler: TCP/IP receive. 
[1724] +OK - Success.  
[1724]  
[... 700 lines cut ...]
[4848] doCommand: CDTA  
[4848]  
[4848] ev_handler: TCP/IP receive. 
[4848] 0  
[4848] +OK - Success.  
[4848]  
[4848] TCP/IP line: 0   
[4848] TCP/IP line: +OK - Success.   
[4848] 0  
[4848] +OK - Success.  
[4848] checkReturnValue: Command success! 
[4848] doCommand: CDTA  
[4848]  
[4848] ev_handler: TCP/IP receive. 
[4848] 0  
[4848] +OK - Success.  
[4848]  
[4848] TCP/IP line: 0   
[4848] TCP/IP line: +OK - Success.   
[4848] 0  
[4848] +OK - Success.  
[4848] checkReturnValue: Command success! 
[4848] ev_handler: TCP/IP close. 
[4848] clientTcpIpWorkerThread: Terminating. 
```

grodansparadis, 2015-05-10:
Thanks. It's a lot of inflow at the moment but I will have a look. 
Hopefully early next week. Yes I know I promised Friday...

Andreas Merkle skrev den 2015-05-05 21:34:

> Hi Ake, here you see the problem with debug logging:
>
> Good case:
>
> |[6660] doCmdVersion: VERS
> [6660] 1,0,1
> [6660] +OK - Success.
> [6660] checkReturnValue: Command success!
> [6660] doCmdVersion: [...]

BlueAndi, 2015-05-10:
No problem Ake, we will get the bug! :-)

grodansparadis, 2015-09-02:
Is this still a problem?

BlueAndi, 2015-09-02:
Yes, it is still a problem.
If you take a look to the debug output above, you will see in the bad case that it looks like the relation between command and responses is lost.

Can I send you my executable to test it at your side?

grodansparadis, 2015-09-02:
Den 2015-09-02 kl. 20:46, skrev Andreas Merkle:

>
> Yes, it is still a problem.
> If you take a look to the debug output above, you will see in the bad 
> case that it looks like the relation between command and responses is 
> lost.
OK
>
> Can I send you my executable to test it at your side?
Yes do that.

BlueAndi, 2015-09-02:
Note, problem exists under linux too.

grodansparadis, 2015-09-02:
OK I see the problem and can repeat it. Will investigate further. It 
seem to happen two of ten times or something here so this should be 
pretty easy to solve.

Den 2015-09-02 kl. 21:27, skrev Andreas Merkle:

>
> Note, problem exists under linux too.

BlueAndi, 2015-09-02:
That sounds good, because this bug is very annoying. :-)

grodansparadis, 2015-09-16:
I have run a works against the demo server and it works 100%. And I have really done MANY tests. But when I run your code it pause (temporarily hangs) after the connection on some of the runs. That is it never hangs for me. Always resumes after a while. I will build some test code and see if I can repeat it. It is strange as the same code is used.

grodansparadis, 2015-09-16:
???

BlueAndi, 2015-09-16:
Tried once and it happened.

The log above shows that the relation between incoming command and corresponding response is lost at the very beginning. Because of the loose coupling it never heals.
At my side I got this error about 50% of my connections.

grodansparadis, 2015-09-16:
I have run it all day without problems since the fix.  Are you sure!!!

Den 2015-09-16 kl. 22:32, skrev Andreas Merkle:

> Tried once and it happened.
> 
> The log above shows that the relation between incoming command and
> corresponding response is lost at the very beginning. Because of the
> loose coupling it never heals.
> At my side I got this error about 50% of my connections.

BlueAndi, 2015-09-16:
Sorry ... but yes ... I called it now several times just some minutes ago. :-(

grodansparadis, 2015-09-16:
Strange I tested again also. and got it.  Probably have run your app.
300 times today without any problems.  OK back to it tomorrow then :(

Den 2015-09-16 kl. 22:36, skrev Andreas Merkle:

> Sorry ... but yes ... I called it now several times just some minutes
> ago. :-(

grodansparadis, 2015-09-16:
I don't know if you have a Linux machine bit if you do could you test
src/vscp/helperlib/linux/examples/example1 from your location a couple
of time.

/Ake

Den 2015-09-16 kl. 22:36, skrev Andreas Merkle:

> Sorry ... but yes ... I called it now several times just some minutes
> ago. :-(

BlueAndi, 2015-09-16:
I will, but tomorrow ... my wife is in my back and I have to leave the computer now. ;-)
Was late back today, because of a long meeting with a customer.

grodansparadis, 2015-09-16:
No problems.

Den 2015-09-16 kl. 22:42, skrev Andreas Merkle:

> I will, but tomorrow ... my wife is in my back and I have to leave the
> computer now. ;-)
> Was late back today, because of a long meeting with a customer.

BlueAndi, 2015-09-17:
Log from linux, which fails too:

```
20:30:05: Debug: initializing
20:30:05: Debug: application constructor
20:30:05: Debug: constructed

VSCP level 1 node
Version: 0.3.0
(c) 2014-2015 Andreas Merkle

Keys:
?       Show this key table
e       Dump EEPROM
h       Enable/Disable node heartbeat
i       Start node segment initialization
l       Show the state of all lamps
q       Quit program
1-5     Send info button event

2015-09-17 20:30:05 eeprom.c (171) - Info: EEPROM file found.
2015-09-17 20:30:05 eeprom.c (251) - Info: EEPROM file loaded.
2015-09-17 20:30:05 Lamp blinking slow
2015-09-17 20:30:05 Connecting ...
20:30:05: Debug: ============================================================
20:30:05: Debug: Connect in progress with server 104.236.235.81:9598
20:30:05: Debug: ============================================================
20:30:05: Debug: Checking server response
20:30:05: Debug: Welcome to the VSCP daemon.
20:30:05: Debug: Version: 1.0.1.52 Neon
20:30:05: Debug: Copyright (C) 2000-2015, Paradise of the Frog AB, http://www.paradiseofthefrog.com
20:30:05: Debug: +OK - Success.
20:30:05: Debug: checkReturnValue: Command success!
20:30:05: Debug: +OK found from server.
20:30:05: Debug: doCommand: USER admin

20:30:06: Debug: +OK - Success.
20:30:06: Debug: checkReturnValue: Command success!
20:30:06: Debug: Username OK
20:30:06: Debug: doCommand: PASS secret

20:30:06: Debug: +OK - User name accepted, password please
20:30:06: Debug: checkReturnValue: Command success!
20:30:06: Debug: Password OK
20:30:06: Debug: Successful log in to VSCP server
20:30:06: Debug: doCommand: NOOP

20:30:06: Debug: +OK - Success.
20:30:06: Debug: checkReturnValue: Command success!
20:30:06: Debug: doCommand: SFLT 0,0,0,00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00

20:30:06: Debug: +OK - Success.
20:30:06: Debug: checkReturnValue: Command success!
20:30:06: Debug: doCommand: SMSK 0,64512,0,00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00

20:30:06: Debug: +OK - Success.
20:30:06: Debug: checkReturnValue: Command success!
20:30:06: Debug: doCommand: CLRA

20:30:07: Debug: +OK - Success.
20:30:07: Debug: checkReturnValue: Command success!
2015-09-17 20:30:07 vscp_tp_adapter.c (542) - Info: Connected.
20:30:07: Debug: doCommand: VERS

20:30:07: Debug: +OK - All events cleared.
20:30:07: Debug: checkReturnValue: Command success!
2015-09-17 20:30:07 vscp_tp_adapter.c (547) - Warning: vscphlp_getVersion failed: -1
2015-09-17 20:30:07 Used VSCP dll version 0x00000004
2015-09-17 20:30:07 Vendor of driver: Paradise of the Frog AB, Sweden
2015-09-17 20:30:07 Connection successful.
[... 394 lines cut ...]
20:30:13: Debug: doCommand: CDTA

20:30:13: Debug: ev_handler: TCP/IP receive.
20:30:13: Debug: 0
+OK - Success.

 0:30:13: Debug: TCP/IP line: 0
 0:30:13: Debug: TCP/IP line: +OK - Success.
20:30:13: Debug: 0
20:30:13: Debug: +OK - Success.
20:30:13: Debug: checkReturnValue: Command success!
20:30:13: Debug: doCommand: CDTA

20:30:13: Debug: ev_handler: TCP/IP receive.
20:30:13: Debug: 0
+OK - Success.

 0:30:13: Debug: TCP/IP line: 0
 0:30:13: Debug: TCP/IP line: +OK - Success.
20:30:13: Debug: 0
20:30:13: Debug: +OK - Success.
20:30:13: Debug: checkReturnValue: Command success!
20:30:14: Debug: ev_handler: TCP/IP close.
20:30:14: Debug: clientTcpIpWorkerThread: Terminating.
```

grodansparadis, 2015-09-17:
I messed up the changes as it looks. Test again tomorrow.

Den 2015-09-17 kl. 20:35, skrev Andreas Merkle:

> Log from linux, which fails too:
>
> |20:30:05: Debug: initializing 20:30:05: Debug: application constructor
> 20:30:05: Debug: constructed VSCP level 1 node Version: 0.3.0 (c)
> 2014-2015 Andreas Merkle [...]

BlueAndi, 2015-09-17:
The following line should not be there, I guess. Maybe that helps ...

```
20:30:05: Debug: doCommand: USER admin

20:30:06: Debug: +OK - Success.
20:30:06: Debug: checkReturnValue: Command success!
20:30:06: Debug: Username OK
20:30:06: Debug: doCommand: PASS secret

20:30:06: Debug: +OK - User name accepted, password please <---------
20:30:06: Debug: checkReturnValue: Command success!
20:30:06: Debug: Password OK
```

grodansparadis, 2015-09-17:
When you get some time can you check again?

grodansparadis, 2015-09-17:
Increase the vscphlp_setAfterCommandSleep ro 100 or 200 milliseconds if you still have problems. That will add that amount to the command wait.  200 works fine here but on VSCP works I will use 200

BlueAndi, 2015-09-18:
I will check it, but first I have to recompile the helper dll. I fear timeouts, because they are per default wrong. Is there any other possibility to ensure that the server delivered the data?

The bad thing is ... I need to write something like a connection test to be able to automatically determine the timeout values. Otherwise it will work for me, but maybe not for others. Have to think about that ...

grodansparadis, 2015-09-18:
As soon as you enter a timeout of any sort you enters that state. I 
tried not to. But when there yo have to find a good value and with a 
configuration possibility you are often fine.

But never mind with it at all. I am rather pissed of with your attitude 
Andy. I would be happy if you left this project and go somewhere else.

Thanks for all the contributions.

Den 2015-09-18 kl. 20:42, skrev Andreas Merkle:

> I will check it, but first I have to recompile the helper dll. I fear
> timeouts, because they are per default wrong. Is there any other
> possibility to ensure that the server delivered the data?
>
> The [...]

BlueAndi, 2015-09-18:
Ake? Sorry? Did I something wrong?

BlueAndi, 2015-09-18:
I am sitting here now for an hour and your statement makes me sad, because I thought our relationship was good and I am aware of no guilt. I hope I didn't write something stupid in english, as it is not my native language.

But as you said, it is not a democratic project and I have to respect that. Please remove me as a collaborator, I will stop any kind of contributions and any kind of stupid questions to you.

grodansparadis, 2015-09-18:
You walk your own way anyway so just continue with that.

I have removed you from the projects and will switch back the HTML5 project.

Den 2015-09-18 kl. 22:08, skrev Andreas Merkle:

> I am sitting here now for an hour and your statement makes me sad,
> because I thought our relationship was good and I am aware of no guilt.
> I hope I didn't write something stupid in english, as it is [...]

commit 6fbbe1ac (2015-09-16): Fixed #128

Open again if not
closed 2015-09-18 by grodansparadis