grodansparadis/vscp #131: vscpworks.exe is not a valid Win32 application (vscp-1.0.1.2-x86.exe)

waaslandwolf, 2015-05-04:
After installing vscp-1.0.1.2-x86.exe, when running vscpworks.exe, I get an error "c:\Program Files\VSCP\vscpworks.exe is not a valid Win32 application."

Did a proper uninstall of the previous version + deleted the "c:\Program Files\VSCP" manually before installing the new version.

grodansparadis, 2015-05-04:
OS?

Works OK here. Could maybe be a mismatch of 32/64 bit dll's. I will 
check that. But don't have time tight now.

waaslandwolf skrev den 2015-05-04 20:14:

>
> After installing vscp-1.0.1.2-x86.exe, when running vscpworks.exe, I 
> get an error "c:\Program Files\VSCP\vscpworks.exe is not a valid Win32 
> application."
>
> Did a proper uninstall of the previous version + deleted the 
> "c:\Program Files\VSCP" manually before installing the new version.

waaslandwolf, 2015-05-04:
OS is Windows XP SP3 with latest updates.

I assume Win development isn't high on your priority list. I don't 
really need the new win version, if I can get 1.0.0.43 to connect to the 
daemon then that's fine as well.

David.

---
This email has been checked for viruses by Avast antivirus software.
http://www.avast.com

grodansparadis, 2015-05-04:
waaslandwolf skrev den 2015-05-04 20:36:
> OS is Windows XP SP3 with latest updates.
OK   XP is not supported any longer but if you could run 1.0.0 then it 
probably should work.

>
> I assume Win development isn't high on your priority list. I don't
> really need the new win version, if I can get 1.0.0.43 to connect to the
> daemon then that's fine as well.
1.0.0 will never do that.

/Ake

-- 
Be Hungry - Stay Foolish!
Åke Hedman, Grodans Paradis AB / Paradise of the Frog
Brattbergavägen 17, 82050 LOS, SWEDEN
Phone: +46 (0)840011835 CellPhone: +46 (0)730-533146
http://www.grodansparadis.com http://www.vscp.org

---
Detta e-postmeddelande har sökts igenom efter virus med antivirusprogram från Avast.
http://www.avast.com

closed 2015-09-02 by grodansparadis