grodansparadis/vscp #136: "Failed to update data" when writing register

waaslandwolf, 2015-05-08:
Context:
- VSCP 1.0.1.12 on Ubuntu 14.04 LTS
- VSCP daemon running locally, VSCPworks running locally
- USB2CAN converter to a bus with 21 Mespelare v9 modules

History:
- ip link set can0 up
- /etc/init.d/vscpd start
- /usr/local/bin/vscpworks & 
- Start VSCP client window, connect to daemon
- Start VSCP device configuration window, connect to daemon

Known good baseline:
- read registers for device 0x0D, read successful
- change value for register 0x27, click Update, write successful, functionality as expected (shows register write worked)

Bug:
- wait half an hour or so
- change value for register 0x27, click Update. Lots of debug messages in the terminal window where VSCPworks was started, "Failed to update data" message window. 
- kill -9 [vscpd and vscpworks], daemon restart, vscpworks restart, register now reads and writes correctly.
- When testing functionality or re-reading the node's registers, it is clear that the data was written to the node correctly, but VSCPworks still threw an error.

grodansparadis, 2015-05-10:
Ok will check

waaslandwolf skrev den 2015-05-08 21:13:

> Context:
>
>   * VSCP 1.0.1.12 on Ubuntu 14.04 LTS
>   * VSCP daemon running locally, VSCPworks running locally
>   * USB2CAN converter to a bus with 21 Mespelare v9 modules
>
> History:
>
>   * ip link set can0 [...]

grodansparadis, 2015-09-16:
I have not forgotten about this one. Just haven't found it. Sorry for the delay.  Bit I can acknowledge the behavior.

commit bc6cbb3b (2015-09-28): Fixes #136  Open again if not.
closed 2015-09-28 by grodansparadis