grodansparadis/vscp #146: Issue with Scan function?

waaslandwolf, 2015-06-01:
VSCPworks 1.0.1.21 Neon on Ubuntu LTS14.0

- daemon already running for several days/weeks
- open _VSCP Client window_, select interface "_TCPIP: localhost_"
- client window opens, can see events on the bus

- Open "_Scan for device..._" window
- Select interface "_TCP/IP: localhost_"
- Scan window opens
- Click "_Scan_" button, _scan from_ 1 _scan to_ 255 was already filled in
- Scan window opens window "_Scanning for VSCP devices, reading registers_"
- VSCPworks main window throws error window "_No interface specified. Please select one OK_" Clicking _OK_ closes 2 above windows and returns to scan window. No devices found.

grodansparadis, 2015-06-01:
Yes you must select an interface for this to work. Yes I know the naming is confusing, But on the daemon the interfaces is each driver, client etc connected. This is because a nickname have no meaning at this level. Only GUID's have. And it would take a lifetime to scan for the.m

closed 2015-06-01 by grodansparadis

waaslandwolf, 2015-06-02:
Correct - because TCP/IP: localhost I assumed that it would have an interface assigned, which wasn't the case. Clicked Edit and selected level2socketcan driver, now all is well.

grodansparadis, 2015-06-02:
Great.

waaslandwolf skrev den 2015-06-02 13:43:

>
> Correct - because TCP/IP: localhost I assumed that it would have an 
> interface assigned, which wasn't the case. Clicked Edit and selected 
> level2socketcan driver, now all is well.