grodansparadis/vscp #147: XML error message when opening works

mamac, 2015-06-05:
How to repeate:
- Running Linux
- Open WORKS and open the device configuration window
- search for a node and load all the registers
- close WORKS
- re-open WORKS, then you got the following error message:
[image]

The xml file looks as above:

```
<?xml version = "1.0" encoding = "UTF-8" ?>
<vscpworksconfig>

<general>
<width>640</width>
<height>311</height>
<xpos>0</xpos>
<ypos>0</ypos>
<path2tempfile>/tmp</path2tempfile>
<path2logfile enable="false" level="debug" >/tmp\vscpworks.log</path2logfile>
<CANALReadMaxRetries>3</CANALReadMaxRetries>
<CANALReadResendTimeout>1000</CANALReadResendTimeout>
<CANALReadErrorTimeout>5000</CANALReadErrorTimeout>
<TCPIPResponseTimeout>3</TCPIPResponseTimeout>
<TCPIPReadMaxRetries>3</TCPIPReadMaxRetries>
<TCPIPReadResendTimeout>2000</TCPIPReadResendTimeout>
<TCPIPReadErrorTimeout>6000</TCPIPReadErrorTimeout>
<NumberBase>Hex</NumberBase>
<ConfirmDelete>true</ConfirmDelete>
<GuidWriteEnable>do-not-use</GuidWriteEnable>
<ManufacturerGuid>00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00</ManufacturerGuid>
<ManufacturerId>0x00000000</ManufacturerId>
<ManufacturerSubId>0x00000000</ManufacturerSubId>
</general>

<mdfurl>
<translate from="/home/mamac/Electronic/v" to="" />
</mdfurl>

<vscpclient>
<Autoscroll>true</Autoscroll>
<VscpRcvPyjamasStyle>true</VscpRcvPyjamasStyle>
<VscpUseSymbolicNames>true</VscpUseSymbolicNames>
<VscpRcvRowLineHeight>20</VscpRcvRowLineHeight>
<VscpRcvShowField0  enable="true" ></VscpRcvShowField0>
<VscpRcvShowField1  enable="true" ></VscpRcvShowField1>
<VscpRcvShowField2  enable="true" ></VscpRcvShowField2>
<VscpRcvShowField3  enable="true" ></VscpRcvShowField3>
<VscpRcvFieldOrder0>0</VscpRcvFieldOrder0>
<VscpRcvFieldOrder1>1</VscpRcvFieldOrder1>
<VscpRcvFieldOrder2>2</VscpRcvFieldOrder2>
<VscpRcvFieldOrder3>3</VscpRcvFieldOrder3>
<VscpRcvFieldText0>Dir</VscpRcvFieldText0>
<VscpRcvFieldText1>Class</VscpRcvFieldText1>
<VscpRcvFieldText2>Type</VscpRcvFieldText2>
<VscpRcvFieldText3>Note</VscpRcvFieldText3>
<VscpRcvFieldWidth0>30</VscpRcvFieldWidth0>
<VscpRcvFieldWidth1>170</VscpRcvFieldWidth1>
<VscpRcvFieldWidth2>170</VscpRcvFieldWidth2>
<VscpRcvFieldWidth3>700</VscpRcvFieldWidth3>
<VscpTrmitShowField0  enable="true" ></VscpTrmitShowField0>
<VscpTrmitShowField1  enable="true" ></VscpTrmitShowField1>
<VscpTrmitShowField2  enable="true" ></VscpTrmitShowField2>
<VscpTrmitShowField3  enable="true" ></VscpTrmitShowField3>
<VscpTrmitShowField4  enable="true" ></VscpTrmitShowField4>
<VscpTrmitShowField5  enable="true" ></VscpTrmitShowField5>
<VscpTrmitFieldOrder0>0</VscpTrmitFieldOrder0>
<VscpTrmitFieldOrder1>1</VscpTrmitFieldOrder1>
<VscpTrmitFieldOrder2>2</VscpTrmitFieldOrder2>
<VscpTrmitFieldOrder3>3</VscpTrmitFieldOrder3>
<VscpTrmitFieldOrder4>4</VscpTrmitFieldOrder4>
<VscpTrmitFieldOrder5>30</VscpTrmitFieldOrder5>
<VscpTrmitFieldText0>x</VscpTrmitFieldText0>
<VscpTrmitFieldText1>Name</VscpTrmitFieldText1>
<VscpTrmitFieldText2>Period</VscpTrmitFieldText2>
<VscpTrmitFieldText3>Count</VscpTrmitFieldText3>
<VscpTrmitFieldText4>Trigger</VscpTrmitFieldText4>
<VscpTrmitFieldText5>Event</VscpTrmitFieldText5>
<VscpTrmitFieldWidth0>30</VscpTrmitFieldWidth0>
<VscpTrmitFieldWidth1>200</VscpTrmitFieldWidth1>
<VscpTrmitFieldWidth2>60</VscpTrmitFieldWidth2>
<VscpTrmitFieldWidth3>60</VscpTrmitFieldWidth3>
<VscpTrmitFieldWidth4>120</VscpTrmitFieldWidth4>
<VscpTrmitFieldWidth5>900</VscpTrmitFieldWidth5>
<VscpRcvFrameRxTextColour  r="0" g="0" b="128" ></VscpRcvFrameRxTextColour>
<VscpRcvFrameRxBgColour  r="255" g="255" b="255" ></VscpRcvFrameRxBgColour>
<VscpRcvFrameTxTextColour  r="128" g="128" b="128" ></VscpRcvFrameTxTextColour>
<VscpRcvFrameTxBgColour  r="255" g="255" b="255" ></VscpRcvFrameTxBgColour>
<VscpRcvFrameLineColour  r="0" g="0" b="0" ></VscpRcvFrameLineColour>
</vscpclient>

<vscpinterface>
<vscpdaemon>
<description>lljm</description>
<host>192.168.1.2:9598</host>
<username>admin</username>
<password>secret</password>
<interfacename></interfacename>
<level2>false</level2>
<guid>10:76:1A:01:00:00:00:00:00:00:00:00:00:00:00:00</guid>
<filter  priority="0"  class="0"  type="0"  GUID="00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00"></filter>
<mask  priority="0"  class="0"  type="0"  GUID="00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00"></mask>
</vscpdaemon>

</vscpinterface>

</vscpworksconfig>
```
Thank you,
JM

grodansparadis, 2015-06-06:
Hmmm strange. Looks OK. I will check on Monday. Off this weekend (sort 
of...)

mamac skrev den 2015-06-05 23:57:

>
> How to repeate:
>
>   * Running Linux
>   * Open WORKS and open the device configuration window
>   * search for a node and load all the registers
>   * close WORKS
>   * re-open WORKS, then you got the following error message: image
>     <https://cloud.githubusercontent.com/assets/7148802/8015941/b888be3a-0bdd-11e5-8298-a4d1d88cd5d2.png>
>
>
> The xml file looks as above:
>
> |<?xml version = "1.0" encoding = "UTF-8" ?>
> <vscpworksconfig>
>
> <general>
> <width>640</width>
> <height>311</height>
> <xpos>0</xpos>
> <ypos>0</ypos>
> <path2tempfile>/tmp</path2tempfile>
> <path2logfile enable="false" level="debug" >/tmp\vscpworks.log</path2logfile>
> <CANALReadMaxRetries>3</CANALReadMaxRetries>
> <CANALReadResendTimeout>1000</CANALReadResendTimeout>
> <CANALReadErrorTimeout>5000</CANALReadErrorTimeout>
> <TCPIPResponseTimeout>3</TCPIPResponseTimeout>
> <TCPIPReadMaxRetries>3</TCPIPReadMaxRetries>
> <TCPIPReadResendTimeout>2000</TCPIPReadResendTimeout>
> <TCPIPReadErrorTimeout>6000</TCPIPReadErrorTimeout>
> <NumberBase>Hex</NumberBase>
> <ConfirmDelete>true</ConfirmDelete>
> <GuidWriteEnable>do-not-use</GuidWriteEnable>
> <ManufacturerGuid>00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00</ManufacturerGuid>
> <ManufacturerId>0x00000000</ManufacturerId>
> <ManufacturerSubId>0x00000000</ManufacturerSubId>
> </general>
>
> <mdfurl>
> <translate from="/home/mamac/Electronic/v" to="" />
> </mdfurl>
>
> <vscpclient>
> <Autoscroll>true</Autoscroll>
> <VscpRcvPyjamasStyle>true</VscpRcvPyjamasStyle>
> <VscpUseSymbolicNames>true</VscpUseSymbolicNames>
> <VscpRcvRowLineHeight>20</VscpRcvRowLineHeight>
> <VscpRcvShowField0  enable="true" ></VscpRcvShowField0>
> <VscpRcvShowField1  enable="true" ></VscpRcvShowField1>
> <VscpRcvShowField2  enable="true" ></VscpRcvShowField2>
> <VscpRcvShowField3  enable="true" ></VscpRcvShowField3>
> <VscpRcvFieldOrder0>0</VscpRcvFieldOrder0>
> <VscpRcvFieldOrder1>1</VscpRcvFieldOrder1>
> <VscpRcvFieldOrder2>2</VscpRcvFieldOrder2>
> <VscpRcvFieldOrder3>3</VscpRcvFieldOrder3>
> <VscpRcvFieldText0>Dir</VscpRcvFieldText0>
> <VscpRcvFieldText1>Class</VscpRcvFieldText1>
> <VscpRcvFieldText2>Type</VscpRcvFieldText2>
> <VscpRcvFieldText3>Note</VscpRcvFieldText3>
> <VscpRcvFieldWidth0>30</VscpRcvFieldWidth0>
> <VscpRcvFieldWidth1>170</VscpRcvFieldWidth1>
> <VscpRcvFieldWidth2>170</VscpRcvFieldWidth2>
> <VscpRcvFieldWidth3>700</VscpRcvFieldWidth3>
> <VscpTrmitShowField0  enable="true" ></VscpTrmitShowField0>
> <VscpTrmitShowField1  enable="true" ></VscpTrmitShowField1>
> <VscpTrmitShowField2  enable="true" ></VscpTrmitShowField2>
> <VscpTrmitShowField3  enable="true" ></VscpTrmitShowField3>
> <VscpTrmitShowField4  enable="true" ></VscpTrmitShowField4>
> <VscpTrmitShowField5  enable="true" ></VscpTrmitShowField5>
> <VscpTrmitFieldOrder0>0</VscpTrmitFieldOrder0>
> <VscpTrmitFieldOrder1>1</VscpTrmitFieldOrder1>
> <VscpTrmitFieldOrder2>2</VscpTrmitFieldOrder2>
> <VscpTrmitFieldOrder3>3</VscpTrmitFieldOrder3>
> <VscpTrmitFieldOrder4>4</VscpTrmitFieldOrder4>
> <VscpTrmitFieldOrder5>30</VscpTrmitFieldOrder5>
> <VscpTrmitFieldText0>x</VscpTrmitFieldText0>
> <VscpTrmitFieldText1>Name</VscpTrmitFieldText1>
> <VscpTrmitFieldText2>Period</VscpTrmitFieldText2>
> <VscpTrmitFieldText3>Count</VscpTrmitFieldText3>
> <VscpTrmitFieldText4>Trigger</VscpTrmitFieldText4>
> <VscpTrmitFieldText5>Event</VscpTrmitFieldText5>
> <VscpTrmitFieldWidth0>30</VscpTrmitFieldWidth0>
> <VscpTrmitFieldWidth1>200</VscpTrmitFieldWidth1>
> <VscpTrmitFieldWidth2>60</VscpTrmitFieldWidth2>
> <VscpTrmitFieldWidth3>60</VscpTrmitFieldWidth3>
> <VscpTrmitFieldWidth4>120</VscpTrmitFieldWidth4>
> <VscpTrmitFieldWidth5>900</VscpTrmitFieldWidth5>
> <VscpRcvFrameRxTextColour  r="0" g="0" b="128" ></VscpRcvFrameRxTextColour>
> <VscpRcvFrameRxBgColour  r="255" g="255" b="255" ></VscpRcvFrameRxBgColour>
> <VscpRcvFrameTxTextColour  r="128" g="128" b="128" ></VscpRcvFrameTxTextColour>
> <VscpRcvFrameTxBgColour  r="255" g="255" b="255" ></VscpRcvFrameTxBgColour>
> <VscpRcvFrameLineColour  r="0" g="0" b="0" ></VscpRcvFrameLineColour>
> </vscpclient>
>
> <vscpinterface>
> <vscpdaemon>
> <description>lljm</description>
> <host>192.168.1.2:9598</host>
> <username>admin</username>
> <password>secret</password>
> <interfacename></interfacename>
> <level2>false</level2>
> <guid>10:76:1A:01:00:00:00:00:00:00:00:00:00:00:00:00</guid>
> <filter  priority="0"  class="0"  type="0"  GUID="00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00"></filter>
> <mask  priority="0"  class="0"  type="0"  GUID="00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00"></mask>
> </vscpdaemon>
>
> </vscpinterface>
>
> </vscpworksconfig>
> |
>
> Thank you,
> JM

BlueAndi, 2015-09-01:
Hi JM,
can you try it again, but this time replace the
`<translate from="/home/mamac/Electronic/v" to="" />`
to
`<translate from="/home/mamac/Electronic/v" to=""></translate>`
?

Maybe the parser has a problem with that kind of tags.

mamac, 2015-09-01:
Hi Andi

The suggested fix works.

However I reinstalled my computer since I created this issue and therefore installed the latest version of WORKS.

Here is what I have now:
`
<mdfurl>
<translate from="/^@^@^@h^@^@^@o^@^@^@m^@^@^@e^@^@^@/^@^@^@m^@^@^@a^@^@^@m^@^@^@a^@^@^@c^@^@^@/^@^@^@E^@^@^@l^@^@^@e^@^@^@c^@^@^@t^@^@^@r^@^@^@o^@^@^@n^@^@^@i^@^@^@c^@^@^@/^@^@^@v" to="" />
</mdfurl>
`

Thank you,
JM

grodansparadis, 2015-09-01:
2.8.12 ?

mamac, 2015-09-02:
WORKS 1.0.1.30
Wxwidgets 2.8.12

mamac, 2015-09-04:
This is not a show stopper, it just resets the configuration everytime I use the configuraiton window, so I have to re-enter the connection détails but it's not a big deal, this is a low priority.

grodansparadis, 2015-09-04:
I will take a look (and fix) next week.

Den 2015-09-04 kl. 08:46, skrev mamac:

>
> This is not a show stopper, it just resets the configuration everytime 
> I use the configuraiton window, so I have to re-enter the connection 
> détails but it's not a big deal, this is a low priority.

commit f0030438 (2015-09-14): Fixed #147. Open again please if still a problem.
closed 2015-09-14 by grodansparadis