grodansparadis/vscp #148: Illegal operation for vscpd

nos86, 2015-06-12:
Hi Akhe,

I tried to compile and install the vscpd on raspberry without success, unfortunately

I followed all operation described in http://78.20.42.79:8080/posts/Installing%20VSCP%20on%20a%20Raspberry%20Pi/

and during compilation, I have no error and no warning but as soon as I try to start the daemon following message appears: 
sudo /etc/init.d/vscpd start
[....] Starting VSCP daemon: vscpdIllegal instruction
 failed!

Launching daemon with GDB, error seems related to libwx3.0:
Reading symbols from /usr/local/bin/vscpd...done.
(gdb) run
Starting program: /usr/local/bin/vscpd
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".

Program received signal SIGILL, Illegal instruction.
0xb6b773d0 in call_gmon_start () from /usr/lib/arm-linux-gnueabihf/libwx_baseu-3.0.so.0

Can you help me to understand what is happening?
I updated all package to last version, but nothing is changeda

grodansparadis, 2015-06-12:
Strange. Do you get the same error if you start the daemon with vscpd -s from the command line?

grodansparadis, 2015-06-12:
In an other instance I tried a daemon with head code and jade some problems with a driver. Can you try this also and disable all drivers. /etc/vscp/vscpd.conf   Change all enable="true" to enable="false" for drivers.

nos86, 2015-06-12:
I tried to disable all drivers (level1 and 2) but I still have the same problem.. Should be something with library wx3.0? I checked and it is installed

Cheers

grodansparadis, 2015-06-12:
2.8.12 should be available as a package also as an option.

Salvo Musumeci skrev den 2015-06-12 14:43:

>
> I tried to disable all drivers (level1 and 2) but I still have the 
> same problem.. Should be something with library wx3.0? I checked and 
> it is installed
>
> Cheers

nos86, 2015-06-15:
Hello Akhe,

finally, I fixed the issue: following the Steeman's guide, I downloaded the wxWidget 3.0 from a Debian Distribution. 
As soon as I removed it and compiled the source, vscpd started to work.
Now only alert I receive is:
../src/common/socket.cpp(295): assert "app" failed in Init(): sockets can't be initialized without wxApp
00:41:11: Debug: ControlObject: Configfile =/etc/vscp/vscpd.conf

but, I can confirm you that it is, just, a warning; telnet, web-server and remote vscpWorks are working correctly.

Thanks for the support and for all work you did (and are doing) on this project.

Salvo

closed 2015-06-15 by nos86

grodansparadis, 2015-06-16:
Sorry for a late reply my mail has been down.

Ok that warning I have on my list. Good you are up an running. Thanks 
for the encouragement.

/Ake

Salvo Musumeci skrev den 2015-06-15 02:42:

> Hello Akhe,
>
> finally, I fixed the issue: following the Steeman's guide, I
> downloaded the wxWidget 3.0 from a Debian Distribution.
> As soon as I removed it and compiled the source, vscpd started to [...]