grodansparadis/vscp #150: feature request: read and display strings in vscpworks dev.conf.window abstraction view

datenheim, 2015-06-15:
It would be very straight forward to have works able to read and present strings also in the device configuration window view - just as it already does with other values.
[image]

grodansparadis, 2015-06-16:
I thought that was working. Will have a look.

Stefan Langer skrev den 2015-06-15 09:59:

>
> It would be very straight forward to have works able to read and 
> present strings also in the device configuration window view - just as 
> it already does with other values.
> devcfgabstr 
> <https://cloud.githubusercontent.com/assets/101634/8155492/1cb995be-1345-11e5-9ca7-4b3ed216e94b.jpg>

grodansparadis, 2019-03-25:
VSCP Works is deprecated

closed 2019-03-25 by grodansparadis