grodansparadis/vscp #151: device config window - make lines editable / writing from abstraction view

datenheim, 2015-06-15:
It would be a very useful feature to enable writing abstracted values the same manner as it works in the register view of the device configuration window.

Clicking on a value in the abstractions already opens an input window, but entered value is:
* ignored totally for floats and strings
*  sometimes reacts when entering value for Unsigned 8-bit Integers but then throws an error (could not write register)

While writing I re-tested and after editing any type of value in abstraction view they are ignored and no reaction at all happens. Neither restarting works nor power cycling the node can change that. Unluckily I can not provide the error message right now.

If there is any code interpreting any values in abstraction view it is either incomplete or has some bugs...

[vscpworks 1.0.0.43, usb2can, AVR-CAN-node based on digiprobe project]

grodansparadis, 2015-06-16:
Agree and noted.   Abstractions should be easier to use than registers 
but today is the opposite way around in VSCP works. But I expect that to 
change when wizards are in place. Be patient.

/Ake

Stefan Langer skrev den 2015-06-15 10:23:

> It would be a very useful feature to enable writing abstracted values
> the same manner as it works in the register view of the device
> configuration window.
>
> Clicking on a value in the abstractions [...]

grodansparadis, 2019-03-25:
VSCP works deprecated

closed 2019-03-25 by grodansparadis