grodansparadis/vscp #155: Daemon respond with websocket authentication AUTH0 twice

BlueAndi, 2015-06-21:
15:10:42.222 Cmd: C;CHALLENGE
15:10:42.226 Rsp: +;AUTH0;124971ae8a2df676a1379238cf0842a8
15:10:42.232 Cmd: C;AUTH;e61ff4fca8f0a23a52374de21cf2f0b5
15:10:42.331 Rsp: +;AUTH0;124971ae8a2df676a1379238cf0842a8
15:10:42.631 Rsp: -;7;Not authorized

If the second AUTH0 is not answered with a "AUTH" command by the client, the authentication aborts with an negative reply.

grodansparadis, 2015-06-22:
OK noted. Will check.

grodansparadis, 2015-09-16:
This is correct as "CHALLANGE" is sent. If "CHALLANGE" is not sent one should be received.

closed 2015-09-16 by grodansparadis

BlueAndi, 2015-09-17:
Sry ... I am standing on the line ;-) ...
I see two commands and three responses ... don't get it why there are more responses, than commands?

grodansparadis, 2015-09-17:
The "CHALLANGE" command is not needed. It just trigger the same response
that is sent out on a connect.

Den 2015-09-17 kl. 20:39, skrev Andreas Merkle:

> Sry ... I am standing on the line ;-) ...
> I see two commands and three responses ... don't get it why there are
> more responses, than commands?

BlueAndi, 2015-09-17:
Ahh ... ok ... now I got it. Thanks for explanation.