grodansparadis/vscp #159: Websocket RESETVAL is missing in the specification

BlueAndi, 2015-06-22:
*LOL*

grodansparadis, 2015-06-23:
Fixed. writevar and restevar now return type and value.

BlueAndi, 2015-06-23:
Great Ake, thanks. I will update the library accordingly. Did you update the demoserver too? I use it at the moment for testing purposes. And btw. the demosever is REALLY helpful!!!! Like it a lot. And my PC nodes now makes a lot more sense, as I connect them there too.

grodansparadis, 2015-06-23:
Hi,

yes I updated the demoserver also. Not tested anything though so let me know if something went wrong. Well I know you will!   Good that it comes to use.

BlueAndi, 2015-06-24:
Hi Ake,
my tests shows the following:
20:35:23.383 Cmd: C;RESETVAR;DEMOVARIABLE
20:35:23.500 Rsp: +;RESETVAR;DEMOVARIABLE;2;false
20:35:32.408 Cmd: C;WRITEVAR;DEMOVARIABLE;true
20:35:32.524 Rsp: +;WRITEVAR

RESETVAR is fine, but WRITEVAR responses nothing. Can you check please?

grodansparadis, 2015-06-24:
Should be fixed now. Updated on demo server.

BlueAndi, 2015-06-24:
Not all ;-)

21:01:35.485 Cmd: C;WRITEVAR;DEMOVARIABLE;false
21:01:35.601 Rsp: +;WRITEVAR;false;2;false <--- Fine

21:01:41.719 Cmd: C;WRITEVAR;DEMOVARIABLE;asdfsadfasdf
21:01:41.835 Rsp: +;WRITEVAR;asdfsadfasdf;2;false <---- Wrong :-)

21:01:55.894 Cmd: C;READVAR;DEMOVARIABLE
21:01:56.130 Rsp: +;READVAR;DEMOVARIABLE;2;false

grodansparadis, 2015-06-24:
This is actually correct as "false" is the default. You will see the same for a numerical variable if you try to set it to a string it will be zero.  This is by design as I said but also feel like this maybe should be more strict when I see it now. But in the meanwhile...  I have tons of other stuff at the moment I must finish before a two week vacation. So this have to be later.

BlueAndi, 2015-06-24:
But "false" in this case is the persistency, take a look to the first parameter after WRITEVAR ... this is the value, which is "asdfsadfasdf".

BlueAndi, 2015-06-24:
Another example shows it again different.
Precondition, the variable is false and persistency is false.

Now set value to "true".
21:21:18.400 Cmd: C;WRITEVAR;DEMOVARIABLE;true
21:21:18.522 Rsp: +;WRITEVAR;true;2;true

Returns now true for the type too ... hmmm ...

Now set a invalid value:
21:21:25.801 Cmd: C;WRITEVAR;DEMOVARIABLE;asfsafdsadf
21:21:25.923 Rsp: +;WRITEVAR;asfsafdsadf;2;false

Returns the wrong value, the type and looks like the new value. Ahhhh ..... the value is returned now in the way that I see the written and the new value. Ok ... I understand ... that means the current specification is wrong ... or the other way around.

BlueAndi, 2015-06-24:
And ... take your time ... there is no need to fix that now.

grodansparadis, 2015-06-24:
A variable was overwritten.

Andreas Merkle skrev den 2015-06-24 21:16:

>
> But "false" in this case is the persistency, take a look to the first 
> parameter after WRITEVAR ... this is the value, which is "asdfsadfasdf".

grodansparadis, 2015-06-24:
I am building on the demo server right now but must attend to other things now. Hopefully it works.

BlueAndi, 2015-06-24:
Thanks and as I said ... don't hurry ... no problem!

grodansparadis, 2015-06-24:
Built now.

Andreas Merkle skrev den 2015-06-24 21:26:

> Another example shows it again different.
> Precondition, the variable is false and persistency is false.
>
> Now set value to "true".
> 21:21:18.400 Cmd: C;WRITEVAR;DEMOVARIABLE;true
> 21:21:18.522 Rsp: [...]

BlueAndi, 2015-06-24:
Great, works as expected!!

21:39:24.293 Cmd: C;WRITEVAR;DEMOVARIABLE;true
21:39:24.412 Rsp: +;WRITEVAR;DEMOVARIABLE;2;true

21:39:35.320 Cmd: C;WRITEVAR;DEMOVARIABLE;asdfasdf
21:39:35.152 Rsp: +;WRITEVAR;DEMOVARIABLE;2;false

closed 2015-06-24 by BlueAndi