grodansparadis/vscp #160: Websocket RESETVAL returns no value back

BlueAndi, 2015-06-22:
It would be good to receive the value with the positive reply back. Otherwise after the variable is reset, you have to read it back again.

grodansparadis, 2015-06-22:
In the rest case the default value is at least known. If a positive reply the variable have the default value. Listed here http://www.vscp.org/docs/vscpd/doku.php?id=vscp_daemon_variable_string_write_format

closed 2015-06-22 by grodansparadis

BlueAndi, 2015-06-22:
Nevertheless getting the reset value back would be better. If all "known" reset values are implemented hardcoded, the sourcecode will always be false, because the specification changed (murphys law). And bad is, that I have to know the variable type, otherwise I can not imagine how the reset value is.

And its quite easy to return it back. ;-)

grodansparadis, 2015-06-22:
On the other hand you can use writevar to obtain what you want. resetvar  "reste".