grodansparadis/vscp #162: Wesocket feature: Variable shall be sent automatically onChange / onCreate / onRemoved

BlueAndi, 2015-07-21:
Instead of polling a variable, it would be better to be notified about a new variable, a removed variable and in case a variable value changed.

I thought about a subscribe mechanism for every variable.

Advantage:
* Easier implementation on client side
* Less communication load
* Less latency, therefore better usability ... the user see the change not a second later

grodansparadis, 2015-07-27:
Yes agree. This was actually in the first version but I lacked the time 
to implement. Will do this.

Den 2015-07-21 kl. 13:25, skrev Andreas Merkle:

> Instead of polling a variable, it would be better to be notified about
> a new variable, a removed variable and in case a variable value changed.
>
> I thought about a subscribe mechanism for every [...]

grodansparadis, 2020-12-10:
No tvalid anymore but will maybe come in a driver

closed 2020-12-10 by grodansparadis