grodansparadis/vscp #164: Standardize websocket GT and LISTVAR

BlueAndi, 2015-07-24:
LISTVAR return every variable with +;LiSTVAR;.... but didn't notfiy about how many variables will follow.
GT return every table row with +;TR; .... and +;GT;XXX notify about how many table rows will follow.

It would be good to standardize the behavior of them and future commands.

I prefer the approach of GT, because it is clear how many items will follow. Using LISTVAR, a timeout is necessary to determine the end of the list.

grodansparadis, 2015-07-27:
Good. I will fix/change that.

Den 2015-07-24 kl. 15:23, skrev Andreas Merkle:

> LISTVAR return every variable with +;LiSTVAR;.... but didn't notfiy
> about how many variables will follow.
> GT return every table row with +;TR; .... and +;GT;XXX notify about
> how many table rows [...]

grodansparadis, 2018-02-05:
Should be fixed.

closed 2018-02-05 by grodansparadis