grodansparadis/vscp #167: Rename JSON property "session-id" to session_id

dinguluer, 2015-08-12:
There is a bug in the naming convection for session-id JSON property.

Please change the name of JSON property "session-id" to session_id, because :--

JSON object properties naming convection it recommends that:--
1/ Property names must be camel-cased, ascii strings.
2/ The first character must be a letter, an underscore (_) or a dollar sign ($).

BlueAndi, 2015-09-01:
Can you verify the bugfix please Dinesh?

closed 2015-09-01 by grodansparadis

dinguluer, 2015-09-02:
> via 53bfdaa

>> Can you verify the bugfix please Dinesh?
Is this change made to the vscpd demon ?

Regards,
Dinesh

grodansparadis, 2015-09-02:
Send this to the group also as more people are affected.

Yes it is.  Some other changes to.  So read the docs.  Will do some work 
on this today myself and later tonight hopefully update the demo server 
with the latest code.

To be consistent I have removed '-' everywhere so session-id is now 
vscpsession. Also the parameter session is now vscpsession and is the 
same for both POST's and GETS.  user is vscpuser and password is 
vscpsecret. etc etc

Cheers
/Ake

ps I NEVER change API's normally. This is an exception and that rule 
will apply to this API also. ds

Den 2015-09-02 kl. 08:01, skrev Dinesh Guleria:

> >> Can you verify the bugfix please Dinesh?
> Is this change made to the vscpd demon ?
>
> Regards,
> Dinesh
>
>
> > Closed #167 <https://github.com/grodansparadis/vscp_software/issues/167>
> > via 53bfdaa

commit 53bfdaa0 (2015-09-01): Fixes #167 and  #168