grodansparadis/vscp #170: Setup recipes xml example contains errors

BlueAndi, 2015-08-23:
Fixed example:

```
<setup>
    <recipe>
        <name>Blink-channel0</name>
        <description lang="en">
        Set channel 0 to output and blink with 10 Hz.
        </description>
        <!-- Set channel as output -->
        <bit-in-reg pos="0" page="0" offset="" value="false" />
        <register page="0" offset="0" value="0" />
        <!-- Read frequency from user -->
        <messagebox>
            <function>input</function>
            <head lang="en">Beijing I/O node</head>
            <description lang="en">With what frequency should channel blink?</description>
            <variable type="byte" name="frequency" />
        </messagebox>
        <!-- Write frequency to abstraction register -->
        <abstraction name="blink-frequency0" value="$frequency" />
    </recipe>
</setup>
```
Some notes:
* You should not use a tag name several times, but with different content, e.g. register. This makes it complex and bad to write a xml schema later.
* If a tag e.g. messagebox can appear more than once, best practice is to group them with a separate tag e.g. messageboxes. Makes it easier to read.
* Rename the tag head to name. So every typical combination is always name and description.

grodansparadis, 2015-09-01:
Den 2015-08-23 kl. 15:37, skrev Andreas Merkle:
>
> Fixed example:
>
> |<setup> <recipe> <name>Blink-channel0</name> <description lang="en"> 
> Set channel 0 to output and blink with 10 Hz. </description>  <bit-in-reg pos="0" page="0" offset="" 
> value="false" /> <register page="0" offset="0" value="0" />  <messagebox> <function>input</function> <head 
> lang="en">Beijing I/O node</head> <description lang="en">With what 
> frequency should channel blink?</description> <variable type="byte" 
> name="frequency" /> </messagebox>  <abstraction name="blink-frequency0" value="$frequency" 
> /> </recipe> </setup>|
In the docs.

> ||
>
> Some notes:
>
>   * You should not use a tag name several times, but with different
>     content, e.g. register. This makes it complex and bad to write a
>     xml schema later.
>
But for register and abstraction it really make sense IMO.

>   * If a tag e.g. messagebox can appear more than once, best practice
>     is to group them with a separate tag e.g. messageboxes. Makes it
>     easier to read.
>
But this is linear. Things should be carried out in the order it is read.

>   * Rename the tag head to name. So every typical combination is
>     always name and description.
>

Good suggestion! Done!

/Ake

Be Hungry - Stay Foolish! Åke Hedman, Grodans Paradis AB / Paradise of 
the Frog Brattbergavägen 17, 82050 LOS, SWEDEN Phone: +46 (0)840011835 
CellPhone: +46 (0)730-533146 http://www.grodansparadis.com 
http://www.vscp.org

closed 2015-09-01 by grodansparadis

BlueAndi, 2015-09-01:
>>
>> Some notes:
>>
>>   * You should not use a tag name several times, but with different
>>     content, e.g. register. This makes it complex and bad to write a
>>     xml schema later.
>>

>But for register and abstraction it really make sense IMO.

The abstraction tag inside the abstractions tag (*funny*) describes an abstract value. The abstraction tag inside the recipe tag describes how to set the abstraction value for a specific result. IMO and experience it is better to differ between them. Let define a <access-abstraction>, <access-register> or maybe <write-abstraction>, <write-register>.
Why making it more complex by being not precise enough?

>>   * If a tag e.g. messagebox can appear more than once, best practice
>>     is to group them with a separate tag e.g. messageboxes. Makes it
>>     easier to read.
>>

>But this is linear. Things should be carried out in the order it is read.

I understand your opinion. But who will read the specification to know which elements can appear several times and which not? I learned in business the 2nd one is the bigger problem. But its still a proposal, of course.

BlueAndi, 2015-09-01:
BTW the xml example in the docs is still not correct
The <function>input</function> is there twice. :-)

grodansparadis, 2015-09-01:
Better now?

Den 2015-09-01 kl. 19:43, skrev Andreas Merkle:

>
> BTW the xml example in the docs is still not correct
> The input is there twice. :-)

grodansparadis, 2015-09-01:
Den 2015-09-01 kl. 18:43, skrev Andreas Merkle:
>
>         Some notes:
>
>           * You should not use a tag name several times, but with
>             different content, e.g. register. This makes it complex
>             and bad to write a xml schema later.
>
>     But for register and abstraction it really make sense IMO.
>
> The abstraction tag inside the abstractions tag (/funny/) describes an 
> abstract value. The abstraction tag inside the recipe tag describes 
> how to set the abstraction value for a specific result. IMO and 
> experience it is better to differ between them. Let define a , or 
> maybe , .
> Why making it more complex by being not precise enough?
>
OK suggest names.

> * If a tag e.g. messagebox can appear more than once, best
>             practice is to group them with a separate tag e.g.
>             messageboxes. Makes it easier to read.
>
>     But this is linear. [...]

Yes I understand the problem. Every top-tag  (under script) can be there 
several times as far as I see it.  Name and description being the two 
exceptions which works well anyway if parsing them just replace the 
previous content. So read a "top-tag" and do what it says then parse the 
next and do what is says.

As the order is actually important maybe there should be a <step></step> 
tag to clarify that it is.

But sure it is possible to define elements that can be used later and 
reuse them and even have recipes that call other recipes etc but the 
task grows. And eventually maybe that is the thing to do. But then again 
the Javascript embedded road is a better way to go.

/Ake

-- 
Be Hungry - Stay Foolish!
Åke Hedman, Grodans Paradis AB / Paradise of the Frog
Brattbergavägen 17, 82050 LOS, SWEDEN
Phone: +46 (0)840011835 CellPhone: +46 (0)730-533146
http://www.grodansparadis.com http://www.vscp.org

BlueAndi, 2015-09-01:
Sorry ... I already suggested names in the previous comment, but the markup was wrong, so they were hidden. Comment updated now.

grodansparadis, 2015-09-01:
PK we go for "write..." if that feels OK