grodansparadis/vscp #177: Failed to write a string variable via websocket, which contains a semicolon inside.

BlueAndi, 2015-09-05:


grodansparadis, 2015-09-05:
Hard to solve in a nice way.  Would it be OK to disallow it and make it 
possible by inserting it with an escape as "\;"

Den 2015-09-05 kl. 13:34, skrev Andreas Merkle:

BlueAndi, 2015-09-05:
Good question. I think we could live with it for now. I will update the javascript lib, so that it is automaticallty done.

grodansparadis, 2015-09-05:
This reflects the tcp/ip variable setting as well I guess.  Have to 
think a bit about this. Good find.

Den 2015-09-05 kl. 17:09, skrev Andreas Merkle:

>
> Good question. I think we could live with it for now. I will update 
> the javascript lib, so that it is automaticallty done.

grodansparadis, 2015-09-16:
Added info to the docs.

closed 2015-09-16 by grodansparadis

BlueAndi, 2015-09-16:
That a semicolon is not allowed for variable names is fine.
But it is now not allowed as value too? Also it is escaped like '\;'?

grodansparadis, 2015-09-16:
No we have to handle that outside of the interface. It could work with
some other escape as a semicolon screw things up. One way would be to
url encode values.But for now this will be the solution.

Den 2015-09-16 kl. 22:41, skrev Andreas Merkle:

> That a semicolon is not allowed for variable names is fine.
> But it is now not allowed as value too? Also it is escaped like '\;'?