grodansparadis/vscp #179: Creating a empty string variable at the daemon is not possible.

BlueAndi, 2015-09-05:
I am sure it was possible this morning to create a empty string variable on the demoserver, but now it isn't anymore.

grodansparadis, 2015-09-05:
I have updated it to the latest greatest a few minutes ago... ;-/

Den 2015-09-05 kl. 17:14, skrev Andreas Merkle:

>
> I am sure it was possible this morning to create a empty string 
> variable on the demoserver, but now it isn't anymore.

BlueAndi, 2015-09-05:
ahh ... latest greatest ... *LOL*

BlueAndi, 2015-09-09:
Could you update the demoserver with the last changes?

grodansparadis, 2015-09-09:
Will do that tomorrow.  To late?

Den 2015-09-09 kl. 23:21, skrev Andreas Merkle:

> Could you update the demoserver with the last changes?

BlueAndi, 2015-09-09:
Yes, thats ok ... will go to bed now. Good night!

grodansparadis, 2015-09-09:
Goodnight

Den 2015-09-09 kl. 23:24, skrev Andreas Merkle:

> Yes, thats ok ... will go to bed now. Good night!

BlueAndi, 2015-09-10:
Successful tested.

closed 2015-09-10 by BlueAndi

grodansparadis, 2015-09-10:
Great!

Den 2015-09-10 kl. 19:19, skrev Andreas Merkle:

> Successful tested.