grodansparadis/vscp #181: Wrong description of GUID in specification for vscpEvent

BlueAndi, 2015-09-11:
See
http://www.vscp.org/docs/vscphelper/doku.php?id=helper_lib_api_communication#vscphlp_sendeventex
```
uint8_t  GUID[ 16 ];    // Node globally unique id LSB(0) -> MSB(15
```

Should be MSB(0) -> LSB(15), or?

grodansparadis, 2015-09-11:
Yes that is right.

Den 2015-09-11 kl. 21:29, skrev Andreas Merkle:

> See
> http://www.vscp.org/docs/vscphelper/doku.php?id=helper_lib_api_communication#vscphlp_sendeventex
> 
> |uint8_t GUID[ 16 ]; // Node globally unique id LSB(0) -> MSB(15 |
> 
> Should be MSB(0) -> LSB(15), or?

BlueAndi, 2015-09-11:
Can you check this diagram please? Is it correct or did I miss something?

http://www.websequencediagrams.com/cgi-bin/cdraw?lz=dGl0bGUgTm9kZSBkaXNjb3ZlcnkKCnBhcnRpY2lwYW50IFdlYmNsaWVudAAJDURhZW1vbgAcDUFsbCBMMSBpbnRlcmZhY2VzAAwSMgAGGAAsDCB3aXRoIEdVSUQgeHkKCm5vdGUgb3ZlcgCBAAo6IFRoZQAcBmluIHRoZSBldmVudCBzdHJ1Y3R1cmUgaXMAEgUASgpHVUlELgoKYWwAZwUAKgVzICgwIDw9IGNsYXNzIDwgNTEyKQogIGFsAIFpCyB0byBMMSBub2RlCiAgIACCAwotPgCBfAY6AEYJABsFAIIRBi0-AIF7EQAeClxuKG9BZGRyID0AgS8HR1VJRFsxNV0pCiAgZW5kAHwHAG8HIHRvAIJ3CyAgIACCVRIAbB8AgisLAHgLAHEKID0gMDA6OgCBDwUAewhlbmQAgiMJAIJ2BUwyAIIrCTUxMgCCKgwxMDI0KQoAggIYADANXG4oaW5jbHVkZXMAgwsPIHh5IGluIGRhdGEpCgCCNw8AhCgMAEISCiAgAIQXCgCDCwhDb252ZXIAgzUIAIMWCACDFAgAhFMZAIJ_JQCCEQoAggkKMTAyNACEPgw2NTUzNgCBfRgAgkEJAIFKHwCCaggAgwgF&s=modern-blue

BlueAndi, 2015-09-11:
I try to get a clear picture about the relation between Levels, GUIDs and node ids. Maybe we can add this later to the specification too.

grodansparadis, 2015-09-11:
Looks very nice.

The daemon is a special case to.

On Ethernet for example you can sit with a device and need to discover
nodes. Bluetooth the dsame

Generally wireless like 6LowPan

From a BT becon.

Multicast.

etc.

Maybe could be called "cloud"

Den 2015-09-11 kl. 22:32, skrev Andreas Merkle:

> Can you check this diagram please? Is it correct or did I miss something?
> 
> http://www.websequencediagrams.com/cgi-bin/cdraw?lz=dGl0bGUgTm9kZSBkaXNjb3ZlcnkKCnBhcnRpY2lwYW50IFdlYmNsaWVudAAJDURhZW1vbgAcDUFsbCBMMSBpbnRlcmZhY2VzAAwSMgAGGAAsDCB3aXRoIEdVSUQgeHkKCm5vdGUgb3ZlcgCBAAo6IFRoZQAcBmluIHRoZSBldmVudCBzdHJ1Y3R1cmUgaXMAEgUASgpHVUlELgoKYWwAZwUAKgVzICgwIDw9IGNsYXNzID
> wgNTEyKQogIGFsAIFpCyB0byBMMSBub2RlCiAgIACCAwotPgCBfAY6AEYJABsFAIIRBi0-AIF7EQAeClxuKG9BZGRyID0AgS8HR1VJRFsxNV0pCiAgZW5kAHwHAG8HIHRvAIJ3CyAgIACCVRIAbB8AgisLAHgLAHEKID0gMDA6OgCBDwUAewhlbmQAgiMJAIJ2BUwyAIIrCTUxMgCCKgwxMDI0KQoAggIYADANXG4oaW5jbHVkZXMAgwsPIHh5IGluIGRhdGEpCgCCNw8AhCgMAEISCiAgAIQXCgCDCwhDb252ZXIAgzUIAIMWCACDFAgAhFMZAIJ_JQCCEQoAggkKMTAyNACEPgw2NTUzNgCBfRgAgkEJAIFKHwCCaggAgwgF&s=modern-blue
> <http://www.websequencediagrams.com/cgi-bin/cdraw?lz=dGl0bGUgTm9kZSBkaXNjb3ZlcnkKCnBhcnRpY2lwYW50IFdlYmNsaWVudAAJDURhZW1vbgAcDUFsbCBMMSBpbnRlcmZhY2VzAAwSMgAGGAAsDCB3aXRoIEdVSUQgeHkKCm5vdGUgb3ZlcgCBAAo6IFRoZQAcBmluIHRoZSBldmVudCBzdHJ1Y3R1cmUgaXMAEgUASgpHVUlELgoKYWwAZwUAKgVzICgwIDw9IGNsYXNzIDwgNTEyKQogIGFsAIFpCyB0byBMMSBub2RlCiAgIACCAwotPgCBfAY6AEYJABsFAIIRBi0-AIF7EQAeClxuKG9BZGRyID0AgS8HR1VJRFsxNV0pCiAgZW5kAHwHAG8HIHRvAIJ3CyAgIACCVRIAbB8AgisLAHgLAHEKID0gMDA6OgCBDwUAewhlbmQAgiMJAIJ2BUwyAIIrCTUxMgCCKgwxMDI0KQoAggIYADANXG4oaW5jbHVkZXMAgwsPIHh5IGluIGRhdGEpCgCCNw8AhCgMAEISCiAgAIQXCgCDCwhDb252ZXIAgzUIAIMWCACDFAgAhFMZAIJ_JQCCEQoAggkKMTAyNACEPgw2NTUzNgCBfRgAgkEJAIFKHwCCaggAgwgF&s=modern-blue>

grodansparadis, 2015-09-11:
Definitly!

Den 2015-09-11 kl. 22:34, skrev Andreas Merkle:

> I try to get a clear picture about the relation between Levels, GUIDs
> and node ids. Maybe we can add this later to the specification too.

BlueAndi, 2015-09-11:
Think I didn't get your whole answer ;-)
Can you change it directly here and share the link with the code back?

http://www.websequencediagrams.com/?lz=dGl0bGUgLi4uCgpwYXJ0aWNpcGFudCBXZWJjbGllbnQACQ1EYWVtb24AHA1BbGwgTDEgaW50ZXJmYWNlcwAMEjIABhgALAwgd2l0aCBHVUlEIHh5Cgpub3RlIG92ZXIAgQAKOiBUaGUAHAZpbiB0aGUgZXZlbnQgc3RydWN0dXJlIGlzABIFAEoKR1VJRC4KCmFsAGcFACoFcyAoMCA8PSBjbGFzcyA8IDUxMikKICBhbACBaQsgdG8gTDEgbm9kZQogICAAggMKLT4AgXwGOgBGCQAbBQCCEQYtPgCBexEAHgpcbihvQWRkciA9AIEvB0dVSURbMTVdKQogIGVuZAB8BwBvByB0bwCCdwsgICAAglUSAGwfAIIrCwB4CwBxCiA9IDAwOjoAgQ8FAHsIZW5kAIIjCQCCdgVMMgCCKwk1MTIAgioMMTAyNCkKAIICGAAwDVxuKGluY2x1ZGVzAIMLDyB4eSBpbiBkYXRhKQoAgjcPAIQoDABCEgogIACEFwoAgwsIQ29udmVyAIM1CACDFggAgxQIAIRTGQCCfyUAghEKAIIJCjEwMjQAhD4MNjU1MzYAgX0YAIJBCQCBSh8AgmoIAIMIBQ&s=modern-blue

grodansparadis, 2015-09-12:
Fixed.  I have opened the wiki up now for changes. Keep it that way if it works.

closed 2015-09-12 by grodansparadis

grodansparadis, 2015-09-12:
OK I will do that. Not today though. But you have to redo it anyway
because I am terrible at everything like this. En result ALWAYS looks awful.

Den 2015-09-11 kl. 22:47, skrev Andreas Merkle:

> Think I didn't get your whole answer ;-)
> Can you change it directly here and share the link with the code back?
> 
> http://www.websequencediagrams.com/?lz=dGl0bGUgLi4uCgpwYXJ0aWNpcGFudCBXZWJjbGllbnQACQ1EYWVtb24AHA1BbGwgTDEgaW50ZXJmYWNlcwAMEjIABhgALAwgd2l0aCBHVUlEIHh5Cgpub3RlIG92ZXIAgQAKOiBUaGUAHAZpbiB0aGUgZXZlbnQgc3RydWN0dXJlIGlzABIFAEoKR1VJRC4KCmFsAGcFACoFcyAoMCA8PSBjbGFzcyA8IDUxMikKICBhbACBaQsgdG8gTDEgbm9kZQogICAAggMKLT4AgXwGOg
> BGCQAbBQCCEQYtPgCBexEAHgpcbihvQWRkciA9AIEvB0dVSURbMTVdKQogIGVuZAB8BwBvByB0bwCCdwsgICAAglUSAGwfAIIrCwB4CwBxCiA9IDAwOjoAgQ8FAHsIZW5kAIIjCQCCdgVMMgCCKwk1MTIAgioMMTAyNCkKAIICGAAwDVxuKGluY2x1ZGVzAIMLDyB4eSBpbiBkYXRhKQoAgjcPAIQoDABCEgogIACEFwoAgwsIQ29udmVyAIM1CACDFggAgxQIAIRTGQCCfyUAghEKAIIJCjEwMjQAhD4MNjU1MzYAgX0YAIJBCQCBSh8AgmoIAIMIBQ&s=modern-blue
> <http://www.websequencediagrams.com/?lz=dGl0bGUgLi4uCgpwYXJ0aWNpcGFudCBXZWJjbGllbnQACQ1EYWVtb24AHA1BbGwgTDEgaW50ZXJmYWNlcwAMEjIABhgALAwgd2l0aCBHVUlEIHh5Cgpub3RlIG92ZXIAgQAKOiBUaGUAHAZpbiB0aGUgZXZlbnQgc3RydWN0dXJlIGlzABIFAEoKR1VJRC4KCmFsAGcFACoFcyAoMCA8PSBjbGFzcyA8IDUxMikKICBhbACBaQsgdG8gTDEgbm9kZQogICAAggMKLT4AgXwGOgBGCQAbBQCCEQYtPgCBexEAHgpcbihvQWRkciA9AIEvB0dVSURbMTVdKQogIGVuZAB8BwBvByB0bwCCdwsgICAAglUSAGwfAIIrCwB4CwBxCiA9IDAwOjoAgQ8FAHsIZW5kAIIjCQCCdgVMMgCCKwk1MTIAgioMMTAyNCkKAIICGAAwDVxuKGluY2x1ZGVzAIMLDyB4eSBpbiBkYXRhKQoAgjcPAIQoDABCEgogIACEFwoAgwsIQ29udmVyAIM1CACDFggAgxQIAIRTGQCCfyUAghEKAIIJCjEwMjQAhD4MNjU1MzYAgX0YAIJBCQCBSh8AgmoIAIMIBQ&s=modern-blue>