grodansparadis/vscp #189: Node label in VSCPworks

waaslandwolf, 2015-10-14:
Ake,

The VSCPworks monitor window at the moment spews event data which is kind of cryptic. Difficult to see where received events come from.

You know what would be cool? If VSCPworks wouldn't just display the nodeid of each event received, but if you could assign labels to each nodeid. So you'd be able to see, in human understandable form, what that node is meant for. 

For example; instead of seeing

RX CLASS1_INFORMATION BUTTON Nodeid=1

you'd see

RX CLASS1_INFORMATION BUTTON Nodeid=1 *Kitchen_buttons*

grodansparadis, 2015-10-14:
Hi,

this is planed. The automation interface of the daemon is not supposed 
to not just send heartbeats as today  but also collect info from a new 
nodes. When this is in place a GUID can be connected to a symbolic name 
and then  what you suggest is possible. Instead of using GUID's then 
names can be used all over. This will also give another dimension to 
node discovery.  I think that will be clearer when the Frankfurt BT is 
in place.

I have one thing left on this side before this can be done and that is 
the rewrite of the webserver. A lot of work.  It will take a while and 
this release will come first.

Thanks a lot for input. It is much more fun to work when it comes from 
time to time instead of everything going out to /dev/null.

/Ake

grodansparadis, 2019-03-25:
VSCP Works is deprecated

closed 2019-03-25 by grodansparadis