grodansparadis/vscp #194: vscpd webserver seems not to start (sodium, debian wheezy)

BlueAndi, 2015-10-27:
/vscp folder is not accessible via port 8080
websocket is down too

the default configuration (vscp.conf) is used

the log files says "webserver is active"

tested on a beaglebone black, debian wheezy, libwxbase2.8-dev

grodansparadis, 2015-10-27:
Anything in the logs?

Den 2015-10-27 kl. 22:50, skrev Andreas Merkle:

>
> /vscp folder is not accessible via port 8080
> websocket is down too
>
> the default configuration (vscp.conf) is used
>
> the log files says "webserver is active"
>
> tested on a beaglebone black, debian wheezy, libwxbase2.8-dev

grodansparadis, 2015-10-28:
Just tested with a fresh installation here and it looks like it work as 
it should.

Den 2015-10-27 kl. 22:50, skrev Andreas Merkle:

>
> /vscp folder is not accessible via port 8080
> websocket is down too
>
> the default configuration (vscp.conf) is used
>
> the log files says "webserver is active"
>
> tested on a beaglebone black, debian wheezy, libwxbase2.8-dev

BlueAndi, 2015-10-28:
Strange ... I just powered the beaglebone on again a minute ago, with same software as yesterday ... now it works here as well.

grodansparadis, 2015-10-28:
Good to here. Could it be two daemons running at the same time?

Den 2015-10-28 kl. 21:41, skrev Andreas Merkle:

>
> Strange ... I just powered the beaglebone on again a minute ago, with 
> same software as yesterday ... now it works here as well.

BlueAndi, 2015-10-28:
No. I stopped the daemon and checked it with
ps -Al | grep vscpd

grodansparadis, 2015-10-28:
Strange then.

Den 2015-10-28 kl. 21:44, skrev Andreas Merkle:

>
> No. I stopped the daemon and checked it with
> ps -Al | grep vscpd

BlueAndi, 2015-10-30:
Same effect again ... still the same software.
Powered up the beaglebone and the webserver seems to be dead. No websocket available and etc.

What can I check now?

grodansparadis, 2015-10-31:
Can it be something else on port 8080? Try to change it to something 
else on the daemon.

grodansparadis, 2017-07-12:
old

closed 2017-07-12 by grodansparadis