grodansparadis/vscp #196: RPi & Frankfurt daemon not woking

grodansparadis, 2015-10-28:
On rpi & Frankfurt it's a semi-success:

the canal interface works fine

through daemon it's weird: using vscpd -s  I can see the Frankfurt interface is opened.

when using works to see interfaces, Frankfurt isn't in the list. Same with web-interface.

git pull from yesterday.

svenZ80, 2015-10-28:
more info:
interfaces ok, forgot to use sudo to restart daemon
I removed the simulation driver, and now daemon works fine with Frankfurt

grodansparadis, 2015-10-28:
OK you added it here. Thanks! Perfect.

Den 2015-10-28 kl. 20:14, skrev svenZ80:

>
> more info:
> interfaces ok, forgot to use sudo to restart daemon
> I removed the simulation driver, and now daemon works fine with Frankfurt

closed 2016-01-18 by grodansparadis