grodansparadis/vscp #201: Menu items for VSCPworks Configuration window commands

waaslandwolf, 2016-02-28:
Ake,

I am running VSCPworks on Ubuntu on a little eeePC. It works fine, however in the Configuration window the lower buttons are below my lower screen limit and thus unavailable. The main annoyance is the Update button which I can't push after making changes to the registers. So essentially I can't use VSCPworks on the eeePC to make config changes. Which is a pity since it is a nice little machine to walk around the house with while troubleshooting.

Would it be possible to make an item in the menu which lists the most used commands for the Configuration window? Shortcuts would be nice too (ctrl-u to update the registers?)

Thanks,
David.

grodansparadis, 2016-02-28:
OK I will have a look.  Want more f this stuff.

closed 2016-03-02 by grodansparadis

waaslandwolf, 2016-03-03:
> Closed #201 <https://github.com/grodansparadis/vscp/issues/201> via
> 7b957fb

Ake,

Thanks for your quick work :-) I just did a git pull, ./configure, make but
compilation fails with the following error. Have fun troubleshooting :-)

Looks like something lua related.

/wx-3.0 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -D VSCP_ENABLE_LUA
-I/usr/include/lua5.2 -DwxUSE_GUI=1 -D MG_ENABLE_THREADS -I. -I../common
-I../../common  -c -o ../common/canalsuperwrapper.o
../common/canalsuperwrapper.cpp
g++ -g -O2  -I/usr/lib/i386-linux-gnu/wx/include/gtk2-unicode-3.0
-I/usr/include/wx-3.0 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -D
VSCP_ENABLE_LUA -I/usr/include/lua5.2 -DwxUSE_GUI=1 -D MG_ENABLE_THREADS
-I. -I../common -I../../common  -c -o ../common/dllwrapper.o
../common/dllwrapper.cpp
g++ -g -O2  -I/usr/lib/i386-linux-gnu/wx/include/gtk2-unicode-3.0
-I/usr/include/wx-3.0 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -D
VSCP_ENABLE_LUA -I/usr/include/lua5.2 -DwxUSE_GUI=1 -D MG_ENABLE_THREADS
-I. -I../common -I../../common  -c -o ../common/guid.o ../common/guid.cpp
g++ -g -O2  -I/usr/lib/i386-linux-gnu/wx/include/gtk2-unicode-3.0
-I/usr/include/wx-3.0 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -D
VSCP_ENABLE_LUA -I/usr/include/lua5.2 -DwxUSE_GUI=1 -D MG_ENABLE_THREADS
-I. -I../common -I../../common  -c -o ../common/vscphelper.o
../common/vscphelper.cpp
g++ -g -O2  -I/usr/lib/i386-linux-gnu/wx/include/gtk2-unicode-3.0
-I/usr/include/wx-3.0 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -D
VSCP_ENABLE_LUA -I/usr/include/lua5.2 -DwxUSE_GUI=1 -D MG_ENABLE_THREADS
-I. -I../common -I../../common  -c -o ../common/vscpeventhelper.o
../common/vscpeventhelper.cpp
../common/vscpeventhelper.cpp: In constructor
‘VSCPInformation::VSCPInformation()’:
../common/vscpeventhelper.cpp:75:1: note: variable tracking size limit
exceeded with -fvar-tracking-assignments, retrying without
 VSCPInformation::VSCPInformation( void )
 ^
g++ -g -O2  -I/usr/lib/i386-linux-gnu/wx/include/gtk2-unicode-3.0
-I/usr/include/wx-3.0 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -D
VSCP_ENABLE_LUA -I/usr/include/lua5.2 -DwxUSE_GUI=1 -D MG_ENABLE_THREADS
-I. -I../common -I../../common  -c -o ../common/vscpremotetcpif.o
../common/vscpremotetcpif.cpp
g++ -g -O2  -I/usr/lib/i386-linux-gnu/wx/include/gtk2-unicode-3.0
-I/usr/include/wx-3.0 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -D
VSCP_ENABLE_LUA -I/usr/include/lua5.2 -DwxUSE_GUI=1 -D MG_ENABLE_THREADS
-I. -I../common -I../../common  -c -o ../common/mdf.o ../common/mdf.cpp
g++ -g -O2  -I/usr/lib/i386-linux-gnu/wx/include/gtk2-unicode-3.0
-I/usr/include/wx-3.0 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -D
VSCP_ENABLE_LUA -I/usr/include/lua5.2 -DwxUSE_GUI=1 -D MG_ENABLE_THREADS
-I. -I../common -I../../common  -c -o ../common/register.o
../common/register.cpp
g++ -g -O2  -I/usr/lib/i386-linux-gnu/wx/include/gtk2-unicode-3.0
-I/usr/include/wx-3.0 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -D
VSCP_ENABLE_LUA -I/usr/include/lua5.2 -DwxUSE_GUI=1 -D MG_ENABLE_THREADS
-I. -I../common -I../../common  -c -o ../common/canalconfobj.o
../common/canalconfobj.cpp
g++ -g -O2  -I/usr/lib/i386-linux-gnu/wx/include/gtk2-unicode-3.0
-I/usr/include/wx-3.0 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -D
VSCP_ENABLE_LUA -I/usr/include/lua5.2 -DwxUSE_GUI=1 -D MG_ENABLE_THREADS
-I. -I../common -I../../common  -c -o ../common/vscpmulticast.o
../common/vscpmulticast.cpp
g++ -g -O2  -I/usr/lib/i386-linux-gnu/wx/include/gtk2-unicode-3.0
-I/usr/include/wx-3.0 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -D
VSCP_ENABLE_LUA -I/usr/include/lua5.2 -DwxUSE_GUI=1 -D MG_ENABLE_THREADS
-I. -I../common -I../../common  -c -o ../common/knownnodes.o
../common/knownnodes.cpp
g++ -o vscpworks vscpworks.o dlgvscpmsg.o frmmain.o frmvscpsession.o
dlgnewvscpsession.o dlgconfiguration.o dlgvscpinterfacesettings.o
dlgvscpfilter.o vscpbootloader.o dlgvscptrmit.o frmdeviceconfig.o
frmscanfordevices.o dlgselectdaemoninterface.o frmmdfeditor.o dlgabout.o
vscptxobj.o gridtable.o dlgeditlevel1dmrow.o dlgsetfiltermask.o
dlgabstractionedit.o devicebootloaderwizard.o bootdevice.o
bootdevice_pic1.o bootdevice_vscp.o merlin.o
frmdaemondecisionmartrixeditor.o frmdaemonvariableeditor.o
frmsimpleuserinterfacegenerator.o dlgeventfilter.o dlgclasstypeselect.o
threadvscpwork.o dlgsetmanufactdata.o ../../common/crc.o
../../common/crc8.o ../../common/mongoose.o ../common/canalsuperwrapper.o
../common/dllwrapper.o ../common/guid.o ../common/vscphelper.o
../common/vscpeventhelper.o ../common/vscpremotetcpif.o ../common/mdf.o
../common/register.o ../common/canalconfobj.o ../common/vscpmulticast.o
../common/knownnodes.o   -L@top_wxbuilddir@lib  -lssl -lpcap -lz -lrt -lm
-lcrypto -lpthread -ldl  -L/usr/lib/i386-linux-gnu -pthread
-lwx_gtk2u_xrc-3.0 -lwx_gtk2u_html-3.0 -lwx_gtk2u_qa-3.0 -lwx_gtk2u_adv-3.0
-lwx_gtk2u_core-3.0 -lwx_baseu_xml-3.0 -lwx_baseu_net-3.0 -lwx_baseu-3.0
-llua5.2
/usr/bin/ld: cannot find -llua5.2
collect2: error: ld returned 1 exit status
make[1]: *** [vscpworks] Error 1
make[1]: Leaving directory `/home/john/vscp_software/src/vscp/vscpworks'
make: *** [all] Error 2
root@john-D410:/home/john/vscp_software#

grodansparadis, 2016-03-03:
Yes LUA is a demand now whcih can be disabled with --disable-lua on 
configure. But

sudo apt-get install lua5.2 lua5.2-dev

will be added to the build instructions one day.

The head code should probably not be run in a production environment yet 
as there is a lot of changes, done and a lot is untested yet. On the 
other hand if no one run it things will remain untested....

/Ake

commit 7b957fbc (2016-03-02): Fixes #201