grodansparadis/vscp #220: Add labels to VSCP Works client window

waaslandwolf, 2016-11-10:
The VSCP Works client window shows events as they are received on the bus. The following colums of information are shown:
- Direction (TX/RX)
- Class & Type
- Note (showing NodeID & GUID)

But the content of the event isn't shown in a clear way. One has to click on an event only to see some details appear in the window on the right. Even there the payload is only shown as HEX and DEC numbers which isn't easy to interpret. One has to know in the back of ones mind which zone and subzone numbers represent what in real-life, which nodeID is connected to what device in real life etc. 

Wouldn't it be cool if VSCP had a translation tabel that was able to cross-reference the HEX and DEC numbers to some real-life use case? 

For example; 
Nickname 10 = "Master bedroom 8-way button faceplate"
Zone 1 = "1st floor"
Subzone 253 = "Master bedroom button 8"
Index 8 = "button 8"

That way the client window (and the rest of the VSCP software suite) could represent a "CLASS1_INFORMATION, ON with data 0x08, 0x01, 0xFD" event as:
Master bedroom 8-way button faceplate; 1st floor, Master bedroom button 8, button 8

Much more useful, don't you agree? :-)

Happy coding!
David

vscp, 2017-03-14:
Hi David. For some reason this was not sent to me so I just saw it now. Yes this will come.

waaslandwolf, 2017-03-15:
Cool! I'm sure It will be much more user -friendly covering up All the hex codes ;-)

grodansparadis, 2019-03-25:
VSCP Works is deprecated

closed 2019-03-25 by grodansparadis