grodansparadis/vscp #221: Demon sends wrong time since epoch

ajje, 2016-12-05:
The SEGCTRL_HEARTBEAT sent out by the demon sends wrong time since epoch. And it also sends it at an odd interval of 61 seconds.

My system configuration:
VSCP 1.12.4.0 Magnesium
Windows 10 Pro X64
I'm running a x86 VSCP release as X64 crashes at start.

Here are some values that I have logged with VSCP works:

2016-12-05 22:33:00		0x58,0x45,0xEE,0x0C
2016-12-05 22:34:01		0x58,0x45,0xEE,0x49
2016-12-05 22:35:02		0x58,0x45,0xEE,0x86
2016-12-05 22:36:03		0x58,0x45,0xEE,0xC3
2016-12-05 22:37:04		0x58,0x45,0xEF,0x00
2016-12-05 22:38:05		0x58,0x45,0xEF,0x3D
2016-12-05 22:39:06		0x58,0x45,0xEF,0x7A
2016-12-05 22:40:07		0x58,0x45,0xEF,0xB7
2016-12-05 22:41:08		0x58,0x45,0xEF,0xF4
2016-12-05 22:42:09		0x58,0x45,0xEF,0x31		Going back in time (0xEF not increasing)
2016-12-05 22:43:10		0x58,0x45,0xEF,0x6E
2016-12-05 22:44:11		0x58,0x45,0xEF,0xAB
2016-12-05 22:45:12		0x58,0x45,0xEF,0xE8
2016-12-05 22:46:13		0x58,0x45,0xF0,0x25
2016-12-05 22:47:14		0x58,0x45,0xF0,0x62
2016-12-05 22:48:15		0x58,0x45,0xF0,0x9F
2016-12-05 22:49:16		0x58,0x45,0xF0,0xDC
2016-12-05 22:50:17		0x58,0x45,0xF0,0x19		Going back in time (0xF0 not increasing)
2016-12-05 22:51:18		0x58,0x45,0xF0,0x56
2016-12-05 22:52:19		0x58,0x45,0xF0,0x93
2016-12-05 22:53:20		0x58,0x45,0xF0,0xD0
2016-12-05 22:54:21		0x58,0x45,0xF1,0x0D
2016-12-05 22:55:22		0x58,0x45,0xF1,0x4A
2016-12-05 22:56:23		0x58,0x45,0xF1,0x87
2016-12-05 22:57:24		0x58,0x45,0xF1,0xC4
2016-12-05 22:58:25		0x58,0x45,0xF1,0x01		Going back in time (0xF1 not increasing)
2016-12-05 22:59:26		0x58,0x45,0xF1,0x3E
2016-12-05 23:00:27		0x58,0x45,0xF1,0x7B
2016-12-05 23:01:28		0x58,0x45,0xF1,0xB8
2016-12-05 23:02:29		0x58,0x45,0xF1,0xF5
2016-12-05 23:03:30		0x58,0x45,0xF2,0x32
2016-12-05 23:04:31		0x58,0x45,0xF2,0x6F
2016-12-05 23:05:32		0x58,0x45,0xF2,0xAC
2016-12-05 23:06:33		0x58,0x45,0xF2,0xE9
2016-12-05 23:07:34		0x58,0x45,0xF2,0x26		Going back in time (0xF2 not increasing)
2016-12-05 23:08:35		0x58,0x45,0xF2,0x63
2016-12-05 23:09:36		0x58,0x45,0xF2,0xA0
2016-12-05 23:10:37		0x58,0x45,0xF2,0xDD
2016-12-05 23:11:38		0x58,0x45,0xF3,0x1A

grodansparadis, 2016-12-06:
>
> The SEGCTRL_HEARTBEAT sent out by the demon sends wrong time since 
> epoch. And it also sends it at an odd interval of 61 seconds.
Fixed in 1.12.15.2

> My system configuration:
> VSCP 1.12.4.0 Magnesium
> Windows 10 Pro X64
> I'm running a x86 VSCP release as X64 crashes at start.

Almost always related to the configuration file and/or drivers. You cant 
run a 32-bit version rith 64-bit drivers and vice versa.

> Here are some values that I have logged with VSCP works:
>
> 2016-12-05 22:33:00 0x58,0x45,0xEE,0x0C
> 2016-12-05 22:34:01 0x58,0x45,0xEE,0x49
> 2016-12-05 22:35:02 0x58,0x45,0xEE,0x86
> 2016-12-05 22:36:03 [...]

This was a strange one. Will check it out. This may be a thing worth a 
Nobel price on the other hand... ;-)

Thanks for your input. VERY valuable!

/Ake

> <https://github.com/grodansparadis/vscp/issues/221>, or mute the 
> thread

grodansparadis, 2016-12-06:
This (wrong epoch) was due ta 9 instead of a 8 in a byte shift. Thanks!

closed 2016-12-06 by grodansparadis

ajje, 2016-12-06:
Great you found it!

commit b91dee0b (2016-12-06): Fixes #221