grodansparadis/vscp #222: Demon sends out Level2 events on Level1 interface

ajje, 2016-12-06:
When I add one extra CAN adapter to my VSCP network I can see that demon outputs Level2 HEART_BEAT on CAN bus. Look at attached screen dumps.

When I set "heartbeat-event" interval to 20 seconds. The demon outputs it every 21 second.

When I change "segmentcontoller-event" enable or interval nothing changes. The demon outputs SEGCTRL_HEARTBEAT every 61 seconds anyway.

[image]
[image]
[image]

grodansparadis, 2016-12-06:
>
> When I add one extra CAN adapter to my VSCP network I can see that 
> demon outputs Level2 HEART_BEAT on CAN bus. Look at attached screen dumps.
It looks like you are on a tcp/ip interface in the session on the 
picture. This is a Level II interface for the system so you will receive 
Level II events there. It will not go out on a Level I interface.

> When I set "heartbeat-event" interval to 20 seconds. The demon outputs 
> it every 21 second.
Fixed in #221

> When I change "segmentcontoller-event" enable or interval nothing 
> changes. The demon outputs SEGCTRL_HEARTBEAT every 61 seconds anyway.
As far as I can see this woks in 1.12.15.4   You can see 
http://localhost:8080/vscp/configure here (with your host) if the value 
actually changed when you changed the config.

Breaking glass today?

> canal-demon 
> <https://cloud.githubusercontent.com/assets/19362676/20919495/55a64ba0-bb9c-11e6-8700-04fd7a1d48d1.png>
> canal-usb 
> <https://cloud.githubusercontent.com/assets/19362676/20919497/58a582d0-bb9c-11e6-9a07-ffca06bb8ef6.png>
> vscpd conf 
> <https://cloud.githubusercontent.com/assets/19362676/20919500/5bbed160-bb9c-11e6-9c63-14ed304131b8.png>
>
> <https://github.com/grodansparadis/vscp/issues/222>, or mute the 
> thread

ajje, 2016-12-06:
When demon outputs Level2 HEART_BEAT on CAN bus it is received as a GLASS_BREAK on the other interface... Look at the data... Both screen dumps are on the same bus at the same time.

I will have a look at http://localhost:8080/vscp/configure....

grodansparadis, 2016-12-06:
OK that was strange, will check how that.

ajje, 2016-12-06:
Here is a screen dump of the configure window.

1.  segmentcontroller-event set to disable and interval 5 seconds.
     Configure indicates enabled 60 seconds.
     Demon outputs SEGCTRL_HEARTBEAT every 61 seconds.

2. heartbeat-event set to disable and interval 10 seconds.
    Configure indicates enabled 10 seconds.
    Demon don't output Level2 HEART_BEAT.

I'm on latest release from 2016-05-13 "VSCP & Friends Version 1.12.4 - Magnesium".
I'm on Windows x86 release because i need to use can232 driver from Neon release, that was not released as X64.

[image]

grodansparadis, 2016-12-06:
> settings error

This appears to be fixed in 1.12.15.5 because I can't repeat it there.

For the Level II event sent on Level I interface I can confirm a bug. A 
fix is on the way. Good catch!

closed 2016-12-06 by grodansparadis

ajje, 2016-12-06:
Thanks for your support!

Is it safe to try the master branch on a raspberry pi?
(I can't compile for windows my self but have sucessfully done so for linux).

grodansparadis, 2016-12-06:
Yes it should be possible to work with it as it is now. The problem is 
the database. Will not add any updating functionality to it as work 
continues until things are at some sort of alpha state.  If you continue 
to use the xml file for settings you should be in the clear. Possibly 
you should also removing the databases in /srv/vscp (.sqlite3) when you 
update your code to the headcode after you have setup a working system. 
At least if you have problems. I don't see any database updates coming 
but can't promise they will not.

How do you see he need for a windows version. Is it good to have or not?

Thanks for you work. Very (VERY!) valuable here.

ajje, 2016-12-06:
VSCP-Works for sure is good to have in Windows as Windows is my main operating system. I have also used the demon in Windows now durring development (feels faster to work with for me). And will continue to use the demon in Windows for development setup as long as it is available. But for my system that will run 24/7 I will use a Raspberry Pi for sure.. 

Windows version YES..

grodansparadis, 2016-12-07:
On 06/12/16 14:56, Anders Forsgren wrote:

> VSCP-Works for sure is good to have in Windows as Windows is my main
> operating system. I have also used the demon in Windows now durring
> development (feels faster to work with for me). And will [...]

I used to do all development on Windows before and it was easier back 
then.  But we have to see. Works will always be available on windows to. 
But I have to think bit about the daemon and the drivers. It is quite a 
bit of work to maintain the two.

-- 
Stay Hungry - Stay Foolish!
Åke Hedman
Brattbergavägen 17, 82050 LOS, SWEDEN
http://www.vscp.org http://www.akehedman.se

ajje, 2016-12-08:
I haven't actually been successful in compiling a working demon on linux yet.
- I have tried on an debian machine with socketcan and slcan.
- I have tried with Raspberry Pi with can232 driver (didn't have slcan, so have not tried that).
- I have successfully compiled VSCP-Works on and virtual Ubuntu 16.04 machine.
I manage to compile and start the demon and when I look with my Windows connected adapter it sends SEGCTRL_HEARTBEAT on the CAN bus. I can even login to the http://localhost:8080/vscp/configure interface. But when I connect to the TCP/IP interface with VSCP-works (both from Windows and Ubuntu) I lose a lot of CAN packages, I even lose SEGCTRL_HEARTBEAT packages. I have tested with latest Magnesium version and latest master. On latest master I get a package drop of 95% and on latest Magnesium I get 5% packet drop. The Ubuntu VSCP-Works works against Windows demon over TCP/IP interface without any packet drops. For me it feels like the demon TCP/IP on linux is broken!!

Maybe I doing something wrong when I compile on linux, I don't know...
Is it difficult to make deb packages for linux?

grodansparadis, 2016-12-09:
On 08/12/16 08:31, Anders Forsgren wrote:
> I haven't actually been successful in compiling a working demon on
> linux yet.
>
>   * I have tried on an debian machine with socketcan and slcan.
>   * I have tried with Raspberry Pi with can232 driver [...]

So I understand fully.  You have this problem on head code or on other 
Linux version code to?

> Is it difficult to make deb packages for linux?
>
No but it takes time. It's been in the lop a long time but just haven't 
found the time to do it.

-- 
Stay Hungry - Stay Foolish!
Åke Hedman
Brattbergavägen 17, 82050 LOS, SWEDEN
http://www.vscp.org http://www.akehedman.se

ajje, 2017-02-07:
Hi Åke,

I have now since a couple of weeks successfully been running VSCP daemon on an Rasperry Pi 3 using socketcan driver. I'm had no success with master branch, but using my own fork works. It also controls my outdoor light on my house with Sunrice/Sunset. Working great!!

More news coming...

grodansparadis, 2017-02-07:
NIce to here Anders.

commit e3185ab5 (2016-12-06): Fixes #222
commit f6b9d41f (2016-12-13): Should fix #222 (due to change in Mongoose 6.6)