grodansparadis/vscp #223: Error while compiling

mamac, 2017-03-14:
Got the error below while trying to compile the latest version:
`====================================================
Building in dir  src/vscp/vscpworks
====================================================
make[1]: Entering directory '/home/vscp/vscp_software/src/vscp/vscpworks'
g++ -g -O2  -I/usr/lib64/wx/include/gtk2-unicode-3.0 -I/usr/include/wx-3.0 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -D MG_ENABLE_SSL -D VSCP_ENABLE_LUA -I/usr/include/lua5.2 -DwxUSE_GUI=1 -D MG_ENABLE_THREADS -I. -I../common -I../../common  -c -o vscpworks.o vscpworks.cpp
In file included from ./vscpworks.h:57:0,
                 from ./frmdeviceconfig.h:61,
                 from vscpworks.cpp:99:
../common/canalsuperwrapper.h: In member function ‘int CCanalSuperWrapper::getVariableString(wxString&, wxString*)’:
../common/canalsuperwrapper.h:350:62: error: no matching function for call to ‘VscpRemoteTcpIf::getRemoteVariable(wxString&, wxString*&)’
         return m_vscptcpif.getRemoteVariable( name, strValue ); 
                                                              ^
../common/canalsuperwrapper.h:350:62: note: candidate is:
In file included from ../common/canalsuperwrapper.h:42:0,
                 from ./vscpworks.h:57,
                 from ./frmdeviceconfig.h:61,
                 from vscpworks.cpp:99:
../common/vscpremotetcpif.h:559:9: note: int VscpRemoteTcpIf::getRemoteVariable(const wxString&, wxString&)
     int getRemoteVariable( const wxString& name, wxString& strVariable );
         ^
../common/vscpremotetcpif.h:559:9: note:   no known conversion for argument 2 from ‘wxString*’ to ‘wxString&’
In file included from ./vscpworks.h:57:0,
                 from ./frmdeviceconfig.h:61,
                 from vscpworks.cpp:99:
../common/canalsuperwrapper.h: In member function ‘int CCanalSuperWrapper::setVariableString(wxString&, wxString&)’:
../common/canalsuperwrapper.h:361:28: error: ‘class VscpRemoteTcpIf’ has no member named ‘setVariableString’
         return m_vscptcpif.setVariableString( name, strValue ); 
                            ^
../common/canalsuperwrapper.h: In member function ‘int CCanalSuperWrapper::getVariableBool(wxString&, bool*)’:
../common/canalsuperwrapper.h:372:28: error: ‘class VscpRemoteTcpIf’ has no member named ‘getVariableBool’
         return m_vscptcpif.getVariableBool( name, bValue ); 
                            ^
../common/canalsuperwrapper.h: In member function ‘int CCanalSuperWrapper::setVariableBool(wxString&, bool)’:
../common/canalsuperwrapper.h:383:28: error: ‘class VscpRemoteTcpIf’ has no member named ‘setVariableBool’
         return m_vscptcpif.setVariableBool( name, bValue ); 
                            ^
../common/canalsuperwrapper.h: In member function ‘int CCanalSuperWrapper::getVariableInt(wxString&, int*)’:
../common/canalsuperwrapper.h:394:28: error: ‘class VscpRemoteTcpIf’ has no member named ‘getVariableInt’
         return m_vscptcpif.getVariableInt( name, value ); 
                            ^
../common/canalsuperwrapper.h: In member function ‘int CCanalSuperWrapper::setVariableInt(wxString&, int)’:
../common/canalsuperwrapper.h:405:28: error: ‘class VscpRemoteTcpIf’ has no member named ‘setVariableInt’
         return m_vscptcpif.setVariableInt( name, value ); 
                            ^
../common/canalsuperwrapper.h: In member function ‘int CCanalSuperWrapper::getVariableLong(wxString&, long int*)’:
../common/canalsuperwrapper.h:416:28: error: ‘class VscpRemoteTcpIf’ has no member named ‘getVariableLong’
         return m_vscptcpif.getVariableLong( name, value );  
                            ^
../common/canalsuperwrapper.h: In member function ‘int CCanalSuperWrapper::setVariableLong(wxString&, long int)’:
../common/canalsuperwrapper.h:427:28: error: ‘class VscpRemoteTcpIf’ has no member named ‘setVariableLong’
         return m_vscptcpif.setVariableLong( name, value );  
                            ^
../common/canalsuperwrapper.h: In member function ‘int CCanalSuperWrapper::getVariableDouble(wxString&, double*)’:
../common/canalsuperwrapper.h:438:28: error: ‘class VscpRemoteTcpIf’ has no member named ‘getVariableDouble’
         return m_vscptcpif.getVariableDouble( name, value ); 
                            ^
../common/canalsuperwrapper.h: In member function ‘int CCanalSuperWrapper::setVariableDouble(wxString&, double)’:
../common/canalsuperwrapper.h:449:28: error: ‘class VscpRemoteTcpIf’ has no member named ‘setVariableDouble’
         return m_vscptcpif.setVariableDouble( name, value ); 
                            ^
../common/canalsuperwrapper.h: In member function ‘int CCanalSuperWrapper::getVariableMeasurement(wxString&, wxString&)’:
../common/canalsuperwrapper.h:461:28: error: ‘class VscpRemoteTcpIf’ has no member named ‘getVariableMeasurement’
         return m_vscptcpif.getVariableMeasurement( name, strValue ); 
                            ^
../common/canalsuperwrapper.h: In member function ‘int CCanalSuperWrapper::setVariableMeasurement(wxString&, wxString&)’:
../common/canalsuperwrapper.h:473:28: error: ‘class VscpRemoteTcpIf’ has no member named ‘setVariableMeasurement’
         return m_vscptcpif.setVariableMeasurement( name, strValue ); 
                            ^
../common/canalsuperwrapper.h: In member function ‘int CCanalSuperWrapper::getVariableEvent(wxString&, vscpEvent*)’:
[... 26 lines cut ...]
                            ^
../common/canalsuperwrapper.h: In member function ‘int CCanalSuperWrapper::setVariableVSCPdata(wxString&, uint16_t, uint8_t*)’:
../common/canalsuperwrapper.h:565:28: error: ‘class VscpRemoteTcpIf’ has no member named ‘setVariableVSCPdata’
         return m_vscptcpif.setVariableVSCPdata( name, pData, sizeData ); 
                            ^
../common/canalsuperwrapper.h: In member function ‘int CCanalSuperWrapper::getVariableVSCPclass(wxString&, uint16_t*)’:
../common/canalsuperwrapper.h:576:28: error: ‘class VscpRemoteTcpIf’ has no member named ‘getVariableVSCPclass’
         return m_vscptcpif.getVariableVSCPclass( name, vscp_class ); 
                            ^
../common/canalsuperwrapper.h: In member function ‘int CCanalSuperWrapper::setVariableVSCPclass(wxString&, uint16_t)’:
../common/canalsuperwrapper.h:587:28: error: ‘class VscpRemoteTcpIf’ has no member named ‘setVariableVSCPclass’
         return m_vscptcpif.setVariableVSCPclass( name, vscp_class ); 
                            ^
../common/canalsuperwrapper.h: In member function ‘int CCanalSuperWrapper::getVariableVSCPtype(wxString&, uint16_t*)’:
../common/canalsuperwrapper.h:598:28: error: ‘class VscpRemoteTcpIf’ has no member named ‘getVariableVSCPtype’
         return m_vscptcpif.getVariableVSCPtype( name, vscp_type ); 
                            ^
../common/canalsuperwrapper.h: In member function ‘int CCanalSuperWrapper::setVariableVSCPtype(wxString&, uint16_t)’:
../common/canalsuperwrapper.h:610:28: error: ‘class VscpRemoteTcpIf’ has no member named ‘setVariableVSCPtype’
         return m_vscptcpif.setVariableVSCPtype( name, vscp_type ); 
                            ^
make[1]: *** [<builtin>: vscpworks.o] Error 1
make[1]: Leaving directory '/home/vscp/vscp_software/src/vscp/vscpworks'
make: *** [Makefile:13: all] Error 2
`

mamac, 2017-03-14:
Tried to dig into this myself but I have to admit it's a bit too difficult for me...

closed 2017-03-14 by grodansparadis

mamac, 2017-03-15:
Thank you Ake. I whish I could sort this out myself...

grodansparadis, 2017-03-15:
It was not easy as I had changed some method names. Entirely my fault.

Strange that I never saw your initial post just your second comment.

/Ake

commit 99bb0c0a (2017-03-14): Fixes #223 Thansk J-M