grodansparadis/vscp #224: Compile error

mamac, 2017-03-31:
Trying to make VSCP to work on a BananaPI running Raspbian wheezy.

Got the message below while compiling:

`====================================================
Building in dir  src/vscp/drivers/level2/sim/linux
====================================================
make[1]: Entering directory '/home/vscp/vscp/src/vscp/drivers/level2/sim/linux'
g++ -g -O2  -fPIC -I. -I../../../../common -I../../../../../common -I include   -I/usr/local/lib/wx/include/gtk3-unicode-3.1 -I/usr/local/include/wx-3.1 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -D MG_ENABLE_SSL -D VSCP_ENABLE_LUA -I/usr/include/lua5.2 -D__LINUX__ -D MG_ENABLE_THREADS -fPIC -I../../../../common -I../../../../../common  -c -o ../common/simulation.o ../common/simulation.cpp
g++ -g -O2  -fPIC -I. -I../../../../common -I../../../../../common -I include   -I/usr/local/lib/wx/include/gtk3-unicode-3.1 -I/usr/local/include/wx-3.1 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -D MG_ENABLE_SSL -D VSCP_ENABLE_LUA -I/usr/include/lua5.2 -D__LINUX__ -D MG_ENABLE_THREADS -fPIC -I../../../../common -I../../../../../common  -c -o ../../../../common/vscpremotetcpif.o ../../../../common/vscpremotetcpif.cpp
g++ -g -O2  -fPIC -I. -I../../../../common -I../../../../../common -I include   -I/usr/local/lib/wx/include/gtk3-unicode-3.1 -I/usr/local/include/wx-3.1 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -D MG_ENABLE_SSL -D VSCP_ENABLE_LUA -I/usr/include/lua5.2 -D__LINUX__ -D MG_ENABLE_THREADS -fPIC -I../../../../common -I../../../../../common  -c -o ../../../../common/vscphelper.o ../../../../common/vscphelper.cpp
g++ -g -O2  -fPIC -I. -I../../../../common -I../../../../../common -I include   -I/usr/local/lib/wx/include/gtk3-unicode-3.1 -I/usr/local/include/wx-3.1 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -D MG_ENABLE_SSL -D VSCP_ENABLE_LUA -I/usr/include/lua5.2 -D__LINUX__ -D MG_ENABLE_THREADS -fPIC -I../../../../common -I../../../../../common  -c -o ../../../../common/guid.o ../../../../common/guid.cpp
cc -D__LINUX__ -fPIC -I. -I../../../../common -I../../../../../common -I include   -I/usr/local/lib/wx/include/gtk3-unicode-3.1 -I/usr/local/include/wx-3.1 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -D MG_ENABLE_SSL -D VSCP_ENABLE_LUA -I/usr/include/lua5.2 -D__LINUX__ -D MG_ENABLE_THREADS -fPIC -I../../../../common -I../../../../../common  -c -o ../../../../../common/crc.o ../../../../../common/crc.c
cc -D__LINUX__ -fPIC -I. -I../../../../common -I../../../../../common -I include   -I/usr/local/lib/wx/include/gtk3-unicode-3.1 -I/usr/local/include/wx-3.1 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -D MG_ENABLE_SSL -D VSCP_ENABLE_LUA -I/usr/include/lua5.2 -D__LINUX__ -D MG_ENABLE_THREADS -fPIC -I../../../../common -I../../../../../common  -c -o ../../../../../common/crc8.o ../../../../../common/crc8.c
cc -D__LINUX__ -fPIC -I. -I../../../../common -I../../../../../common -I include   -I/usr/local/lib/wx/include/gtk3-unicode-3.1 -I/usr/local/include/wx-3.1 -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -D MG_ENABLE_SSL -D VSCP_ENABLE_LUA -I/usr/include/lua5.2 -D__LINUX__ -D MG_ENABLE_THREADS -fPIC -I../../../../common -I../../../../../common  -c -o ../../../../../common/mongoose.o ../../../../../common/mongoose.c
../../../../../common/mongoose.c:3985:25: fatal error: openssl/ssl.h: No such file or directory
compilation terminated.
<builtin>: recipe for target '../../../../../common/mongoose.o' failed
make[1]: *** [../../../../../common/mongoose.o] Error 1
make[1]: Leaving directory '/home/vscp/vscp/src/vscp/drivers/level2/sim/linux'
Makefile:20: recipe for target 'install' failed
make: *** [install] Error 2
`
Thank you.

prfi, 2017-03-31:
Did you try to install `libssl-dev`?

mamac, 2017-04-03:
Hi

I installed libssl-dev and restarted build but I noticed that there are issues related to lua, I supposed I had to install  liblua5.2-dev too... restarting to build at the moment...

Thank you!

mamac, 2017-04-03:
OK after installation of libssl-dev and  liblua5.2-dev.
Thank you!

closed 2017-04-03 by mamac