grodansparadis/vscp #225: Web interface hangs on variable list

mamac, 2017-05-13:
VSCP Daemon version: 1.12.21.6 Magnesium
Operating system: Linux 4.9.16-gentoo x86_64 64-bit Little endian
Free memory: 5799202816 bytes

After fresh installation of vscpd, when accessing the web interface on port 8080, going to configuration menu then 'Variable - List', the web page is loading but nothing happens.

Last messages from vscpd on the terminal is below:

`2017-05-13T22:29:14 - Webserver: Host=[127.0.0.1] - req=[/images/logo/vscp_new.png] query=[] method=[GET] 
Failed to write message to log database. Error is: (null) Message is: Webserver: Host=[127.0.0.1] - req=[/favicon.ico] query=[] method=[GET] 

2017-05-13T22:29:14 - Webserver: Host=[127.0.0.1] - req=[/favicon.ico] query=[] method=[GET] 
Failed to write message to log database. Error is: (null) Message is: Webserver: Host=[127.0.0.1] - req=[/vscp] query=[] method=[GET] 

2017-05-13T22:29:21 - Webserver: Host=[127.0.0.1] - req=[/vscp] query=[] method=[GET] 
Failed to write message to log database. Error is: (null) Message is: Webserver: Host=[127.0.0.1] - req=[/vscp/configure] query=[] method=[GET] 

2017-05-13T22:29:23 - Webserver: Host=[127.0.0.1] - req=[/vscp/configure] query=[] method=[GET] 
Failed to write message to log database. Error is: (null) Message is: Webserver: Host=[127.0.0.1] - req=[/vscp/variables] query=[] method=[GET] 

2017-05-13T22:29:25 - Webserver: Host=[127.0.0.1] - req=[/vscp/variables] query=[] method=[GET] `

Thank you

mamac, 2017-05-13:
Just adding that this happens also on ARMBian on a bananaPi so I suspect it is not related to operating system setup.

grodansparadis, 2017-05-13:
Hi,

the variable list, the DM edit, session window is moving to the vscp-ux 
external code (see git) so I will not fix things in the current web 
administrative interface (rather remove the current code). The variable 
functionality is more or less in place in the vscp-ux code now so please 
test it out.

So I will close this.

Cheers
/Ake

closed 2017-05-13 by grodansparadis