grodansparadis/vscp #226: Not possible to connect to remote server

mamac, 2017-11-07:
With version:12.29.2.8 Magnesium

Trying to connect to an older vscpd server, I got the message below.
[image]

Thank you

grodansparadis, 2017-11-07:
Same version of VSCP Works and server?

mamac, 2017-11-08:
Server is VSCP Daemon version: 1.0.1.7 Neon
WORKS is version 12.29.2.8 Magnesium

grodansparadis, 2017-11-08:
I don't say it is but that can be the problem. The reason is the addition of the datetime earlier which broke all old interfaces. So a late version of VSCP Works to an old server and vice versa will have problem.  But VSCP Works worktrough for the upcoming release is still  not done so it can be some other problem also.  Hopefully I will be there in a week or two.

grodansparadis, 2017-11-09:
I can confirm that this was a problem also on later versions, all due to a missing compile switch.  This is fixed in 12.29.2.12 Magnesium.

mamac, 2017-11-09:
Indeed, it works now :-)
Thank you very much Ake!

closed 2017-11-09 by mamac

grodansparadis, 2017-11-09:
Thanks for the heads up!