grodansparadis/vscp #227: All events with same node ID

mamac, 2017-11-09:
In VSCPWORKS, in the event list, all events have the same node ID:
[image]
I can't remember it was like this with previous versions.

I have about 20 nodes with different IDs.

grodansparadis, 2017-11-09:
Yes that looks strange. Will check.
Is the data correct?

mamac, 2017-11-09:
The data is not even correct, for example it receive a temperature message with value -73K, but in reality temperature is positive and sent in Celsius.
Also I cannot connect to nodes with the device configuration window, maybe it is related...

grodansparadis, 2017-11-09:
Probably has to do with the timedate addition.  Sadly I can't check/fix  until Sunday.

grodansparadis, 2017-11-12:
I tested this on my local systems and it works here as far as I can see. The upcoming release (head code) will not be backward compatible so client and servers need to update to major version 12.

mamac, 2017-11-13:
OK, I need to update my server then :-)

closed 2017-11-13 by mamac