grodansparadis/vscp #229: CLASS1_PROTOCOL - PROTOCOL_GENERAL

waaslandwolf, 2017-12-05:
Open VSCP works 12.29.2.6 on Ubuntu
Open Client window
All events, even known good events such as CLASS1_PROTOCOL, HEARTBEAT events, show as CLASS1_PROTOCOL - PROTOCOL_GENERAL.

A previously known good, saved, txset that sends CLASS1_INFORMATION, BUTTON events now shows "Unknown class", "unknown type" event when sent.

grodansparadis, 2018-01-02:
Open this again if it still is a problem in later versions.

waaslandwolf, 2018-01-04:
I did a git pull and build 2 days ago and it can confirm that this problem still persists.

grodansparadis, 2018-01-04:
Strange! I use it on a daily basis here.  Can you save received events and send to me. Also a screen dump would be fine.

waaslandwolf, 2018-01-04:
Ok will do!

 

Dit e-mailbericht is gecontroleerd op virussen met Avast antivirussoftware.
https://www.avast.com/antivirus

grodansparadis, 2018-03-19:
I can't still repeat this behavior. Open if still there.

closed 2018-03-19 by grodansparadis