grodansparadis/vscp #230: Fatal error while compiling

mamac, 2017-12-26:
Hi,

Compiling latest VSCP from master branch.
I saw the message below while compiling:
`../../../../../common/fastpbkdf2.c:20:25:` fatal error: openssl/sha.h: No such file or directory
 `#include <openssl/sha.h>
`                         ^
`compilation terminated.
`Makefile:127: recipe for target 'fastpbkdf2.o' failed
`make[1]: *** [fastpbkdf2.o] Error 1
`make[1]: Leaving directory '/home/vscp/vscp/src/vscp/drivers/level2/wire1/linux'

Compilation keeps going on but I though it was worth a notification.

Thank you,
JM

mamac, 2017-12-27:
Actually same error at the end of compilation, below are the last lines:
`cc -std=c99  -I/usr/lib/arm-linux-gnueabihf/wx/include/gtk2-unicode-3.0 -I/usr/include/wx-3.0 `-D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -pthread -DCBC -I../common -I../../common -DMG_ENABLE_THREADS -DM
`G_ENABLE_IPV6 -DMG_ENABLE_SSL  -DMG_ENABLE_GETADDRINFO -c ../../common/mongoose.c -o mongoose.o
`../../common/mongoose.c:4466:25: fatal error: openssl/ssl.h: No such file or directory
` #include <openssl/ssl.h>
`                         ^
`compilation terminated.
`Makefile:279: recipe for target 'mongoose.o' failed
`make[1]: *** [mongoose.o] Error 1
`make[1]: Leaving directory '/home/vscp/vscp/src/vscp/vscpworks'
`Makefile:14: recipe for target 'all' failed
`make: *** [all] Error 2

I will try to compile on my laptop to see if I got the same error.

grodansparadis, 2017-12-27:
Looks like openssl developer headers is not installed. Try

apt-get install libssl-dev

Cheers
/Ake

mamac, 2017-12-27:
Oups, sorry for that.
Compilation seems to go well now.
Thank you!

closed 2017-12-27 by mamac